adafruit/circuitpython #8382: OSError on PyPortal while writing to SD card

caternuson, 2023-09-08:
### CircuitPython version

```python
Works with 6.3.0.

Tested the following versions and they show the issue:
* 7.3.3
* 8.2.4
* 8.2.5

Example banner from 8.2.5:

Adafruit CircuitPython 8.2.5 on 2023-09-07; Adafruit PyPortal with samd51j20
```
```

### Code/REPL

```python
Code is from Learn guide here:
https://learn.adafruit.com/cleveland-museum-of-art-pyportal-frame/code
```

### Behavior

There is other output, including URLs that include API keys, so not adding that here. At the point the code fails, here's the output:
```python
Reply is OK!
Saving data to  /sd/cache.bmp
.Traceback (most recent call last):
  File "adafruit_pyportal/network.py", line 191, in process_image
  File "adafruit_portalbase/network.py", line 310, in wget
OSError: [Errno 5] Input/output error

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "code.py", line 52, in <module>
  File "adafruit_pyportal/__init__.py", line 336, in fetch
  File "adafruit_pyportal/network.py", line 194, in process_image
OSError: 

No writable filesystem found for saving datastream. Insert an SD card or set internal filesystem to be unsafe by setting 'disable_concurrent_write_protection' in the mount options in boot.py

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```

**NOTE** The message about the filesystem is a red herring. It's just the print in the except block.

### Additional information

Related forum thread:
https://forums.adafruit.com/viewtopic.php?t=204273

caternuson, 2023-09-09:
Here is a simplified version of the Cleveland Museum code that can be used to demonstrate the issue:
```python
import board
from adafruit_pyportal import PyPortal

WIDTH = board.DISPLAY.width
HEIGHT = board.DISPLAY.height

APIURL = "https://openaccess-api.clevelandart.org/api/artworks?cc0=1&has_image=1&indent=2&limit=1&skip=1234"

pyportal = PyPortal(
                    image_json_path=["data", 0, "images", "web", "url"],
                    image_dim_json_path=(["data", 0, "images", "web", "width"],
                                         ["data", 0, "images", "web", "height"]),
                    image_resize=(WIDTH, HEIGHT - 15),
                    image_position=(0, 0),
                    json_path=["data", 0, "title"],
                    )

pyportal.fetch(APIURL)
```

caternuson, 2023-09-09:
**SD CARD INFO** - forgot to mention that a 16GB SD Card formatted FAT32 was placed in the PyPortal for all of the above testing.

Note that this same SD card worked fine when reverting to 6.3.0 for testing. So it does not appear to be an SD card size/speed/format issue. Unless older CP firmware was somehow more forgiving in this regard.

dhalbert, 2023-09-12:
> Here is a simplified version of the Cleveland Museum code that can be used to demonstrate the issue:

I tried this code on a PyPortal with 8.2.5, and it's working for mepc . SD card is a SanDisk 4GB.
```
Adafruit CircuitPython 8.2.5 on 2023-09-07; Adafruit PyPortal with samd51j20
>>> 
>>> import test
Connecting to AP PBE2
Retrieving data...Reply is OK!
image dim: 969 893
original URL: https://openaccess-cdn.clevelandart.org/1973.160/1973.160_web.jpg
convert URL: https://io.adafruit.com/api/v2/danhalbert/integrations/image-formatter?x-aio-key=0629ddf76d6a4be085e6a868a77d5187&width=320&height=225&output=BMP16&url=https://openaccess-cdn.clevelandart.org/1973.160/1973.160_web.jpg
Fetching stream from https://io.adafruit.com/api/v2/danhalbert/integrations/image-formatter?x-aio-key=0629ddf76d6a4be085e6a868a77d5187&width=320&height=225&output=BMP16&url=https://openaccess-cdn.clevelandart.org/1973.160/1973.160_web.jpg
Reply is OK!
Saving data to  /sd/cache.bmp
.............................................................................................................................................................................................................................................................................................Created file of 144138 bytes in 8.7 seconds
>>> 
```
I am running NINA-FW version 1.7.5 on the AirLift ESP32 co-processor. What version is on your ESP32 co-processor.

Maybe this is a certificate issue, since 1.7.5 is needed to talk to AdafruitIO now.

caternuson, 2023-09-12:
Looks like the PyPortal I was using for testing has 1.6.1:
```python
>>> esp.firmware_version
bytearray(b'1.6.1\x00')
>>> 
```

Let me try upgrading...

caternuson, 2023-09-12:
Nina FW updated:
```python
Adafruit CircuitPython 8.2.5 on 2023-09-07; Adafruit PyPortal with samd51j20
>>> import board
>>> from digitalio import DigitalInOut
>>> from adafruit_esp32spi import adafruit_esp32spi
>>> esp32_cs = DigitalInOut(board.ESP_CS)
>>> esp32_ready = DigitalInOut(board.ESP_BUSY)
>>> esp32_reset = DigitalInOut(board.ESP_RESET)
>>> spi = board.SPI()
>>> esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)
>>> esp.firmware_version
bytearray(b'1.7.5\x00')
>>> 
```

But still getting the same error:
```python
Adafruit CircuitPython 8.2.5 on 2023-09-07; Adafruit PyPortal with samd51j20
>>> import test
retrieving url: https://openaccess-api.clevelandart.org/api/artworks?cc0=1&has_image=1&indent=2&limit=1&skip=15987
Connecting to AP [removed]
Retrieving data...Reply is OK!
image dim: 1153 893
original URL: https://openaccess-cdn.clevelandart.org/1964.371.4/1964.371.4_web.jpg
convert URL: [URL removed]
Fetching stream from [URLremoved]
Reply is OK!
Saving data to  /sd/cache.bmp
.Traceback (most recent call last):
  File "adafruit_pyportal/network.py", line 191, in process_image
  File "adafruit_portalbase/network.py", line 310, in wget
OSError: [Errno 5] Input/output error

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "test.py", line 52, in <module>
  File "adafruit_pyportal/__init__.py", line 336, in fetch
  File "adafruit_pyportal/network.py", line 194, in process_image
OSError: 

No writable filesystem found for saving datastream. Insert an SD card or set internal filesystem to be unsafe by setting 'disable_concurrent_write_protection' in the mount options in boot.py
>>>
```

dhalbert, 2023-09-12:
This is so weird. I will re-update my PyPortal to 1.7.5 again, in case my 1.7.5 is not the same. I think I was testing when NINA-FW was being updated.

caternuson, 2023-09-13:
**This is possibly SD card related.** :(

Since the one used above worked when reverting to an older version of CP, it was considered OK. However, just did the simple check of swapping to another SD card - and it now works! Running simple test from above:
```python
Reply is OK!
Saving data to  /sd/cache.bmp
............................................................................................................................................................................................................................................................................................Created file of 144138 bytes in 13.0 seconds
>>>
```

Here are some benchmark data on the two SD cards using Ubuntu's Disk utility.

**Problematic SD Card:**
[image]

**Working SD Card:**
[image]

caternuson, 2023-09-15:
I'm going to consider this to be an SD card issue and close.

Asked the original poster to try this as well, but as yet have not heard back:
https://forums.adafruit.com/viewtopic.php?p=985638#p985638

closed 2023-09-15 by caternuson

anglerfish27, 2023-09-23:
Hello, I was the original person who posted this issue in the Adafruit forums. I apologize for the delay. My mother has some scary medical conditions happening right now that I have been tending to.  Just FYI on my setup. I am the using the Titanto with NINA-FW version 1.7.5 upgraded as well. First thing I did was upgrade it. Per @caternuson I ran an older release, 7.3 and the issue went away (same SD card no change). I have no upgraded to 8.x and I ignored the portions of the "how-to-guide" to add the libs because they are already baked into 8.2.6. I nuked the SPI flash and started fresh. Verified I am on 1.75 of the wifi firmware.

Currently testing, its running on 8.2.6. I hope It doesn't crash with the error mentioned in the adafruit forum. If it does I have several SD cards to try in hopes that solves it somehow. I'm in a wait and see pattern. I'll bench mark my SD cards as well to see what they are doing (what program on Ubuntu are you using?)  I dont own any cheapo brand SD cards they are all big name high speed cards, Samsung, San Disk, Kingston, ect.. of course it could be a bad card. I have a bunch probably around 10 cards so I can test the heck out of them. Hopefully this is a moot point and its fixed!

anglerfish27, 2023-09-23:
FYI I'm using a random program for the Apple store to bench the microSD card. Its writing/reading a random 250MB file to the card and its I/O reads are anywhere from 35-60MB/sec with an overall average of about 39/40MB/sec. Reads are about double that averaging 85MB/sec. I dont know the quality of this program though...I'll keep looking for something a little better.

anglerfish27, 2023-09-23:
So I found another tool, that let me test several gigs of random IO, still averaging slightly higher numbers than my last post 50's for write and 90's for read. This matched up with the Mac's built in system monitor I watched the IO on there as well. Suffice to say the numbers are not dropping anywhere near the danger zone. Now I have to let my microSD card cool off its toasty hot! :)   Cheers.

caternuson, 2023-09-24:
As a simple initial test, can try just running the original example as is using latest release of CircuitPython. See if you can get *any* SD cards working there. That's all I did.

The info about the SD card performance is secondary.

anglerfish27, 2023-09-24:
Yeah I have it up and running on the latest 8x build as you recommended and it is working with a SanDisk microSD card. Now I'm dealing with the secondary bug I seem to have stumbled across its in the forums and they have a github ticket open with the team!