adafruit/circuitpython #8389: Connection between two Pi Pico via i2c does not work

Flashton91, 2023-09-12:
### CircuitPython version

```python
I'm using the latest version 8.2.4 on both Pi Pico
```

### Code/REPL

```python
# same code into both microcontrollers
import busio
import board
from adafruit_bus_device.i2c_device import I2CDevice

i2c = busio.I2C(board.GP3, board.GP2)
while not i2c.try_lock():
    pass
ad = i2c.scan()
print(ad)
```

### Behavior

Hello. I'm trying to connect two Raspberry Pi Pico microcontrollers using i2c.

I'm scanning the i2c bus using scan(). I loaded the same code into both microcontrollers. I'm getting an empty list "[ ]" on both microcontrollers.

Please tell me how to get the address of the second Pi Pico to transfer data to it?

cleverca22, 2023-09-12:
one device needs to act as an i2c controller, the other needs to act as an i2c device

having 2 controllers wont work

Flashton91, 2023-09-12:
Thank you
Please tell me - how to make one of the controllers a slave?

jepler, 2023-09-12:
One board needs to run a board that uses the "i2ctarget" module. there is a simple example shown in the documentation. https://docs.circuitpython.org/en/8.2.x/shared-bindings/i2ctarget/index.html

```python
import board
from i2ctarget import I2CTarget

regs = [0] * 16
index = 0

with I2CTarget(board.SCL, board.SDA, (0x40, 0x41)) as device:
    while True:
        r = device.request()
        if not r:
            # Maybe do some housekeeping
            continue
        with r:  # Closes the transfer if necessary by sending a NACK or feeding dummy bytes
            if r.address == 0x40:
                if not r.is_read:  # Main write which is Selected read
                    b = r.read(1)
                    if not b or b[0] > 15:
                        break
                    index = b[0]
                    b = r.read(1)
                    if b:
                        regs[index] = b[0]
                elif r.is_restart:  # Combined transfer: This is the Main read message
                    n = r.write(bytes([regs[index]]))
                #else:
                    # A read transfer is not supported in this example
                    # If the microcontroller tries, it will get 0xff byte(s) by the ctx manager (r.close())
            elif r.address == 0x41:
                if not r.is_read:
                    b = r.read(1)
                    if b and b[0] == 0xde:
                        # do something
                        pass
```

(target mode is the same as "s--" mode but we prefer not to use that terminology in circuitpython)

dhalbert, 2023-10-25:
Closing due to inactivity.

closed 2023-10-25 by dhalbert