adafruit/circuitpython #8403: [RP2040][bug] microcontroller.cpu.temperature , negative reading

mzhboy, 2023-09-15:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.5 on 2023-09-07; VCC-GND Studio YD RP2040 with rp2040
Board ID:vcc_gnd_yd_rp2040
```

### Code/REPL

```python
>>> import microcontroller
microcontroller.cpu.temperature
-1479.79
>>> board.board_id
'vcc_gnd_yd_rp2040'
```

### Behavior

got negative reading **-1479.79**

mikepaxton, 2023-09-18:
I'm seeing  the same issue but only after connecting the 5v lead from the Pico to the  Adafruit 16x2 RGB Char LCD Shield (Adafruit Part#[714](https://www.adafruit.com/product/714)). 
 If the Shield is not powered on then microcontroller.cpu.temperature returns a correct cpu temp around 34° Celsius.  As soon as I boot the Pico with the Shield attached via i2c and powered from the Pico's 5v vbus the cpu temp is reported as a negative number usually around the -70° Celsius range.

Pico W
CircuitPython 8.2.4

eightycc, 2023-11-20:
@mzhboy There's a solder jumper on the Studio YD boards that needs to be connected in order to supply the ADC_VREF to the RP2040. Without it, the RP2040's ADC can't function so you will not get a valid temperature reading.
[image]

dhalbert, 2023-12-11:
Closing this due to the apparent fix. Please reopen if this does not solve the problem.

closed 2023-12-11 by dhalbert