adafruit/circuitpython #8405: Web-workflow AP fallback

bill88t, 2023-09-15:
This is less of a feature request and more of a request to get down to decide an api for it.
I would like to try implementing it, but I want your input on how it will work.

I was thinking of making a fallback of sorts.
If the STA network `/settings.toml` specifies is not found, or could not be connected, check if AP credentials exist and make an AP.

`CIRCUITPY_AP_SSID`
`CIRCUITPY_AP_PASSWORD`

Also it would perhaps be worth considering installing some random defaults onto the filesystem upon formatting.
This would remove the need to go and do:
```py
a = open("/settings.toml", "w")
a.write(..)
```
on every wipe.

DJDevon3, 2023-09-15:
I agree with this. Having a user create a secrets.py or settings.toml instead of providing one in the file system isn't intuitive especially for a first timer.  For those of us who constantly play with new boards, having to create the file, and fill it with relevant credentials with copy/paste gets tedious.  Even if the settings are blank it would be nice to have a default settings.toml included with all possible fields that can be used in a settings.toml file.

tannewt, 2023-09-15:
> check if AP credentials exist and make an AP.

What do you gain by making an AP? I don't think web workflow works on one yet.

> Even if the settings are blank it would be nice to have a default settings.toml included with all possible fields that can be used in a settings.toml file.

This will take up more flash space because we'd need to store a copy of the default file in the image. I think it'd make it *harder* to write to since you'd need to edit a line instead of appending.

The WebSerial installer on circuitpython.org is meant to alleviate this by installing the file for you with the credentials you enter.

> on every wipe.

How often are you wiping out the file system? It shouldn't happen very often.

anecdata, 2023-09-15:
I don't think we automatically want an AP started by default on every CircuitPython install (or storage.erase_filesystem()) with no user intervention. An AP uses board resources and creates security and privacy vulnerabilities that don't require physical access to the device.

But an easy way to optionally enable web workflow via AP would be very welcome.

bill88t, 2023-09-15:
> This will take up more flash space because we'd need to store a copy of the default file in the image. I think it'd make it harder to write to since you'd need to edit a line instead of appending.

It could just be generated at fs creation time.
There is no reason not to just touch and append to it whatever we generate.
This way, we could use the `#if`'s to very quickly append placeholders for whatever the board has.

> What do you gain by making an AP? I don't think web workflow works on one yet.

Well, "yet". If it did it work it would make it really convenient for when you are outside your home, away from the regular wifi access point. Some boards are portable and suitable for outdoors.
But I guess ftp now exists for those that are serious about getting files in and out of the board.

> creates security and privacy vulnerabilities

Random password, I did not suggest an open network.
The password would only be accessible to whom has access to serial to read it off of `settings.toml`.
Doing a `with open("settings.toml") as f: print(f.read())` is much simplier.

If the wifi security is broken we have much bigger worries.

anecdata, 2023-09-16:
From the OP, this is the key:

> check if AP credentials exist and make an AP

An auto-generated random password doesn't solve the problem of not having serial access to the device. And if you do have serial access, AP might be a convenience, but you could do anything over serial without needing an AP.

 I would not want every new device to start an AP by default, increasing complexity, using resources, broadcasting an SSID, and revealing its permanent hardware MAC address (and by inference, vendor).

But an optional mirror to the Station web workflow mechanism for AP web workflow would be awesome. With either CDC or MSC access, the credentials could easily be added to devices where it's desired.