adafruit/circuitpython #8406: Add Scientific Pitch Notation converter helper to synthio

CedarGroveStudios, 2023-09-16:
In addition to the MIDI note value to frequency converter helper already in synthio, it would be very handy to include a converter for a note's Scientific Pitch Notation (SPN) string to a frequency.

Here's an example of a helper in CedarGroveStudios/CircuitPython_MIDI_Tools (also in the community bundle) that converts SPN to a MIDI note value.

```python
def name_to_note(name):
    """Translates a note name to a MIDI sequential note value. Note names are
    character strings expressed in Scienfic Pitch Notation (NoteOctave) format
    such as 'C4' or 'G#7'. Note names range from 'C-1' (note value 0) to
    'F#9' (note value 127). Note values are of integer type in the range of
    0 to 127 (inclusive). If the input value is outside that range, the value
    of `None` is returned.

    :param str name: The note name input in SPN format. No default value.
    """
    name = name.upper()  # Convert lower to uppercase
    if name[:-1] in NOTE_BASE:
        # Note name is valid
        note = NOTE_BASE.index(name[:-1])
        if "-1" in name:
            # Special case for octave -1
            note = NOTE_BASE.index(name[:-2])
            octave = -1
        else:
            note = NOTE_BASE.index(name[:-1])
            octave = int(name[-1])
        return min(max(0, note + (12 * (octave + 1))), 127)  # MIDI note value
    return None  # Input string is not in NOTE_BASE
```

tannewt, 2023-09-18:
Why should this be in the core instead of a library?

jepler, 2023-09-18:
FWIW I'd say that midi note numbering is supported just as a sort of historical accident: the original synthio was a midi player and couldn't play notes with arbitrary frequency in Hz.

Another reason to use something outside of the core for conversion from musical notation to frequency is to support different kinds of [Musical Temperament](https://en.wikipedia.org/wiki/Musical_temperament), rather than the equal temperament that is the one that you get with MIDI note numbers. we wouldn't want to add all sorts of temperaments in the core.

CedarGroveStudios, 2023-09-18:
Good point and interesting historical reference. The SPN conversion method and frequency to MIDI note value methods are already in the Community Bundle, BTW.

closed 2023-09-18 by CedarGroveStudios