adafruit/circuitpython #8409: `listdir()` vs `listdir("./")` does not work the same way on `/`

bill88t, 2023-09-18:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.1-40-g17ced21e04 on 2023-09-14; M5Stack Timer Camera X with ESP32
```

### Code/REPL

```python
# mount a vfs under /something
import os
os.listdir()
os.listdir("./")
```

### Behavior

```
>>> listdir()
['ramdisk', 'LjinuxRoot', 'code.py', 'lib', 'boot_out.txt', 'ftp_serv.py', 'boot.py', 'repl.py', 'settings.toml']
>>> listdir("./")
['LjinuxRoot', 'code.py', 'lib', 'boot_out.txt', 'ftp_serv.py', 'boot.py', 'repl.py', 'settings.toml']
>>> 
```

### Description

I just started playing with ramdisk more, and I noticed my `ls` was just.. not seeing it..

RetiredWizard, 2023-09-18:
I believe I ran into a similar issue with SD card mounts.

RetiredWizard, 2023-09-18:
When the os.listdir('..') is run from a mounted directory (SD card anyway) the mounted folder is listed rather than the root folder. I.E. the ".." directory shortcut can't parse up thorough a mount point.

PR #10263 (2025-04-22): Add working_directory for subsequent code file
This allows subfolders to be treated similar to / for multiple apps within different folders.

Also, fix up the internal current working directory so it doesn't depend on volumes.

Fixes #9045 and fixes #8409.
commit 33777650 (2025-04-22): Add working_directory for subsequent code file

This allows subfolders to be treated similar to / for multiple
apps within different folders.

Also, fix up the internal current working directory so it doesn't
depend on volumes.

Fixes #9045 and fixes #8409.
closed 2025-04-22 by tannewt