adafruit/circuitpython #8413: Incorrect microcontroller defined for CircuitPython Adafruit monster m4sk

stampe65, 2023-09-18:
### CircuitPython version

```python
CircuitPython 8.2.6
```

### Code/REPL

```python
circuitpython/ports/atmel-samd/boards/monster_m4sk/mpconfigboard.mk

USB_VID = 0x239A
USB_PID = 0x8048
USB_PRODUCT = "Monster M4SK"
USB_MANUFACTURER = "Adafruit Industries LLC"

CHIP_VARIANT = SAMD51J19A
CHIP_FAMILY = samd51

QSPI_FLASH_FILESYSTEM = 1
EXTERNAL_FLASH_DEVICES = "GD25Q64C,W25Q64JVxQ"
LONGINT_IMPL = MPZ
CIRCUITPY_SYNTHIO = 0
```

### Behavior

The Circuit Python .UF2 file downloaded from https://circuitpython.org/board/monster_m4sk/ does not work

### Description

The circuitypython mpconfigboard.h and mpconfigboard.mk files for the monster_m4sk board incorrectly identify the Adafruit MONSTER M4SK microcontroller as a SAMD51J19A chip (QFP64) when it is actually a SAMD51G19 chip (QFP48).  This seems to be the root cause of the Circuit Python .UF2 file downloaded from https://circuitpython.org/board/monster_m4sk/ not working.

dhalbert, 2023-09-18:
It is true that it should be a SAMD51G19A, and we should fix that. but I downloaded CircuitPython 8.2.6 successfully and it came up. When you say "not working", what did not work?

dhalbert, 2023-09-18:
Wrong CPU fixed by #8414 and should be in the next 8.2.x CircuitPython release.

dhalbert, 2023-09-19:
Fixed in 8.2.x by #8414; will be merged into `main` at the next 8.2.x -> main merge.

closed 2023-09-19 by dhalbert