adafruit/circuitpython #8415: Any reason why `time.sleep()` does not support microseconds on RP2040?

ZodiusInfuser, 2023-09-19:
Hi,

I am attempting to implement half duplex uart for a feature on the upcoming Pimoroni Yukon board, and am in need of doing sub millisecond time delays. Micropython has `sleep_us` on their RP2040 build but CircuitPython does not, and although the regular `sleep` function accepts any floating point value, it is quantised to milliseconds.

Here is my analyser output from running `time.sleep()` with any microsecond value:
[image]
In it the uart TX is being routed to the half duplex line based on the direction pin. As you can see, the direction pin changes before all the data is sent, meaning it gets cut off.

Doing a 1 millisecond delay has the direction change take too long:
[image]

Yet, if I add in my own sleep_us function to the CircuitPython build, I can get the direction change happening exactly where it's needed with a `time.sleep_us(350)` delay.
```c
STATIC mp_obj_t time_sleep_us(mp_obj_t microseconds_o) {
    mp_int_t usecs = mp_obj_get_int(microseconds_o);
    if (usecs < 0) {
        mp_raise_ValueError(translate("sleep length must be non-negative"));
    }
    common_hal_time_delay_us(usecs);
    return mp_const_none;
}
MP_DEFINE_CONST_FUN_OBJ_1(time_sleep_us_obj, time_sleep_us);
```
[image]

This was very easy for me to add, which makes me think there is a good reason why CircuitPython does not have this ability? My immediate thought is boards that do not support us delays, which my code above does not check for.

Perhaps someone could point me to how I could make this change just for RP2040, or maybe just for the Pimoroni Yukon, which I already have an in-progress PR for (https://github.com/adafruit/circuitpython/pull/7440)

Thanks

deshipu, 2023-09-19:
https://docs.circuitpython.org/en/latest/shared-bindings/microcontroller/index.html#microcontroller.delay_us

deshipu, 2023-09-19:
The reason why this is not in the time module is compatibility with cpython.

jepler, 2023-09-19:
I'm not sure why time.sleep can't call the microsecond sleep function, though, similar to (untested)
```c
STATIC mp_obj_t time_sleep(mp_obj_t seconds_o) {
    mp_float_t seconds = mp_obj_get_float(seconds_o);
    if (seconds < 0) {
        mp_raise_ValueError(translate("sleep length must be non-negative"));
    }
    if (seconds < MICROPY_FLOAT_CONST(.001)) {
        mp_float_t usecs = MICROPY_FLOAT_CONST(1e-6) * seconds + MICROPY_FLOAT_CONST(0.5);
        common_hal_time_delay_us(usecs);
    } else {
        mp_float_t msecs = MICROPY_FLOAT_CONST(1e-3) * seconds + MICROPY_FLOAT_CONST(0.5);
        common_hal_time_delay_ms(msecs);    
    }   
    return mp_const_none;
}
MP_DEFINE_CONST_FUN_OBJ_1(time_sleep_obj, time_sleep);
```

dhalbert, 2023-09-19:
It could also use the microsecond sleep for that part of the requested sleep that was less than 1 msec, and use the millisecond sleep for the other part. The microsecond sleep is a busy-wait loop. It doesn't turn off interrupts, so any delay might be longer than requested due to intervening interrupt handling. There are routines to turn interrupts off and on.

jepler, 2023-09-19:
I don't think this bug is specific to rp2, the implementation of `time_sleep` that always calls into `delay_ms` is in shared-bindings.

ZodiusInfuser, 2023-09-19:
Thanks @deshipu! I was completely unaware of that function existing, so that explains my confusion about it being seemingly absent.

@jepler That would be nice, though for my needs I wouldn't want to push for a potentially breaking change like that.

tannewt, 2023-09-19:
I'm ok with time.sleep() not being super precise because python code execution isn't really either. A GC or background task can take non-trivial time at nearly any point. Python code really shouldn't assume precise timing.

joehopfield, 2025-01-24:
I propose broadening this enhancement request to include all time related functions that currently round or truncate unnecessarily.

Time related functions need not arbitrarily round time/sleep resolution. Doing so wastes cycles (maths) and makes polling *less* performant. A millisecond is a long time on an mcu running at 240mhz, even in an interpreter.

I disagree with the argument that "not super precise" is ok for python because of interrupts or gc - 
Interrupts, scheduling of other processes or io, these happen on any multi-process system, and the semantics of the various sleep flavors across many platforms point this out.

POSIX says "suspension time may be longer", as does macos, as does normal python.
And Zephyr supports k_usleep() (with those same semantics).

The Zephyr port is a good opportunity to remove arbitrary rounding/loss of precision.

My particular use-case is for servo current monitoring, but this is a general os principle (and mcus will continue to get faster and more capable)

If backwards compatibility is a real concern (for code that can't be tested), a resolution value could be exposed by time_*

joehopfield, 2025-01-24:
asyncio.sleep is the most important of these, sorry I meant to include...