adafruit/circuitpython #8416: TinyS3 board digitalio shared module does not work

qutefox, 2023-09-19:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.6 on 2023-09-12; TinyS3 with ESP32S3
```

### Code/REPL

```python
import digitalio
import time
import board

time.sleep(5) # give us some time to update code via USB when we are stuck in a crash loop.
tmp = digitalio.DigitalInOut(board.D34)
tmp.direction = digitalio.Direction.OUTPUT
```

### Behavior

Board crashes and hard resets

### Description

When I try to set D34 pin as output using the digitalio module the board hard resets.
This always happens. The board doesn't give any error message so finding the root cause was really hard.

### Additional information

One workaround I've found is:
```python
import digitalio
import time
import board

time.sleep(5) # give us some time to update code via USB when we are stuck in a crash loop.
tmp = digitalio.DigitalInOut(board.D34)
tmp.switch_to_output(True)
```
Then the board does not crash during setup. But when I try to set the value it will still crash so the workaround doesn't really help me for long:
```python
tmp.value = False # crashes the board
tmp.switch_to_output(False) # this crashes the board as well, workaround doesn't work anymore
```

tannewt, 2023-09-20:
Do other pins work ok? Does GPIO34 fail on other S3 boards?

anecdata, 2023-09-20:
I can't reproduce`:
```py
Adafruit CircuitPython 8.2.6 on 2023-09-12; TinyS3 with ESP32S3
>>> import digitalio
>>> import board
>>> 
>>> tmp = digitalio.DigitalInOut(board.D34)
>>> tmp.direction = digitalio.Direction.OUTPUT
>>> tmp.value = True
>>> tmp.value = False
>>> 
>>> tmp.switch_to_input()
>>> tmp.value
False
>>> 
>>> tmp.switch_to_output(False)
>>> tmp.value = True
>>> tmp.value = False
>>> 
```

qutefox, 2023-09-20:
Is there a way for me to somehow capture the reason of the crash on my board? Like a debugger just plugged in connected to some of the pins? Unexpected maker also said that I should downgrade the firmware and try 8.2.3 or 8.2.4 because he uses that and that is what I should use as well.

tannewt, 2023-09-20:
`DEBUG=1` builds of CP will usually output the ESP-IDF logging onto the UART. That's how I debug crashes. The adafruit discord is a good place to get help doing that if you need it. https://adafru.it/discord in #circuitpython-dev

qutefox, 2023-09-20:
Okay. I can still reproduce issue on my board. Going to make a build on my Ubuntu machine and come back here with my findings.

qutefox, 2023-09-24:
I have the debug build.

(It was exciting to make it because the build dependencies had a mismatch:
 * esp-idf/install.sh wanted jinja2 version 3.03 
 * the "requirements-dev.txt" wanted to install jinja2 version 3.1.2 (which I guess is the latest
 Build requirment from: https://github.com/adafruit/esp-idf/blob/f50c59e5ec3d28921fda41c1ef207966df6b3233/requirements.txt#L29
"jinja2<3.1; python_version < "3.11"   # See https://github.com/espressif/esp-idf/issues/8760"
So I had to reseacrh a bit and manually swap versions. But now it is working.)

It seems that the example code is not enough. First I had a lot more code running and since I got crashes and hard resets I started to komment out code segements and the very last bit was that digitalio code.
Interestingly when I put back the digitalio my board crashes, if I remove my board does not crash.

Now on to the debug output:
`
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x16a4
load:0x403b6000,len:0xeac
load:0x403ba000,len:0x2f7c
entry 0x403b6328
I (25) boot: ESP-IDF v4.4-dev-3608-gbbe2a1bf34 2nd stage bootloader
I (25) boot: compile time 11:13:59
I (25) boot: chip revision: 0
I (28) qio_mode: Enabling default flash chip QIO
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : QIO
I (43) boot.esp32s3: SPI Flash Size : 8MB
I (47) boot: Enabling RNG early entropy source...
W (53) bootloader_random: RNG for ESP32-S3 not currently supported
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (78) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (86) boot:  2 ota_0            OTA app          00 10 00010000 00200000
I (93) boot:  3 ota_1            OTA app          00 11 00210000 00200000
I (100) boot:  4 uf2              factory app      00 00 00410000 00040000
I (108) boot:  5 ffat             Unknown data     01 81 00450000 003b0000
I (116) boot: End of partition table
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x16a4
load:0x403b6000,len:0xeac
load:0x403ba000,len:0x2f7c
entry 0x403b6328
I (25) boot: ESP-IDF v4.4-dev-3608-gbbe2a1bf34 2nd stage bootloader
I (25) boot: compile time 11:13:59
I (25) boot: chip revision: 0
I (28) qio_mode: Enabling default flash chip QIO
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : QIO
I (43) boot.esp32s3: SPI Flash Size : 8MB
I (48) boot: Enabling RNG early entropy source...
W (53) bootloader_random: RNG for ESP32-S3 not currently supported
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (78) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (86) boot:  2 ota_0            OTA app          00 10 00010000 00200000
I (93) boot:  3 ota_1            OTA app          00 11 00210000 00200000
I (101) boot:  4 uf2              factory app      00 00 00410000 00040000
I (108) boot:  5 ffat             Unknown data     01 81 00450000 003b0000
I (116) boot: End of partition table
I (620) esp_image: segment 0: paddr=00010020 vaddr=3c130020 size=4d710h (317200) map
I (673) esp_image: segment 1: paddr=0005d738 vaddr=3fc9cf70 size=028e0h ( 10464) load
[... 113 lines cut ...]
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x16a4
load:0x403b6000,len:0xeac
load:0x403ba000,len:0x2f7c
entry 0x403b6328
`

It complains about **E (13450) i2c: i2c_driver_delete(411): i2c driver install error**
Trying to understand why. Maybe I should post my whole code.
I use an IO expander over I2C.
`from adafruit_mcp230xx.mcp23017 import MCP23017
mcp23017 = MCP23017(board.I2C())
`
And it initializes correctly. When the chip is not soldered I get an I2C error complaining about address 0x20 not being found.
But now the just works until it crashes.

My number one suspicion is that the  adafruit_mcp230xx library wraps the digitalio mode and offers similar things:
`pin_power_12v_enable = mcp23017.get_pin(2)
 pin_power_12v_enable.switch_to_output(False)`

Maybe the library messes up something and that is why digitalio calls causes some I2C crash.
Will look into the source code of that library.

qutefox, 2023-09-24:
Here is my full code if you are interested:
[tinys3_crashes.zip](https://github.com/adafruit/circuitpython/files/12708235/tinys3_crashes.zip)

dhalbert, 2023-09-24:
Try using the `8.2.x` branch if your main aim is to get a working build. The latest `main` was just updated to ESP-IDF v5.1, and probably has some regression.

@tannewt Note the `jinja2` version skew issue above.

qutefox, 2023-09-25:
Case can be closed. When I pull my gpio low it causes a short on the power rail and that causes the hard reset.
tannewt from the Discord channel noticed that the reason of the startup in the debug output is "POWERON" so it isn't a code problem but a hardware problem on my part.

closed 2023-09-25 by qutefox