adafruit/circuitpython #8417: Cannot read data from usb_cdc.data

forem1, 2023-09-20:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.6 on 2023-09-12; Raspberry Pi Pico with rp2040
```

### Code/REPL

```python
import usb_cdc

usb_cdc.data.timeout = 1

def send_data(data):
    data = bytes(data, 'utf-8')
    usb_cdc.data.write(data)

while True:
    if usb_cdc.data.in_waiting > 0:
        data = usb_cdc.data.read(usb_cdc.data.in_waiting)
        print(data)
        usb_cdc.data.write(bytes("test\n", 'utf-8'))
```

### Behavior

When attempting to retrieve any number of characters or a line from `usb_cdc.data`, the program becomes unresponsive. If set to read timeout, the program will continue to run, but does not provide any data. In all cases, the count of input symbols in the buffer (`in_waiting`) matches the number of characters sent. 
Sending characters functions correctly - the issue only affects reading.

### Description

Under certain conditions (it was not possible to determine which ones) reading works. Soft and hard reboot, power restart doesn't help. If set `console=False` in `boot.py` the reaction doesn't change. Nothing’s connected to the board. In version 7.3.2 read work perfectly.

dhalbert, 2023-09-20:
What is the host computer, and do you have a simple test script you are running on the host? We would like to try to duplicate your setup.

forem1, 2023-09-20:
I use Windows 11 Pro on my computer. To work with the port I tried to use the terminal programs "Termite", "Serial Port Utility" and the web terminal "Serial terminal" from Google chrome labs. All programs receive data from Console, but not from Data.

dhalbert, 2023-09-20:
And could you post your `boot.py`?

forem1, 2023-09-20:
`boot.py` file:
```
import usb_cdc
usb_cdc.enable(console=True, data=True)
```

I would also like to note that the `boot_out.txt` file has no errors.

nhulsey33, 2023-09-27:
I ran into this exact problem when using a serial port with visual basic. To fix this DTR needs to be asserted by the host machine. Can you verify that is the case? Once I did that the program began working. Could be worth checking.....

Under "connected:bool" there is a note that describes this problem.
https://docs.circuitpython.org/en/latest/shared-bindings/usb_cdc/index.html

dhalbert, 2023-09-27:
@nhulsey33 Thanks, I was looking for that reference to C# and couldn't find it.

dhalbert, 2023-11-07:
@forem1 I could not reproduce this problem on a Pi Pico with your exact test program, and with CircuitPython 7.3.3, 8.2.7, and 9.0.0-alpha.2. For all of these, I saw both input and output on the `data` CDC port. I tested on Windows 10 Pro and Windows 11, Pro mostly with the Tera Term program, but also tested 9.0.0-alpha.2 with Termite and the Google Chrome terminal app. I opened both COM ports in Tera Term, and with Termite, I opened the `data` port in Termite, and the console port in Tera Term.

berkut0, 2023-12-26:
Hi
Can confirm that behaviour
**And DTR enabling helps**
I don't think it's really important which board to use, but I'm using rp2040 now

dhalbert, 2023-12-26:
I am going to close this since we believe that enabling DTR is the key. If the original poster is still having trouble, please reopen.

closed 2023-12-26 by dhalbert