adafruit/circuitpython #8420: Wifi authmodes irregularities

bill88t, 2023-09-21:
### CircuitPython version

```python
Latest master, current commit e39fbf1b26b4fd3b66313e51ccc3db0eba7bd58a
Discovered on s3 while testing idf5, also exhibited on regular esp32.
```

### Code/REPL

```python
import wifi
a = wifi.radio.start_scanning_networks()
for i in a:
    print(i.ssid)
    print(i.authmode)
a = wifi.radio.stop_scanning_networks()
```

### Behavior

Something is wrong with authmodes, and it's consistent.

main:
```
WIND_2.4G_422DEA
[wifi.AuthMode.WPA2, wifi.AuthMode.PSK]
COSMOTE-478503
[wifi.AuthMode.WPA2, wifi.AuthMode.PSK]
Rock4CPlus
[wifi.AuthMode.WPA, wifi.AuthMode.PSK]
KeyFalse
[wifi.AuthMode.OPEN]
Forthnet-5DbGR_EXT
[wifi.AuthMode.WPA2, wifi.AuthMode.PSK]
VODAFONE_WIFI_108-07306
[wifi.AuthMode.WPA, wifi.AuthMode.WPA2, wifi.AuthMode.PSK]
COSMOTE-cg29sh
[wifi.AuthMode.WPA2, wifi.AuthMode.PSK]
WIND_2.4G_650701
[wifi.AuthMode.WPA2, wifi.AuthMode.PSK]
Feline34
[wifi.AuthMode.WPA, wifi.AuthMode.WPA2, wifi.AuthMode.PSK]
```

8.2.6:
```
Rock4CPlus
[wifi.AuthMode.WPA, wifi.AuthMode.PSK]
KeyFalse
[wifi.AuthMode.WPA, wifi.AuthMode.WPA2, wifi.AuthMode.PSK]
Forthnet-5DbGR_EXT
[wifi.AuthMode.WPA2, wifi.AuthMode.PSK]
Forthnet-AKzTk
[wifi.AuthMode.WPA, wifi.AuthMode.WPA2, wifi.AuthMode.PSK]
COSMOTE-04829C
[wifi.AuthMode.WPA2, wifi.AuthMode.PSK]
COSMOTE-cg29sh
[wifi.AuthMode.WPA2, wifi.AuthMode.PSK]
Feline34
[wifi.AuthMode.WPA, wifi.AuthMode.WPA2, wifi.AuthMode.PSK]
WIND_2.4G_650701
[wifi.AuthMode.WPA2, wifi.AuthMode.PSK]
```

### Description

This also prevents connections to those networks that are reported incorrectly.
Sometimes the networks are reported to not be found, and sometimes `unknown failure 15`.

tannewt, 2023-09-21:
Is the 8.2.6 output correct?

bill88t, 2023-11-02:
> Is the 8.2.6 output correct?

From a small amount of testing yes. I did not test 8.x in many places.
The number and authmodes of nearby networks affects it.

This is an upstream bug, I filed a bug.
https://github.com/espressif/esp-idf/issues/12517

bill88t, 2023-11-06:
This bug has been confirmed in:
 - CiruitPython
 - Arduino beta esp plugin 3.0.0-alpha2
 - ESP-IDF wifi example scan

On all esp32 chip variants.
If we are to ship idf5, we need to disable all authmode checks.

bill88t, 2024-01-31:
This issue has been fixed upstream with https://github.com/espressif/esp-idf/commit/615d928aa59315f81f052b34f59db7eb502e8199.

Once we upgrade esp-idf in CP, this will be fixed.
Do note however, some more authmode changes have been made since last update.

bill88t, 2024-04-14:
This has been fixed. Closing.

closed 2024-04-14 by bill88t