adafruit/circuitpython #8425: In KMK python firmware, Specifying "KC.MO(1)" in code.py crashes CircuitPython

Bob837217, 2023-09-23:
### CircuitPython version

```python
Version: Adafruit CircuitPython 9.0.0-alpha.1-37-g4563c35908 on 2023-09-12; EK68Custom with rp2040
```

### Code/REPL

```python
Here are two minimal examples:
This code works with no crash:

    import time
    from kmk.kmk_keyboard import KMKKeyboard
    from kmk.modules.layers import Layers
    from kmk.keys import KC
    keyboard = KMKKeyboard()
    FN = KC.MO(1)
    layers = Layers()
    while True:
        print("Waiting")
        time.sleep(2)

And this code causes a crash:

    import time
    from kmk.kmk_keyboard import KMKKeyboard
    from kmk.modules.layers import Layers
    from kmk.keys import KC
    keyboard = KMKKeyboard()
    layers = Layers()
    FN = KC.MO(1)
    while True:
        print("Waiting")
        time.sleep(2)

The difference is that "FN = KC.MO(1)" occurs after "layers = Layers()" - this causes crash.
Also if (in my full keymap), I reference "KC.MO(1)" directly instead of the non working "FN" alias, this will crash the same way.
```

### Behavior

Describe the bug
In my code.py, (for latest KMK firmware) if I put an an alias like "FN = KC.MO(1)" after a "Layers()" instantiation, CircuitPython crashes, enters Safe Mode and reports that a crash occurred.

REPL output:

Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

dhalbert, 2023-09-23:
The 9.0.0 releases are very early, and not meant for general use.  But thanks for reporting this.

What Safe Mode error is reported?

If you use 8.2.6, does it also crash?

Bob837217, 2023-09-23:
How to tell the Safe Mode error? The REPL contains the included text, but no mention of the specific error.

dhalbert, 2023-09-23:
It should print one of these general errors after reporting it was in Safe Mode:
https://learn.adafruit.com/circuitpython-safe-mode/safe-mode-reasons#internal-software-errors-3138327

Bob837217, 2023-09-23:
It's already in the bug description text:
"Hard fault: memory access or instruction error."
This is the third such error from the link you gave,

dhalbert, 2023-09-23:
Ah sorry, I missed that in the original post. In any case, does it also fail on 8.2.6?

Bob837217, 2023-09-23:
Compiling now, will report later.

Bob837217, 2023-09-23:
Managed to checkout remote branch "8.2.x".  The issue is now resolved.  Didn't realise I needed to checkout the latest remote branch for this.  
From REPL:
Adafruit CircuitPython 8.2.6-5-g37180b28dd on 2023-09-23; EK68Custom with rp2040

Will close this off.  Thanks for your assistance.

dhalbert, 2023-09-23:
9.0.0-alpha.1-something reflects a bunch of changes that could have caused regressions, mostly at the moment a merge from Micropython v1.19.1. So the crash you describe is interesting, and we should look at it, and I will reopen this issue for that reason. But if you're trying to get something to run for your board, by all means use 8.2.6.

Bob837217, 2023-09-23:
Thanks @dhalbert .  Good luck with the bug hunt. I'm just trying to get a stable build for my own PCB so I can use it for a keyboard product that I bought, but did not like the limitations that their mystery MCU gave me.  My PCB is mechanically and electrically compatible with the commercial PCB, and is a drop in replacement.

dhalbert, 2023-12-01:
It would be great to have a smaller example that does not require full KMK to reproduce.

tannewt, 2024-01-31:
Please re-test this with a 9.0 beta release. If it is still an issue, then file a new issue referencing this one.

closed 2024-01-31 by tannewt

DLBPointon, 2024-06-23:
@Bob837217 
Apologies for commenting here, but it is the only place I have seen a similar issue to what I am having.
Hi, did you manage to solve this at all?

I've tested this on all major and latest beta releases of CircuitPython since 8.2.8.

Running the layer assignment before appending Layer(), ends up with the keyboard not working due to `Invalid LT key`, adding it  after the assignment there is an `Invalid HT key` both of which stop the board working. removing the new layer results in a working board but invalid HT. The layers are kinda important for me as I am trying to write some KMK firmware for https://artsey.io/

I am also detailing the issue more here: https://kmkfw.zulipchat.com/#narrow/stream/358347-general/topic/Extensions.20not.20working.20for.20mini.20keyboard

Thanks

Bob837217, 2024-06-23:
Hi @tannewt @DLBPointon
It has been a while since I last did any work on that keyboard project.  I have not tested the latest beta etc and do not know if this is still an issue or not.  I did things in a different way in KMK to avoid any issue for my project in the end.  I also note that for my final version (last I looked at it), things were stable for a couple days uptime, then the rp2040/firmware would freeze and need to be power cycled to regain functionality.  I have moved onto other personal projects and won't be looking at this one again unless I have a serious need to.