adafruit/circuitpython #8426: displayio.Bitmap value_count max should be 65536 not 65535

rimwolf-redux, 2023-09-23:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.6 on 2023-09-12; Adafruit Feather M4 Express with samd51j19

Supplying 65536 as the value_count argument to a displayio.Bitmap constructor yields
an error that value must be 1..65535:

>>> import displayio
>>> bitmap = displayio.Bitmap(160, 120, 65536)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: value_count must be 1-65535
>>>

But in a colorspace like RGB565, all 65536 16-bit values are valid. As for user friendliness, all three of the examples in https://github.com/adafruit/Adafruit_CircuitPython_OV7670/blob/main/examples that use displayio.Bitmap supply 65536 as an argument.

The code in https://github.com/adafruit/circuitpython/blob/main/shared-bindings/displayio/Bitmap.c that calculates the bits (per pixel) from the value_count subtracts 1 from the value_count, so correctly yields 16 for 65536 as well as 65535.
```

### Code/REPL

```python
import displayio
bitmap = displayio.Bitmap(160, 120, 65536)
```

### Behavior

```python
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: value_count must be 1-65535
```

### Additional information

Changing the upper limit to 65536 in the definition of value_count in Bitmap.c (line 67 in the current github version) looks like it would resolve this issue.

jepler, 2023-09-25:
I agree! would you be able to submit a pull request to fix this?

rimwolf-redux, 2023-09-25:
It's been some time since I've built CircuitPython, but I can give it a shot. This should be against 8.2.x (git checkout 8.2.x) I guess?

tannewt, 2023-09-25:
> It's been some time since I've built CircuitPython, but I can give it a shot. This should be against 8.2.x (git checkout 8.2.x) I guess?

Yes please. That way it'll get into a stable release more quickly. (We periodically merge it back to main too.)

rimwolf-redux, 2023-09-26:
Submitted PR #8434 to fix this issue.

isacben, 2023-10-09:
I think this issue is solved. ?

PR #8434 (2023-09-27): Fixed displayio/Bitmap value_count range check
This PR changes the upper limit of display/Bitmap value_count from 65535 to 65536. Closes #8426.
closed 2023-10-09 by dhalbert