adafruit/circuitpython #8429: M5Stack Timer Camera X doesn't boot with idf5.1, but does with 5.0

bill88t, 2023-09-24:
### CircuitPython version

```python
From S3 builds:
adafruit-circuitpython-m5stack_timer_camera_x-en_US-20230923-1c0155c <-- Doesn't boot idf5.1
adafruit-circuitpython-m5stack_timer_camera_x-en_US-20230922-d6b284e <-- Boots just fine idf5.0
```

### Code/REPL

```python
N/A
But wifi auto-connect is enabled without web-workflow.
```

### Behavior

Stuck at:
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:312
load:0x40078000,len:14164
load:0x40080400,len:4
load:0x40080404,len:3268
entry 0x40080580
```

### Description

Making a custom debug build reveals:
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7432
load:0x40078000,len:16576
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:4284
entry 0x40080668
I (31) boot: ESP-IDF fb07ab7f00 2nd stage bootloader
I (31) boot: compile time Sep 24 2023 21:14:24
I (31) boot: Multicore bootloader
I (36) boot: chip revision: v3.0
I (39) qio_mode: Enabling default flash chip QIO
I (45) boot.esp32: SPI Speed      : 80MHz
I (49) boot.esp32: SPI Mode       : QIO
I (54) boot.esp32: SPI Flash Size : 4MB
I (58) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (82) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (90) boot:  2 ota_0            OTA app          00 10 00010000 00160000
I (97) boot:  3 ota_1            OTA app          00 11 00170000 00160000
I (105) boot:  4 user_fs          Unknown data     01 81 002d0000 00130000
I (112) boot: End of partition table
I (117) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=4325ch (275036) map
I (201) esp_image: segment 1: paddr=00053284 vaddr=3ff80000 size=00004h (     4) load
I (201) esp_image: segment 2: paddr=00053290 vaddr=3ffb0000 size=04408h ( 17416) load
I (212) esp_image: segment 3: paddr=000576a0 vaddr=40080000 size=08978h ( 35192) load
I (226) esp_image: segment 4: paddr=00060020 vaddr=400d0020 size=ff40ch (1045516) map
[... 127 lines cut ...]
I (5551) gpio: GPIO[27]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (5561) gpio: GPIO[32]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (5571) gpio: GPIO[34]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (5581) gpio: GPIO[35]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (5591) gpio: GPIO[36]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (5601) gpio: GPIO[37]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (5611) gpio: GPIO[38]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (5621) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (5631) dac_oneshot: dac_oneshot_del_channel(60): input parameter 'handle' is NULL
E (5641) dac_oneshot: dac_oneshot_del_channel(60): input parameter 'handle' is NULL
E (5651) task_wdt: esp_task_wdt_delete(822): TWDT was never initialized
I (5721) wifi:mode : sta (08:b6:1f:87:dd:44)
I (5721) wifi:enable tsf
W (5731) CP wifi: sta start
W (8141) CP wifi: disconnected
W (8141) CP wifi: reason 201 0xc9
E (8141) wifi:NAN WiFi stop
W (8151) CP wifi: sta stop
I (8151) wifi:flush txq
I (8151) wifi:stop sw txq
I (8161) wifi:lmac stop hw txq
```
it goes as far as boot.py, but gets stuck during reload. It's a hang, not a hardfault.
This is weird as in the regular build it doesn't print the boot.py text..
Copying the whole filesystem onto an esp32s3 does not make the issue appear.

tannewt, 2023-09-25:
Do you have an adafruit ESP32 board that this breaks on? I don't have any M5Stack boards.

bill88t, 2023-09-25:
I do not have any unfortunately. Or any other ESP32 boards. I only have S2's and S3's.

Is the issue not reproducible on another esp32 board?

If that is the case, I can debug the definition.
It could be espcamera or the psram or something.

tannewt, 2023-09-25:
> Is the issue not reproducible on another esp32 board?

I did just try it on an ESP32 qtpy and had the same issue. It looks like its due to the analogbufio module not being updated for the new adc api.

PR #8435 (2023-09-27): Update ADC and I2S APIs for 5.1
This fixes ESP32 because the BufferedIn used the old ADC API and I2S did too indirectly.

Tested on QT Py ESP32 Pico and Espressif S3 Devkit C N32R8.

Fixes #8429
commit 3fdfb9bd (2023-09-27): Update ADC and I2S APIs for 5.1

This fixes ESP32 because the BufferedIn used the old ADC API and
I2S did too indirectly.

Fixes #8429
closed 2023-09-27 by tannewt