adafruit/circuitpython #8431: Documentation issue

thetazero, 2023-09-25:
### CircuitPython version

```python
n/a
```

### Code/REPL

```python
Just the docs here: https://docs.circuitpython.org/en/latest/shared-bindings/ulab/numpy/linalg/index.html#ulab.numpy.linalg.eig
```

### Behavior

[image]

It says it returns `(eigenvectors, eigenvalues)` while it actually returns `(eigenvalues, eigenvectors)`.

Actual behavior is inline with numpy.

### Additional information

Happy to fix this myself, just need to be pointed in the right direction.

dhalbert, 2023-09-25:
`ulab` is a submodule: the original repo is here: https://github.com/v923z/micropython-ulab/. So submit issues and PR's to that repo.

closed 2023-09-25 by dhalbert

thetazero, 2023-09-27:
This issue appears on the circuit python docs, not on the micropython ulab docs.

jepler, 2023-09-27:
If they've fixed it already, then we will get the fix next time we update ulab. This is likely to happen soon.

dhalbert, 2023-09-27:
@thetazero Could you link to the docs where the order is correct?