adafruit/circuitpython #8437: Re-enable parallelbus on ESP

tannewt, 2023-09-27:
The I2S LCD API went away with 5.x. Not sure what the best approach going forwards is. (Also not sure we use this anywhere.)

michael-ring, 2023-10-14:
Some esp32s3 boards like lilygo t-display-s3 come with a display connected via 8080 interface, so having parallelbus back on ESP would be nice....

tannewt, 2023-10-16:
Do you know of any IDF 5 example code for these boards? What does it use in the IDF to output to it?

michael-ring, 2023-10-17:
Here's the overall documentation:
https://docs.espressif.com/projects/esp-idf/en/release-v5.0/esp32/api-reference/peripherals/lcd.html
https://docs.espressif.com/projects/esp-idf/en/v5.1.1/esp32/api-reference/peripherals/lcd.html

https://github.com/espressif/esp-idf/blob/master/components/esp_lcd/src/esp_lcd_panel_io_i80.c

In the last example I did not see any mention of i2s, there are other examples in the directiory that are i2s specific.

michael-ring, 2023-10-17:
LilyGo uses eSPI library for their board, unfortunately they only provide Arduino samples:

https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/lib/TFT_eSPI

tannewt, 2023-10-17:
Thanks for the IDF links. It looks like the right way to go. I was assuming esp_lcd had an onboard framebuffer but it looks like it may not for parallel displays. (That's what we expect.)

rgrizzell, 2024-01-07:
> Some esp32s3 boards like lilygo t-display-s3 come with a display connected via 8080 interface, so having parallelbus back on ESP would be nice....

I started working on a port for the LILYGO T-Display S3 and ended up here. Its feasible to get the display working inside the build. However, my cursory look leads me to believe the new ESP drivers are quite different from the old ones. I don't have the smarts to try and tackle it myself, but I'd be more than happy to test any feature branches with the hardware I have.

PR #8862 (2024-02-08): Re-enable paralleldisplaybus on ESP
Add lilygo t-display s3 because it has a parallel display.

Fixes #8437