adafruit/circuitpython #8444: Matrix Portal S3 crashes into safe mode after saving code file that uses UART (core code crash)

WoutStandaert, 2023-09-29:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.6 on 2023-09-12; Adafruit MatrixPortal S3 with ESP32S3
```

### Code/REPL

```python
import board
import displayio
import framebufferio
import rgbmatrix
import busio
import digitalio

uart = busio.UART(board.TX, board.RX, baudrate=115200)
uart.timeout = 0

print("Initialized UART")

displayio.release_displays()
matrix = rgbmatrix.RGBMatrix(
    width=64, bit_depth=4,
    rgb_pins=[
        board.MTX_R1,
        board.MTX_G1,
        board.MTX_B1,
        board.MTX_R2,
        board.MTX_G2,
        board.MTX_B2
    ],
    addr_pins=[
        board.MTX_ADDRA,
        board.MTX_ADDRB,
        board.MTX_ADDRC,
        board.MTX_ADDRD
    ],
    clock_pin=board.MTX_CLK,
    latch_pin=board.MTX_LAT,
    output_enable_pin=board.MTX_OE
)
display = framebufferio.FramebufferDisplay(matrix)

print("Initialized Display")

led = digitalio.DigitalInOut(board.LED)
led.direction = digitalio.Direction.OUTPUT

print("Initialized LED")

# Setup the file as the bitmap data source
bitmap = displayio.OnDiskBitmap("/logo-sprites.bmp")

# Create a TileGrid to hold the bitmap
tile_grid = displayio.TileGrid(bitmap, pixel_shader=bitmap.pixel_shader,
                                width = 1,
                                height = 1,
                                tile_width = 32,
                                tile_height = 32)

# Create a Group to hold the TileGrid
group = displayio.Group()

# Add the TileGrid to the Group
group.append(tile_grid)

# Add the Group to the Display
display.show(group)

group.x = 16

# Loop forever so you can enjoy your image
source_index = 0
while True:
    if uart.in_waiting > 0:
        data = uart.readline()
        if data is not None:        
            print(data)

            if (len(data) > 1):         
                boardid = data[0]
                imageid = data[1]
                
                if boardid == 0:
                    if imageid < 3:
                        tile_grid[0] = imageid
```

### Behavior

Upon saving the code file (which is called main.py) in a text editor, the device will crash and reboot into safe mode with the following message:

```
You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.
```

Sometimes, this means that the code file wasn't saved completely (according to Visual Studio Code).

When resetting the device, the code runs without problems, but it won't run forever and will crash after 30min - 1h. I haven't yet checked the serial monitor when that happens to see what the reason is.

### Additional information

I tried using time.sleep(1) before initializing the UART but it still crashes.

WoutStandaert, 2023-09-29:
Possibly related to #8294

DJDevon3, 2023-09-29:
@WoutStandaert does it happen every file save?  #8294 is intermittent hard fault on save.  When it doesn't hard fault the program runs normally fine.  I'd say every 4th-5th file save it hard faults on save.  If it hard faults on every save for you using UART that might be different because I only tested the RGB matrix capability.

WoutStandaert, 2023-09-29:
@DJDevon3 At first there was no issue, as long as I stayed away from UART. Only when I added UART it crashed every time yesterday on save. I had to do a power cycle every time because the reset button wouldn't get out of safe mode either. Today it seems every one in two times. Not sure what's different today. The crash is sometimes so severe that I have to drag the UF file back on the disk because the drive is showing up as matrixboot.

tannewt, 2023-10-23:
@WoutStandaert Please try 8.2.7 and see if the issue is fixed. Thanks!

WoutStandaert, 2023-10-24:
Hi @tannewt, thank you! 

I had to erase the flash memory via the [factory reset](https://learn.adafruit.com/adafruit-matrixportal-s3/factory-reset#factory-reset-and-bootloader-repair-3107941). Just dragging the new 8.2.7 UF2 file onto the matrixBoot drive didn't work. 

But now that I've reflashed the whole thing, I was able to save the code file like 10 times in a row without problems. 

Then, I still made it crash by connecting it to a UART source while saving. So the RX pin might have been receiving data while I was saving the code file, and that still made it crash, thereby possibly corrupting memory somewhere? Because since this crash, even if everything is disconnected from the MatrixPortal, I can make it crash again by just saving the code file. Windows also complains that "this disk contains errors".

Should I do something else to factory reset it? [There's a .bin file I can download here](https://circuitpython.org/board/adafruit_matrixportal_s3/), but I'm not sure if I should flash it, maybe it's the same as the factory reset .bin from the factory reset procedure?

tannewt, 2023-10-24:
Sounds like the issue is still occurring. We'll need to look into it once we've transitioned to bug fixing on 9.0.0.

dhalbert, 2023-11-16:
@WoutStandaert So this is on Windows? Which version? What is the text editor you are using?

Could you say more about the data coming in over the UART port? Is it continuously being sent from the source? If the UART is opened, but you disconnect the data source, do you still see the problem? I am asking so we can try to set up a similar test situation.

If you could also test with 9.0.0-alpha.5, that would be great. Thanks. That updates the underlying Espressif software by a major version.

WoutStandaert, 2023-11-23:
@dhalbert 

The Windows version is 10 22H2, I'm using Visual Studio Code with [joedevivo's CircuitPython extension](https://github.com/joedevivo/vscode-circuitpython)

The data coming in is coming in every second, and is something like this: 
`byte buf[] = {0, 0, 0x0A};`

I think the problem is related to the `data = uart.readline()` line. When I run the code on a MatrixPortal M4 without a UART source attached, eventually the code will crash on that line (but not hard fault). 

So maybe it's receiving some kind of noise on the UART port, enough to trigger `if uart.in_waiting > 0:` - and then it times out on readline() because it never receives a newline byte.

And maybe, while it's buffering the incoming serial data waiting for the newline, it starts overwriting memory it shouldn't access.

From my other tests in the past, it would still hard fault after 30 - 60 minutes, even with a UART source attached. And now, just starting up VS Code also made it hard fault, but it might be because something got corrupted after all these crashes.

I'll update to 9.0.0 a5, see if that makes a difference.

WoutStandaert, 2023-11-23:
@dhalbert I've updated to 9.0.0-alpha.5 and... so far so good! I haven't been able to make it hard fault yet and it's been running for 3 hours without crashing.

WoutStandaert, 2023-11-24:
@dhalbert After running through the night without problems, I think this is solved by the update to version 9.

closed 2023-11-24 by WoutStandaert