adafruit/circuitpython #8448: I2C I/O Error on Teensy 4.1

jksjaz, 2023-09-30:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.3 on 2023-08-11; Teensy 4.1 with IMXRT1062DVJ6A
```

### Code/REPL

```python
import adafruit_bno055
import busio

# Initialize the IMU imu_sensor
imu_i2c = busio.I2C(board.SCL, board.SDA)
imu_sensor = adafruit_bno055.BNO055_I2C(imu_i2c)

last_val = 0xFFFF

def temperature():
    global last_val  # pylint: disable=global-statement
    result = imu_sensor.temperature
    if abs(result - last_val) == 128:
        result = imu_sensor.temperature
        if abs(result - last_val) == 128:
            return 0b00111111 & result
    last_val = result
    return result

while True:
    print("Temperature: {} degrees C".format(imu_sensor.temperature))
    """
    print(
        "Temperature: {} degrees C".format(temperature())
    )  # Uncomment if using a Raspberry Pi
    """
    print("Accelerometer (m/s^2): {}".format(imu_sensor.acceleration))
    print("Magnetometer (microteslas): {}".format(imu_sensor.magnetic))
    print("Gyroscope (rad/sec): {}".format(imu_sensor.gyro))
    print("Euler angle: {}".format(imu_sensor.euler))
    print("Quaternion: {}".format(imu_sensor.quaternion))
    print("Linear acceleration (m/s^2): {}".format(imu_sensor.linear_acceleration))
    print("Gravity (m/s^2): {}".format(imu_sensor.gravity))
    print()

    time.sleep(1)
```

### Behavior

```python
code.py output:
Temperature: 0 degrees C
Accelerometer (m/s^2): (0.0, 0.0, 0.0)
Magnetometer (microteslas): (10.0625, 13.75, -39.375)
Gyroscope (rad/sec): (-0.00109083, 0.00218166, -0.00218166)
Euler angle: (0.0, 0.0, 0.0)
Quaternion: (0.0, 0.0, 0.0, 0.0)
Linear acceleration (m/s^2): (0.0, 0.0, 0.0)
Gravity (m/s^2): (0.0, 0.0, 0.0)

Temperature: 28 degrees C
Traceback (most recent call last):
  File "code.py", line 291, in <module>
  File "adafruit_bno055.py", line 416, in acceleration
  File "adafruit_bno055.py", line 155, in __get__
  File "adafruit_register/i2c_struct.py", line 49, in __get__
OSError: [Errno 5] Input/output error
```

### Description

While testing the Adafruit BNO055 sensor connected using STEMMA QT, I encountered the following error which seems to be an issue with the I2C protocol.

I've submitted the same issue in the specific library as well but no response there so posting here as well. https://github.com/adafruit/Adafruit_CircuitPython_BNO055/issues/118

jksjaz, 2023-10-10:
After some debugging, it would seem the I2C frequency is not compatible with Teensy is why it's failing. After changing it, I was able to make the sensor work:

`import time
import board
import adafruit_bno055
import busio

i2c = busio.I2C(board.SCL, board.SDA, frequency=400000)
sensor = adafruit_bno055.BNO055_I2C(i2c)

last_val = 0xFFFF

def temperature():
    global last_val  # pylint: disable=global-statement
    result = sensor.temperature
    if abs(result - last_val) == 128:
        result = sensor.temperature
        if abs(result - last_val) == 128:
            return 0b00111111 & result
    last_val = result
    return result

while True:
    try:
        print("Temperature: {} degrees C".format(sensor.temperature))
        """
        print(
            "Temperature: {} degrees C".format(temperature())
        )  # Uncomment if using a Raspberry Pi
        """
        print("Accelerometer (m/s^2): {}".format(sensor.acceleration))
        print("Magnetometer (microteslas): {}".format(sensor.magnetic))
        print("Gyroscope (rad/sec): {}".format(sensor.gyro))
        print("Euler angle: {}".format(sensor.euler))
        print("Quaternion: {}".format(sensor.quaternion))
        print("Linear acceleration (m/s^2): {}".format(sensor.linear_acceleration))
        print("Gravity (m/s^2): {}".format(sensor.gravity))
        print()
    except:
        print("Error")

    time.sleep(1)
`

It still does error out randomly sometimes however it's working fine for about 95% of the time which is enough for most applications.

dhalbert, 2025-02-01:
Noted as a troublesome chip on i.MX here: https://learn.adafruit.com/i2c-addresses/troublesome-chips

closed 2025-02-01 by dhalbert