adafruit/circuitpython #8449: OSError: [Errno 5] Input/output error fresh instalation of MacOs Sonoma

suarezluis, 2023-10-01:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.6 on 2023-09-12; Adafruit CircuitPlayground Express with samd21g18
```

### Code/REPL

```python
# Write your code here :-)

print("Hello World!")
```

### Behavior

```
Code stopped by auto-reload. Reloading soon.
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
OSError: [Errno 5] Input/output error

Code done running.
```

### Description

- Error on each save

### Additional information

On a fresh installation of MacOS sonoma every time that code.py is saved a OSError: [Errno 5] Input/output error pops up, eventually the code runs but it takes about 15 seconds, I have tried disabling USB Stores and spotlight with:

```
defaults write DSDontWriteUSBStores -bool false
sudo mdutil -i off -d /Volumes/CIRCUITPY/ 
```

No luck

anecdata, 2023-10-01:
The operating system shouldn't matter for code running on the board.

What IDE or editor are you using? Is there a `safemode.py` file or a `boot.py` file?

choas, 2023-10-01:
Visual Studio Code shows an error and the main.py file is empty.
Using the macOS TextEdit program works for me.

suarezluis, 2023-10-02:
Ive tried, VS Code and Mu, it seems like the OS is delaying the write of the file or writing to it more than once and one of those writes gets the error 🤷‍♂️ 

I could disable auto-reload but that is not the same experience when using Windows or Linux

suarezluis, 2023-10-02:
It seems like the problem was discovered on Sonoma beta not long ago:

https://discord.com/channels/327254708534116352/537365702651150357/1158115199086247946

anecdata, 2023-10-02:
How does a CIRCUITPY drive present differently than an SD card or flash drive?

eightycc, 2023-10-02:
Former Core OS engineer here. Could you get a trace of Mac initiated I/O to the SD card leading up to the I/O error? Is this an Intel or Apple Silicon machine?

eightycc, 2023-10-02:
Apple maintains an OSS version of MacOS called Darwin. Corresponding to the release of Sonoma, there is an update to `msdosfs` to version 608.0.3, the project can be found here [msdosfs](https://github.com/apple-oss-distributions/msdosfs). The source file of interest is [msdosfs_fat.c](https://github.com/apple-oss-distributions/msdosfs/blob/main/msdosfs.kextproj/msdosfs.kmodproj/msdosfs_fat.c).

My best guess is that 608.0.3 is initiating a write outside of the CP filesystem's flash extent. This could be due to a regression in Apple's `msdosfs` or due to a latent bug in CP. Hoping it's the latter or that if it's the former it can be worked around in CP.

dhalbert, 2023-10-02:
> How does a CIRCUITPY drive present differently than a (small) SD card or flash drive, or are those broken too for equivalent FAT12 or FAT16 devices?

It doesn't necessarily. But FAT12, for instance, is not supported on Android. And those may well be broken in the Sonoma  beta

eightycc, 2023-10-02:
I've got a Circuit Playground Express handy. Just need to update a machine to Sonoma and I should be able to reproduce this bug and get some diagnostic data.

tannewt, 2023-10-02:
My guess is delayed write. I see this occasionally with Linux too. I run `sync` from the command line and then the next reload works as expected.

eightycc, 2023-10-02:
> My guess is delayed write.

We will see! Sonoma just finished installing, I've got Playground Express REPL on UART, and I'm dropping some diagnostic prints in now.

eightycc, 2023-10-02:
I don't always get the I/O error, but I always see two consecutive soft reboots whenever I save `code.py`, so it looks like @tannewt 's guess is correct. Sonoma is writing after a delay.

eightycc, 2023-10-02:
The delay is fairly long, on the order of 10 to 20 seconds. I can see one shot of writes when I save the file, the delay, then the remaining writes:
```
flash_write_blocks: block_num=42, num_blocks=1
flash_write_blocks: block_num=43, num_blocks=1
***delay 10 to 20 seconds***
flash_write_blocks: block_num=2, num_blocks=1
flash_write_blocks: block_num=3, num_blocks=1
flash_write_blocks: block_num=4, num_blocks=1
flash_write_blocks: block_num=5, num_blocks=1
flash_write_blocks: block_num=6, num_blocks=1
flash_write_blocks: block_num=7, num_blocks=1
flash_write_blocks: block_num=8, num_blocks=1
flash_write_blocks: block_num=9, num_blocks=1
```
Unmounting then remounting the `CIRCUITPY` filesystem like this:
```
sudo umount /Volumes/CIRCUITPY
sudo mkdir /Volumes/CIRCUITPY
sudo mount -o noasync -t msdos /dev/disk4s1 /Volumes/CIRCUITPY
```
temporarily eliminates the delay. The special device name will vary. There should be a way to make this change permanent in Sonoma's `/etc/fstab`, but so far I'm not finding a `LABEL=` solution works.

dhalbert, 2023-10-03:
This behavior makes macOS more like Linux and Windows. It would be interesting to see if there is also a delay with a regular USB stick, that is, if you do a write and then pull the stick within a few seconds, does it lose the write and/or corrupt the file system?

eightycc, 2023-10-03:
The workaround above is far from ideal as it needs to be repeated each time the device is plugged in to the Mac's USB port or for any other operation that causes MacOS to remount the `CIRCUITPY` filesystem. `/etc/fstab` is a dead end, Sonoma appears to ignore it for at least USB filesystem mounts. There's a huge amount of variability in the delay, I've seen it as low a 3 seconds and as high as 25 seconds, so increasing `CIRCUITPY_AUTORELOAD_DELAY_MS` to 30000 mS is probably not a good option.

Aside from disabling `autoreload` in CP, I'm out of ideas.

Clearly this is a regression in MacOS Sonoma that Apple needs to fix. To get the attention of their triage mechanism, it's best for everyone that's affected by this to file an issue against the same components with similar descriptions and details. I'm filing with the Feedback Assistant app like this:

Descriptive title: Long, unpredictable delay between writes when writing to USB removable media
Which area are you seeing an issue with: USB/Thunderbolt
What type of issue are you reporting: Incorrect/Unexpected Behavior
Is the device still in the problem state right now: No
What time was it when this last occurred: 19:57 pm PDT 10/02/2023

Description:
When updating a file on an MSDOS filesystem residing on a USB device there is a long delay (3 to 25 seconds) between the first and final writes to the filesystem.

The filesystem is automatically mounted with the async default. Some delay on the order of 10's of milliseconds is expected once writing begins. 3 to 25 seconds is not expected.

eightycc, 2023-10-03:
> This behavior makes macOS more like Linux and Windows. It would be interesting to see if there is also a delay with a regular USB stick, that is, if you do a write and then pull the stick within a few seconds, does it lose the write and/or corrupt the file system?

I'd expect it to behave as you describe, it's getting late so I'll write this up and try the stick in the morning.  The issue for Sonoma is not that the writes are delayed, it's that once the writes start it's reasonable to expect them to finish without a big delay in the middle. What I'm seeing is:

1. Saving `code.py` kicks off a couple of writes to the filesystem to update the file.
2. Those writes complete, triggering an autoreload
3. The autoreload counts down 500 mS and starts the reload
4. After an unpredictable delay, the MacOS kicks off more writes updating the filesystem and writing a metadata file `.code.py`. If these writes happen while autoreload is in progress an I/O error results. Otherwise, if the autoreload has finished, the remaining writes trigger another autoreload.

At no time does the OS see the filesystem removed as would happen with a pulled USB stick.

It's interesting that you mention similar behavior on Linux and Windows. On Linux I've noticed the double and triple autoreload, so something similar could be in play.

dhalbert, 2023-10-03:
The writes can be both data writes to the individual files, and also filesystem metadata updates to the FAT. On Windows, the delay was considerably worse on FAT12 than FAT16 and larger. It appeared to be a relic of floppy disk support, to avoid spinning up the disk or wearing it out with frequent writes. See https://github.com/adafruit/circuitpython/issues/111. Supposedly this was fixed at some point in Windows 10, but I haven't had a chance to verify. On Linux, there is a delay for caching reasons. on macOS before Sonoma, the writes happened immediately.

Writes can be forced to occur immediately by doing an "Eject" on Windows and a `sync` on Linux.  It would be interesting to try a `sync` from a Terminal on Sonoma to see if it forces the writes out.

eightycc, 2023-10-03:
@dhalbert That's interesting. Casting about the web, I've found problems reported on versions of MacOS prior to Sonoma. Camera SD cards were getting corrupted, the root cause being removal of the card without dismounting. The posters were looking for a solution that would force `noasync` when the card automounts, but could never find one. In an earlier post to this thread I tried unmounting the Express and then remounting it with `noasync` specified. The delay between groups of writes went away, and all writes occurred immediately, that is within a few milliseconds of each other.

eightycc, 2023-10-03:
@dhalbert Ran the experiment pulling a USB stick after a write:

1. Created a 32M FAT12 filesystem on a USB stick
2. Wrote the canonical `code.py` to the stick and verified
3. Mounted the stick on Sonoma and opened it as a folder with VSCode
4. Edited a minor change into `code.py`
5. Saved `code.py`
6. Counted one potato, two potatoes and pulled the stick

Wound up with a zero length `code.py` and without the expected `.code.py` metadata file.

dhalbert, 2023-10-03:
@eightycc Thanks for testing. If you have a machine that's still running Ventura, it would be interesting to compare. Or maybe someone else can repeat the experiment. I have some but won't have access to them for a while.

eightycc, 2023-10-03:
@dahlbert Under the covers what's going on is that MacOS copy-on-reads blocks of a file into virtual memory and backs that memory with an unallocated area of the underlying filesystem. Writes become stores into virtual memory, triggering immediate page-outs to the filesystem. Because the filesystem mount flags include `MNT_ASYNC`, updates to the FAT and the metadata file are deferred until a timeout mechanism forces a flush. This is why pulling the stick results in a zero length file, the file data is there but the FAT update to point to it hasn't happened yet.

This got me thinking that if `autoreload` were to trigger selectively on just writes to the FAT (probably just two on such a small file system) this would eliminate the I/O error at the expense of a delayed `autoreload`.

> If you have a machine that's still running Ventura, it would be interesting to compare.
I've got an older Monterey (10.12) machine I can test and trace.

dhalbert, 2023-10-03:
> This got me thinking that if `autoreload` were to trigger selectively on just writes to the FAT (probably just two on such a small file system) this would eliminate the I/O error at the expense of a delayed `autoreload`.

That's a good idea, and we've thought about that in the past. But weren't sure that the FAT is the last thing written on all operating systems. The writes on Windows seemed to be a in random order sometimes. Also, writing data files that are in use could cause the program to read bad data or cause program errors, though that's not as common as updating the program files.

Aside: since MSC is block-oriented, not file-oriented, it's very difficult to tell what's going on and when writes are done. MTP would have been a much more satisfactory solution, but Apple has never supported it, I assume due to iPod marketing decisions.

eightycc, 2023-10-03:
> Also, writing data files that are in use could cause the program to read bad data or cause program errors, though that's not as common as updating the program files.

Yes, that scenario is going to fail hard with MacOS, and probably with Linux and Windows too! Sharing a filesystem has lots of pitfalls, and this is clearly one of them. The producer and consumer do need some sort of locking, signaling and synchronizing mechanism to make sharing through the filesystem safe.

choas, 2023-10-04:
copying the code.py directly in a Terminal causes this error:
`cp code.py /Volumes/AL1EN_L/code.py`
cp: /Volumes/AL1EN_L/code.py: Input/output error

anecdata, 2023-10-05:
On a FAT16 CircuitPython device (UM FeatherS3 with 8.2.6), code saves had a long delay before auto-reload, and in one instance had `OSError: [Errno 5] Input/output error`. So it doesn't look like only a FAT12 issue.

dhalbert, 2023-10-09:
Reported in forums: https://forums.adafruit.com/viewtopic.php?t=205044

czei, 2023-10-09:
> The operating system shouldn't matter for code running on the board. But saving a file triggers auto-reload. You can turn off auto-reload with `supervisor.runtime.autoreload = False` if that's more convenient.
> 
> `OSError: [Errno 5]` is unexpected.
> 
> What IDE or editor are you using? Is there a `safemode.py` file or a `boot.py` file?
> 
> _Update:_ other(s) are seeing this issue too

Sorry for the basic question, but where does the supervisor.runtime.autoreload = False go?  I'm new to Python and the MatrixPortal board.  

I tried the REPL, but that didn't work.   Then I tried putting the line into code.py, and got an error "NameError: name 'supervisotr' is not defined.   

Thanks,

suarezluis, 2023-10-09:
> > The operating system shouldn't matter for code running on the board. But saving a file triggers auto-reload. You can turn off auto-reload with `supervisor.runtime.autoreload = False` if that's more convenient.
> > `OSError: [Errno 5]` is unexpected.
> > What IDE or editor are you using? Is there a `safemode.py` file or a `boot.py` file?
> > _Update:_ other(s) are seeing this issue too
> 
> Sorry for the basic question, but where does the supervisor.runtime.autoreload = False go? I'm new to Python and the MatrixPortal board.
> 
> I tried the REPL, but that didn't work. Then I tried putting the line into code.py, and got an error "NameError: name 'supervisor' is not defined.
> 
> Thanks,

Hi @czei , you can do that in the REPL, you just need to do:
`import supervisor
supervisor.runtime.autoreload = False`

if you want to do it in code.py you need to import supervisor as well but it will run every time with your code in code.py

anecdata, 2023-10-09:
or do it in `boot.py` to close off that interval when starting to run `code.py`, where an autoreload could still happen

hopkapi, 2023-10-10:
The awkward moment when you [experience this in July](https://discord.com/channels/327254708534116352/327298996332658690/1139718759205044224) and ask about it in Discord, but forget to file an actual issue, and then Sonoma ships..

Can I suggest that in the meantime, relevant CP/Adafruit guides get updated with a little note about Sonoma issues with a link to a page noting the workarounds? I don't know what the % of CP users on macOS are, but I'd assume it's more than a handful of us, and Sonoma has been out for a couple of weeks now.

czei, 2023-10-10:
Thanks for the suggestions @hopkapi and @anecdata, that's a big help until this bug is fixed.

dhalbert, 2023-10-12:
I think this report in the Raspberry Pi forums is about the same issue: https://forums.raspberrypi.com/viewtopic.php?t=357468.

dhalbert, 2023-10-12:
Some workarounds:
- simply waiting until the write completes: code.py will restart again after the final write. [not tested by me yet]
- Turning off autoreload: `supervisor.runtime.autoreload = False`.
- Doing `sync` in a Terminal window
- Adding an `os.sync()` in the right place in whatever editor is being used.
- Downgrade back to Ventura. I don't know how easy or possible this is. 

I will not have an opportunity to try this on a Sonoma Mac for a few days, but will try to characterize the issue more exactly then by doing some USB monitoring.

todbot, 2023-10-13:
Thanks for the summary, @dhalbert. I submitted a detailed Feedback to Apple (and I recommend everyone do that same, as mentioned upthread).  My work-around at the moment is to run this very bad idea in a separate terminal window:
```sh
while [ 1 ] ; do ; echo "sync" ; sync ; sleep 5; done
```

Pharkie, 2023-10-25:
I can't get copying files to a Pico CIRCUITPY from a Mac (M1 Macbook Pro running Sonoma) to reliably work with all/any of the workarounds i.e. I've added to boot.py
```
import supervisor
supervisor.runtime.autoreload = False
```
I've done `sync` in Terminal. 
From a newly Flashed-to-CIRCUITPY drive on my Pico, I can copy over the small .py files (boot.py, code.py) quite reliably, but it almost always fails with the (small) MP3s including the ones from the Adafruit sample project (slow.mp3). I can never get more than one MP3 on before Error -36. Once it gives that, that's it, no amount of reconnecting or resetting changes it, it just refuses to write the files.
I switched to my PI400 and everything works just fine, so easy. 

Hope this can get fixed soon - that means waiting on Apple? And if anyone has further tips/ideas for workarounds, please let me know.

🙏

czei, 2023-10-25:
I have had good luck with this shell script that remounts the USB drive on the MatrixPort S3.

#!/bin/csh
#
# This works around bug where, by default, macOS 14.0 writes part of a file 
# immediately, and then doesn't update the directory for 20-60 seconds, causing
# the file system to be corrupted.
#

set disky=`df | grep CIRCUITPY | cut -d" " -f1`
sudo umount /Volumes/CIRCUITPY
sudo mkdir /Volumes/CIRCUITPY
sleep 2
sudo mount -v -o noasync -t msdos $disky /Volumes/CIRCUITPY

Pharkie, 2023-10-26:
That does seem to fix it. After running your script, I can now copy over files on Mac Sonoma reliably just like the Pi400. Thank you! 🤩

dhalbert, 2023-10-27:
The problem seems unrelated to the kind of FAT filesystem (FAT12, FAT16, etc.). Maybe of some interest, the `mount` flags differ:

For both a CIRCUITPY FAT12 drive and a regular several-GB USB stick, the mount flags are:
**Ventura:** `msdos, local, nodev, nosuid, noowners`
**Sonoma:** `msdos, local, nodev, nosuid, noowners, noatime`

eightycc, 2023-10-27:
@dhalbert Manually remounting clears the `nodev`, `nosuid`, and `noatime` flags. Manually mounting with or without the `noasync` option works around the problem. Likewise, manually mounting with various permutations of these four flags always works around the problem, including the case where no mount flags are provided. Looks like the mount flags are a red herring, it's some other difference between an automatic vs. a manual mount.

Testing on Sonoma 14.1.

romkey, 2023-10-27:
Apologies if this has been covered already; I skimmed and it looks like most of the focus has been on async and auto-reload.

The problem also happens with the UF2 boot loader drive - for instance, trying to update CircuitPython firmware by dragging its .uf2 file to the boot loader drive under Sonoma fails.

Given that, it also unsurprisingly happens with non-CircuitPython devices - I couldn't install Bluetooth sniffer firmware on a Nordic nRF dongle on a Sonoma Mac. I was able to using Linux.

More surprisingly I also found that I can't write files to a FAT12 regular old USB stick. I prepared a small USB stick with a 4MB FAT12 filesystem under Linux. I was able to read and write files there. Trying to use it on a Sonoma Mac got the same errors as trying to use a UF2 or CircuitPython drive, both from the Finder and the command line (with a fresh filesystem between attempts). Just writing a bunch of Python source code to the drive as one would with CP.

4MB FAT12 mounted `noasync` worked.

Sonoma was able to write to a 64MB FAT12 USB drive though.

So it also appears to also be related to the size of the filesystem.

So summary:
- other non-CircuitPython devices with UF2 filesystems also fail
- Sonoma failed writing small files to a 4MB FAT12 filesystem on a plain USB flash drive (both command line and Finder)
- Sonoma succeeded writing small files to a 4MB FAT12 filesystem mounted with the `noasync` option
- Sonoma succeeded writing small files to a 64MB FAT12 filesystem on a plain USB flash drive without `noasync`

dhalbert, 2023-10-27:
@romkey note that @eightycc's testing seems to show that any manual remount with any flags seems to clear the problem. Your last two tests probably included a manual remount, so maybe that explains what you are seeing.

eightycc, 2023-10-27:
@romkey You _should_ be able to work around _any_ of the FAT filesystem write problems by unmounting, making a mount point, and then re-mounting the device. No need to specify mount option flags. Give it a try and let me know if you run into a case where that doesn't work.

romkey, 2023-10-27:
The first public beta of macOS 14.2 still has the issue.

romkey, 2023-10-27:
@dhalbert Good point! I just re-ran the test:

- prepped a clean 64MB filesystem on a USB flash drive (using `dd` from a filesystem image prepared under Linux)
- unplugged the drive
- rebooted the Mac (just to be absolutely sure)
- plugged the drive back in and let macOS mount it
- confirmed it was empty
- copied a batch of .py files that cause the error on a 4MB filesystem
- successful copy, no errors

```
/dev/disk6 on /Volumes/NO NAME (msdos, local, nodev, nosuid, noowners, noatime)
```

So it still looks to be dependent on the size of the filesystem.

romkey, 2023-10-27:
@eightycc  Just tested this:

- prepped a clean 4MB filesystem on a USB flash drive (using dd from a filesystem image prepared under Linux)
- unplugged the drive
- plugged the drive back in
- confirmed it was empty
-  `mkdir /Volumes/NO\ NAME`
- `mount -t msdos -o nodev,nosuid,noowners,noatime /dev/disk6 /Volumes/NO\ NAME`
- copied a batch of .py files that normally cause the error on a 4MB filesystem
- as you predicted, this actually worked and they copied successfully

```
/dev/disk6 on /Volumes/NO NAME (msdos, local, nodev, nosuid, noowners, noatime)
```

eightycc, 2023-10-27:
@romkey It's fascinating and unexpected that filesystem size makes a difference, thank you for running that down. So that I can get trace data, I'm going build a modified CP that reports a 64M device. Also, thanks for the confirmation on the work around.

romkey, 2023-10-27:
If it really is filesystem-size related I can see how that could easily slip through QA and testing; unless you're familiar with this part of the world you would probably not expect that there were 4MB USB keys in use these days.

romkey, 2023-10-27:
@eightycc I tried again with a few other sizes. 8MB fails while 8.5MB, 10MB, 12MB and 16MB succeed (worked backwards on this, I'm not completely erratic).

Just to be sure I rebooted and 8.5MB still succeeded.

I tried 8MB three times with fresh filesystems and it failed each time.

I'm going to take a wild guess that 8MB is the breakpoint.

eightycc, 2023-10-27:
@romkey That's super helpful. Thank you!

dhalbert, 2023-10-27:
- #2751 has some discussion about faking a FAT16 filesystem.

czei, 2023-10-27:
Thank you @eightycc and @romkey for your diligent investigation.  That should make the fix much easier for the macOS team at Apple.

eightycc, 2023-10-27:
If anyone on this issue hasn't already done so, file an issue using Apple's Feedback Assistant app. When you do, please mention FB13226668 (the issue I opened) so that Apple ties the reports together and raises the priority of the issue. If you've already opened an issue, you can append a note mentioning my issue number. Thank you!

romkey, 2023-10-28:
Easier way to verify, doesn't even need a USB drive.

Create an 8MB FAT12 disk image. Not sure how to do this on macOS, on Linux it's:
```
truncate -s 8M disk8.img
mkfs.vfat -F 12 disk8.img
```
Transfer this to your Mac.

Mount it - either open Disk Utility in the Finder and use File | Open Disk Image...

or

```
diskutil image attach disk8.img
```

This should mount as `/Volumes/NO NAME` and appear in the sidebar in the Finder.

Copy small files to it from the command line or using Finder. You should see errors.

You can unmount it by ejecting the volume in Finder.

I haven't confirmed if the filesystem is corrupted when errors happen during the copy, so it's best to always start with a fresh disk image to be sure you're not accidentally seeing actual filesystem errors.

eightycc, 2023-10-28:
Because the RP2040 bootloader in ROM presents a 128M FAT16 filesystem, the process of flashing an RP2040 with a new .uf2 image should not fail on MacOS Sonoma. If anyone is experiencing otherwise, please add a note to this issue.

dhalbert, 2023-11-09:
@romkey I tried your disk image test on Sonoma 14.1.1 and could not generate errors. However, I was still seeing delayed writes when using Mu with a Pi Pico. What should I do to provoke errors?

I made the image on Linux as described above, and `scp`'d it over to the Mac
```sh
halbert@Mac-mini ~ % diskutil image attach EIGHT_MB.img 
/dev/disk4		/Volumes/NO NAME
halbert@Mac-mini ~ % cat >foo.txt
abc
halbert@Mac-mini ~ % cp foo.txt /Volumes/NO\ NAME 
halbert@Mac-mini ~ % cp foo.txt /Volumes/NO\ NAME/foob.txt
halbert@Mac-mini ~ % cp foo.txt /Volumes/NO\ NAME/fooc.txt
halbert@Mac-mini ~ % cp foo.txt /Volumes/NO\ NAME/food.txt
halbert@Mac-mini ~ % 
```

romkey, 2023-11-09:
@dhalbert Thanks for trying to confirm this. I'm a lot more comfortable if other people are able to confirm it.

Are you on an Intel or Apple Silicon Mac? I think someone had previously suggested it might only be a problem on Apple Silicon.

I'm using an M2 15" MacBook Air. Still running Sonoma 14.1, I oddly haven't updated yet.

It sounds like we're doing the same thing, I'm not sure what might be different.

I just re-ran the test and am still seeing the problem.

On Ubuntu 22.04.3 LTS I ran

```
truncate -s 8M disk8.img
mkfs.fat -F 12 disk8.img
```
and then on the Mac
```
diskutil image attach disk8.img
cp *.py /Volumes/NO\ NAME
```
to get a slew of errors:
```
p: /Volumes/NO NAME/platform_native.py: fcopyfile failed: Input/output error
cp: /Volumes/NO NAME/platform_native.py: fchmod failed: Input/output error
cp: platform_native.py: could not copy extended attributes to /Volumes/NO NAME/platform_native.py: Input/output error
cp: /Volumes/NO NAME/pollen.py: fcopyfile failed: Input/output error
cp: /Volumes/NO NAME/pollen.py: fchmod failed: Input/output error
cp: pollen.py: could not copy extended attributes to /Volumes/NO NAME/pollen.py: Input/output error
cp: /Volumes/NO NAME/splash.py: fcopyfile failed: Input/output error
```

This is a pretty vanilla Mac, no unusual or extra security software installed.

I'll try updating to 14.1.1 and see if that changes anything. 

Here's a screen cap showing it:

https://github.com/adafruit/circuitpython/assets/58883/ea7fd4d9-3529-4a71-a3e8-a43e4924ab4c

dhalbert, 2023-11-10:
@romkey  I updated my M1 Mac Mini to 14.2 Beta (23C5041e). This is the latest developer beta. I don't get the errors you are seeing when copying files to the disk image. But I do still get errors when copying to an 8MB CIRCUITPY drive. Copying to a 16MB CIRCUITPY drive is ok. I also get err

I double-checked the disk image and it is 8MB and FAT12. I created it with  exactly the same commands you did. Only the name is different.

romkey, 2023-11-10:
@dhalbert I just ran the test again with *exactly* the commands you used, and I don't see the error either.

I tried individual cp commands with longer filenames and didn't get the error. I also tried larger files and manually copying didn't get the error.

I do still see the errors while trying to copy that batch of Python files, though.

It appears to be timing-related. I created a shell script to copy the files one at a time with individual cp commands. It caused the errors. I modified it to put a sleep 1 between the commands and it worked.

So this succeeded:
```
cp aqi.py /Volumes/NO\ NAME
sleep 1
cp clock.py /Volumes/NO\ NAME
sleep 1
cp data.py /Volumes/NO\ NAME
sleep 1
cp esp32spi_http_server.py /Volumes/NO\ NAME
sleep 1
cp give_me_a_sign.py /Volumes/NO\ NAME
sleep 1
cp greet.py /Volumes/NO\ NAME
sleep 1
cp http_server.py /Volumes/NO\ NAME
sleep 1
cp image.py /Volumes/NO\ NAME
sleep 1
cp ip.py /Volumes/NO\ NAME
sleep 1
cp message.py /Volumes/NO\ NAME
sleep 1
...
```

This failed:
```
cp aqi.py /Volumes/NO\ NAME
cp clock.py /Volumes/NO\ NAME
cp data.py /Volumes/NO\ NAME
cp esp32spi_http_server.py /Volumes/NO\ NAME
cp give_me_a_sign.py /Volumes/NO\ NAME
cp greet.py /Volumes/NO\ NAME
cp http_server.py /Volumes/NO\ NAME
cp image.py /Volumes/NO\ NAME
cp ip.py /Volumes/NO\ NAME
cp message.py /Volumes/NO\ NAME
```

Timing-related makes some sense given that `noasync` can bypass it.

As frustrating as it is, this is a pretty cool problem ;) I hope someday we find out exactly what was going on.

If you have a chance would you mind trying that, copying a bunch of files quickly like with a wildcard or in a shell script with no delays between the copies?

I was about to update another machine to the 2nd 14.2 beta that was just released today to test it; if you just updated yours then we have the answer that it's still not fixed :(

dhalbert, 2023-11-10:
> If you have a chance would you mind trying that, copying a bunch of files quickly like with a wildcard or in a shell script with no delays between the copies?

Besides the simple copying of tiny `foo.txt`, I also copied a substantial batch of .py files, and could not get the error. But they are all pretty small files, I think. So I'm not worried we are getting different results. Since it's still reproducible on real boards, unfortunately, we'll just have to keep trying the betas.

dhalbert, 2023-11-10:
@romkey I think your test is valuable to submit to Apple, because it requires external components. If you could reliably reproduce it with a canonical set of files, then the whole thing could be submitted to Apple Feedback. I thought someone gave a Feedback report number to cross-reference, but I can't find it right now

[79 more replies]

closed 2024-02-14 by dhalbert