adafruit/circuitpython #8450: Spread operator like `d = [a, b, *c]`

michalpokusa, 2023-10-01:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.4 on 2023-08-22; Adafruit Feather ESP32-S2 TFT with ESP32S2
Board ID:adafruit_feather_esp32s2_tft
```

### Code/REPL

```python
numbers = [1, 2, 3]
new_numbers = [0, *numbers, 4]
print(new_numbers)
# Expected output: [0, 1, 2, 3, 4]
```

### Behavior

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
SyntaxError: *numbers must be assignment target

### Description

Different behaviour between CircuitPython and Python.

I do not know wheter this is intended.

michalpokusa, 2023-10-01:
Actually now when I thiunk about it, maybe it should be labeled as a feature insted of bug.

todbot, 2023-10-01:
The spread operator does work in {Micro,Circuit}Python but I think only as arguments to functions, not in list constructors, e.g. 
```
numbers = [1, 2, 3]
def print_me(a,b,c): print("hi:", a,b,c)
print(numbers)
```

jepler, 2023-10-01:
I marked this as "enhancement" as well as "third-party". If Micropython ever implements this Python language feature we'd get it with some future merge; otherwise, it's not likely to be prioritized by adafruit folks.