adafruit/circuitpython #8452: qrio: add an option to also know the positions of the qr-codes found

deshipu, 2023-10-01:
Looking at the quirc library that is used internally by qrio, I found that the library also returns the positions of all four corners of the QR codes it found, but that information is not passed on to the Python object returned.

I wonder if there would be a simple way of somehow adding an option to get that information. That would allow projects that have a moving camera to center on the QR code, and even to calculate a rough position relative to the QR code, similar to how one can do it with April Tags fiducials.

I know this is a bit advanced, but since the underlying library already does it, it would be nice to expose it somehow, perhaps with an additional method on the decoder, or a flag that makes the decoder return extra information?

I'm willing to work on this if we figure out the best way to do it.

jepler, 2023-10-01:
That would be neat!

deshipu, 2023-10-01:
One way to do it is to add a `positions` argument to the `decode` method, that would default to `False`, and that, if set to `True`, would make the call return a list of 6-tuples, instead of pairs.

Another way would be to add an additional method, such as `get_positions` (or a property called `positions`) that would return a list of 4-tuples with the positions corresponding to the list returned by the last `.decode()` call. It would return an empty list if there was no call previously. A potential problem here is that `decode` skips found qr codes if the data cannot be obtained, so there would be potentially a discrepancy in the number of items returned.

Finally, we could have a `find` method, that would just run the positioning part of the algorithm, without running the data extracting part, and would return 5-tuples of the code kind and positions, independently from any `.decode()` calls. This has the advantage of being faster, but a disadvantage of not being able to correlate the codes with the ones found with the `.decode()` call.

What do you think? Personally, I think the last option is nice, because it lets us home in on the codes fast, and only decode the data once we have at least one of the codes in a decent size.

jepler, 2023-10-02:
As you may imagine, the API was designed to be as simple as possible.

Maybe the extended API would have:
 * QRDecoder.analyze() which feeds the pixel data in and returns a list of 4-tuples (quirc_begin() ... quirc_end())
 * QRDecoder.extract() which, if it's given an integer, does the decode & extract from the previously analyzed image

this would leave .decode() as analyze() + looped extract().

Alternately, if this is about avoiding the attempt to decode an "obviously too small" code perhaps QRDecoder.decode just gains a new parameter for the size, and makes the decisions itself. Furthermore, maybe it should just decode the biggest QR code in the event that multiple corners are detected.

This is all totally off the cuff, I haven't thought about it in ages.

PR #8467 (2023-10-25): Add qrio.QRDecoder.find() to locate codes without decoding
Fix #8452
commit 12b6a9b5 (2023-10-25): Add qrio.QRDecoder.find() to locate codes without decoding

Fix #8452
closed 2023-10-25 by tannewt