adafruit/circuitpython #8454: synthio not built in on Feather M4

jedgarpark, 2023-10-02:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.6 on 2023-09-12; Adafruit Feather M4 Express with samd51j19
```

### Code/REPL

```python
na
```

### Behavior

synthio doesn't appear in the built-ins list for Feather M4 Express

### Description

Not sure if there's a reason synthio isn't on Feather M4 Express be or just a simple omission? Looks like it is built in on ItsyBitsy M4.

gamblor21, 2023-10-02:
I was able to build the Feather M4 express locally with `synthio` but there was mere bytes remaining.
Quickly looking the ItsyBitsy M4 does not include `audiobusio` (as the SAMD51G does not have I2S) so it likely has more space.

So it may be able to be just squeezed in but then the next core change that affects all builds would cause an issue.

FoamyGuy, 2023-10-02:
It looks like this was the commit where it was turned off: https://github.com/adafruit/circuitpython/commit/23baf02399f02f9d4a9fecab2ec9cdc974aacee5

The message on it does reference not having enough space. 

It might be it does fit on the English build, but does not on one or more of the other languages.

jedgarpark, 2023-10-02:
ah I see, thanks for looking into it @gamblor21 @FoamyGuy Is it ever the case that people could somehow make a leaner build of the built-ins in order to accommodate the stuff they want? For example, `floppyio` might not be part of someone's audio project where the do want `synthio`

FoamyGuy, 2023-10-02:
@jedgarpark I do think it's possible to create custom builds and enable / disable any core modules that have build flags. It would require modifying the mk file for the board you're building and then making a local build with the process documented in this guide: https://learn.adafruit.com/building-circuitpython

Alternatively there was functionality added so that custom builds can be created inside of Github via an actions task rather than having to do everything locally. I don't have any experience with it, but my understanding is you could similarly make modifications to an mk file and push those into your own fork then utilize this tool have github make the build with your customizations included. https://github.com/adafruit/circuitpython/pull/7594

tannewt, 2023-10-02:
Since `synthio` has improved significantly, we should try to enable it for SAMD51s. Generally newer modules don't get added to older devices that are out of space.

dhalbert, 2025-02-01:
Still doesn't fitk, as tested by #8630. I think we need to give up on this for now and recommend other boards instead.

closed 2025-02-01 by dhalbert