adafruit/circuitpython #8478: Integer vs Float issues

SimonAnnetts, 2023-10-13:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.6 on 2023-09-12; Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
a=1697226806333; print(a); print("%d" % a); print("%f" % a); print(int(a)); print(float(a));
```

### Behavior

This is the output I get from the above code:
```
1697226806333
1697226806333
1697226670080.000000
1697226806333
1.69723e+12
```

### Description

I get different values between integer and float. I suspect the int to float conversion is losing some precision and rounding is occurring somewhere. Forgive me if I'm missing something obvious about this particular CPU's bit depth or precision, or the way I'm trying to convert an integer to a float.
If I try the same code on Python 3.10.12 running on a Ubuntu desktop machine I get:
1697226806333
1697226806333
1697226806333.000000
1697226806333
1697226806333.0

dhalbert, 2023-10-13:
Floats in CircuitPython use 30 bits: 8 bits of exponents and 22 bits of mantissa. There are about 5 1/2 digits of precision. Desktop Python uses 64-bit floats. So your results are expected. Integers can be of indefinite length on all but the smallest ports.

SimonAnnetts, 2023-10-13:
Ah, that explains it,
I was actually trying to take a unix_timestamp in milliseconds (derived from a api call)  and dividing by 1000 so I could use time.localtime(seconds_since_the_epoch) to get a datetime.time_struct suitable for setting the RTC.

time.localtime seems to want a float, but I guess it then converts that back to an integer?

I guess I need to think in smaller numbers because the unix_timestamp in milliseconds is possibly 48bits long!

Assuming my time in ms is:
1697229699638
So time in secs is:
1697229699 

I guess at 32 bits even this is too big for the float that time.localtime wants?

Yet, this works and resolves to a correct local time for me:
```
>>> import time; print(time.localtime(1697229699))
struct_time(tm_year=2023, tm_mon=10, tm_mday=13, tm_hour=20, tm_min=41, tm_sec=39, tm_wday=4, tm_yday=286, tm_isdst=-1)
```
and this works:
```
>>> import time; print(time.localtime(1697229699.987654321))
struct_time(tm_year=2023, tm_mon=10, tm_mday=13, tm_hour=20, tm_min=41, tm_sec=39, tm_wday=4, tm_yday=286, tm_isdst=-1)
```

but try and change it to a real float, which is what time.localtime expects (according to my intellisense?) and I get a different answer of course due to the int->float error?
```
>>> import time; print(time.localtime(float(1697229699)))
struct_time(tm_year=2023, tm_mon=10, tm_mday=13, tm_hour=20, tm_min=35, tm_sec=12, tm_wday=4, tm_yday=286, tm_isdst=-1)
```

Anyway, thanks for highlighting the limitations of float, I'll be doing my setting of the RTC using integer math from now on,  trying never to convert to float. 
Something like:
```    
    time_seconds = int(json_response['milliseconds_since_epoch']) // 1000
    datetime = time.localtime(time_seconds)
    rtc_clock.datetime = datetime
```
Is the definition for time.localtime wrong in that it should expect an int rather than float?

dhalbert, 2023-10-13:
`time.localtime()` can take an int or a float in "regular" Python. In CircuitPython, it will also take either but it's converted to an int. 

> `>>> import time; print(time.localtime(1697229699.987654321))`

This is a real float, but the significant digits after about 5 1/2 digits are dropped. It's not the magnitude of the value (it can be up to about 10**38), but the precision. So you'll get a time, but it won't be as precise as the floating point value.

If you are trying to set your RTC, I'd say fetch a time from the web, and convert that directly to what the RTC needs, without passing it through CircuitPython's `time`. Then you can set CircuitPython's time from the RTC, instead of the other way round.

Closing this as "solved", but find to ask more questions.

closed 2023-10-13 by dhalbert

SimonAnnetts, 2023-10-13:
well, in CircuitPython the RTC needs a struct_time object, and that is a part of ```time```. I don't know how else to easily convert an integer timestamp (which might be 32 bits+) into a ```struct_time``` object without using the ``time``` library.
But anyway, using an integer timestamp in seconds (rather than a float ms or a float secs) passed to time.localtime seems to work for me and I get the correct struct_time to pass to the RTC, even though that int timestamp is 32bit.

Thanks for letting me brainstorm this with you.

dhalbert, 2023-10-13:
You can create a struct_time directly: https://docs.circuitpython.org/en/latest/shared-bindings/time/index.html#time.struct_time. That would be convenient if the original source of the time (from the web, say), is in struct_time format. But a integer value in seconds in fine too: you probably don't need it to be any more accurate.