adafruit/circuitpython #8488: Ref #4682 - Pixelbus has no attribute _transmit -- problem seems to be back

PaulskPt, 2023-10-19:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.1-74-geae426ed90 on 2023-10-18; ProS3 with ESP32S3
```

### Code/REPL

```python
id = board.board_id

if id == 'unexpectedmaker_pros3':
    use_neopixel = True
    import pros3
    import neopixel
    my_brightness = 0.005
    BLK = 0
    RED = 1
    GRN = 200
else:
    use_neopixel = False
    my_brightness = None
    BLK = None
    RED = None
    GRN = None

# in setup()
if id == 'unexpectedmaker_pros3':
        try:
            # Turn on the power to the NeoPixel
            pros3.set_ldo2_power(True)

            if not use_neopixel:
                pixels = neopixel.NeoPixel(board.NEOPIXEL, 1)
                #for i in range(len(pixels)):
                #    pixels[i] = RED
                neopixel.NeoPixel.brightness = my_brightness
                r,g,b = pros3.rgb_color_wheel( BLK )
                pixels[0] = ( r, g, b, my_brightness)
                pixels.write()
        except ValueError:
            pass
```

### Behavior

 File "code.py", line 375, in setup
AttributeError: 'PixelBuf' object has no attribute '_transmit'

### Description

I just flashed an Unexpected Maker ProS3 with the latest version of Circuitpython
then appeared the error above. There is no  use of  '_transmit' in my code.py script.

I searched and found the discussion of issue #4682.

tannewt, 2023-10-19:
I suspect we need #4688 again and we should add a test this time.

ladyada, 2023-10-21:
bumped into this too - commenting so i get notified when its closed :)

closed 2023-10-25 by dhalbert

PaulskPt, 2023-10-25:
I downloaded the "latest" CPY file:
`2023-10-25T02:50:15.000Z        2.9 MB         [adafruit-circuitpython-unexpectedmaker_pros3-en_US-20231025-702b4a5.uf2](https://adafruit-circuit-python.s3.amazonaws.com/bin/unexpectedmaker_pros3/en_US/adafruit-circuitpython-unexpectedmaker_pros3-en_US-20231025-702b4a5.uf2)
`
 about 9 hours after Dan closed this issue: 

Contents boot_out.txt:
`Adafruit CircuitPython 9.0.0-alpha.1-86-g702b4a58f8 on 2023-10-25; ProS3 with ESP32S3`

Still get: 
```
Error 'PixelBuf' object has no attribute '_transmit'
Traceback (most recent call last):
  File "code.py", line 1629, in <module>
  File "code.py", line 1583, in main
  File "code.py", line 1322, in say_hello
AttributeError: 'PixelBuf' object has no attribute '_transmit'
```
Did I download the latest too soon?

tannewt, 2023-10-25:
I thought I had fixed it but it looks like my test was wrong. Will look more today. Thanks for testing!

tannewt, 2023-10-25:
Looks like I only fixed calling `.show()` directly. The pixel assignment also calls show under the hood when auto_write is enabled. It wasn't fixed by the first change. The new PR fixes it too.

commit 63079c75 (2023-10-25): Re-add passing native methods the subclass instance

Fixes #8488
PR #8522 (2023-10-25): Pass subobject into native subscr
This allows PixelBuf to call transmit after setting a value.

Fixes #8488