adafruit/circuitpython #8489: Weird NeoPixel "jumping" on PropMaker RP2040 when num_pixels=8

todbot, 2023-10-19:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.6 on 2023-09-12; Adafruit Feather RP2040 Prop-Maker with rp2040
```

### Code/REPL

```python
import time, board, digitalio
import audiobusio, audiocore, audiomixer
import rainbowio, neopixel

num_pixels = 8
power_pin = digitalio.DigitalInOut(board.EXTERNAL_POWER)
power_pin.switch_to_output(value=True)
pixels = neopixel.NeoPixel(board.EXTERNAL_NEOPIXELS, num_pixels, brightness=0.2, auto_write=True)

audio = audiobusio.I2SOut(bit_clock=board.I2S_BIT_CLOCK, word_select=board.I2S_WORD_SELECT, data=board.I2S_DATA)
mixer = audiomixer.Mixer(voice_count=1, channel_count=1, sample_rate=22050, buffer_size=2048)
audio.play(mixer)

track1 = audiocore.WaveFile(open("/wav/StreetChicken.wav","rb"))
mixer.voice[0].play(track1, loop=True)

RED = (255,0,0)

def color_chase(wait):
    for i in range(8):
        print(i)
        pixels[i] = RED
        time.sleep(wait)
        pixels[i] = (0, 0, 0)
        time.sleep(wait)

while True:
    color_chase(0.2)
```

### Behavior

The behavior should be that the lit LED marches up the LED strip/ring.  Instead it jumps around seemingly randomly as in this video:

https://github.com/adafruit/circuitpython/assets/274093/d42e5706-e742-4ef5-b202-8409f053f0fe

### Description

It seems you need to have audio set up and going.  It could be there just needs to be some other stuff in memory. But trying to pare down the above code makes the problem go away.

Increasing `num_pixels=12` also makes the problem go away.

It takes about 3 seconds for the issue to present itself.  In the REPL, the index `i` is being printed, and it ramps up 0-7 as expected and does not jump around.

todbot, 2023-10-19:
Also, it seems to happen when the REPL is connected, but not when it's not.

RoscoP, 2023-10-22:
I had a weird LED turning off issue, it seems to be related to the `time.sleep` call.

When I run this code on an RP2040 propmaker my LED will turn off, and occasionally turn back on while sleeping.
```
import board
import pwmio
import time
rled = pwmio.PWMOut(board.D10, frequency=5000, duty_cycle=2**15)
time.sleep(5)
```

Without the sleep the LED stays on as expected.

dhalbert, 2023-12-01:
Check whether 7 pixels works fine or not.

tannewt, 2023-12-01:
Looks like the neopixel transmit won't be using DMA until 12. Without DMA, the PIO FIFO is filled from a while loop that calls background tasks: https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/common-hal/rp2pio/StateMachine.c#L869C13-L869C33

This leaves it susceptible to audio buffer fills that may mess up the neopixel transmission. Perhaps we should lower the dma threshold to where it kicks in any time the FIFO is too small.

closed 2023-12-18 by dhalbert