adafruit/circuitpython #8498: esp32s3: Implement sdioio

RetiredWizard, 2023-10-21:
I'm going to steal Jeff's train of thought from #5179 but for the esp32s3 chip.

The Makerfabs MaTouch TFT 7" board has an SD card slot wired up for 1bit SDIO and using the [link](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/storage/sdmmc.html) Jeff referenced and [example code](https://github.com/espressif/esp-idf/blob/8fc8f3f47997aadba21facabc66004c1d22de181/examples/storage/sd_card/sdmmc/main/sd_card_example_main.c) from the esp-idf, I will take a stab at implementing sdioio in the esp32s3 port but this is definitely a reach for me and I don't expect any usable results soon, so if someone likes this idea and thinks they can throw it together easily, please feel free to speak up and take it over :grin:.

jacobcrigby, 2024-06-20:
Just as a notice, I'm starting to implement this myself. I'll use this issue to keep everyone up to date with my progress.

jacobcrigby, 2024-06-27:
Just finished up a first draft you can see at https://github.com/jacobcrigby/circuitpython/tree/espressif-sdioio 

Still to-do: 
* Test a lot, on different boards
* No sdioio.reset()
* Init failed on soft reset - check and fix

RetiredWizard, 2024-06-27:
I can't wait to start testing this, unfortunately I'm on the road for the next few days and I'm not sure I'll be able to figure out how to build your repo locally so I'll probably have to wait until Monday.... I will bring the MaTouch with me just in case I have more time than I expect or you get a PR posted so the artifacts are available. Thanks for the work on this!!!

By the way, if you do post a PR to Adafruit this weekend, it would make my day if you could enable the CIRCUITPY_SDIOIO flag on the "makerfabs_tft7" port :grin:

RetiredWizard, 2024-07-08:
@jacobcrigby I finally had some time to build the branch from your repo, however when I added the CIRCUITPY_SDIOIO=1 line to the mpconfigboard.mk of the board I was going to test on, the build complained because of a missing ESP-IDF file.

``
[422/959] Building CXX object esp-idf/nvs_flash/C...les/__idf_nvs_flash.dir/src/nvs_partition.cpp.objcommon-hal/sdioio/SDCard.c:11:10: fatal error: sdmmc_cmd.h: No such file or directory
   11 | #include "sdmmc_cmd.h"
      |          ^~~~~~~~~~~~~
``

Is there another update to the CP environment to get that file included in the submodule fetch or did you manually copy the file to your local repository?

RetiredWizard, 2024-07-08:
I was able to build by replacing the include with 
```
#include "ports/espressif/esp-idf/components/sdmmc/include/sdmmc_cmd.h"
```
I assume this isn't the correct way to link to the file but at least I should be able to start testing :grin:

RetiredWizard, 2024-07-08:
I have some more testing to do but once I figured out the data pins needed to be passed as a list, I was able to mount an SD card using 1-pin SDIO. :grin:

RetiredWizard, 2024-07-09:
I've done a fair amount of testing on the makerfabs TFT7" ESP32-S3 board using 1-bit SDIO and it seems to work fine. I don't know if I have an ESP board with a 4-pin SDIO SD card but I think I may have a breakout that supports SDIO so if it's needed I can try and set that up for testing as well.

The only things I've noticed so far is that CircuitPython doesn't like it at all if you demount the sd card `storage.umount(path)`, deinit the sdio.SDCard object and then attempt to use the storage.VfsFat objected created earlier. I tried the same thing on a board with an SPI based SD card and it did the same thing though (crashes into safe mode - I'll open a new issue on that)

I also noticed that if you allocate the SDIO pins (and don't deinit the SDCard object) they are not reset by a ctrl-D, CP throws a `Failed to initialize SDMMC host: 103` error when attempting to mount an SD card until the board is power cycled completely. If you unmount/deinit the objects there's no problem re-mounting as many times as you like. I suspect this is a board level issue with never-reset logic and not really related to the new sdioio module.

jacobcrigby, 2024-07-09:
Thank you so much for doing all of this extra testing! I ended up busy, then went on vacation over the holiday, so I wasn't able to do any extra work on this one. 
I just got a few SD breakout boards in the mail, so I'm going to try out a few other ESP32 and ESP32-S3 boards I have with them to see if 4-bit works too.

RetiredWizard, 2024-07-19:
I dug out my SD card breakout and did some 4-wire testing and it seems to work fine as well. Not surprisingly the 4-wire connections had the same deinit and ctrl-D resetting issue but the deinit is probably pretty easily solved by lifting the changes that Dan referenced. If I have some time, I'll take a look at the board never-reset logic and see if I can make any headway on the ctrl-D not cleaning up the SDIO pins.

There was one additional issue related to the deinit problem that I noticed this time. When the sdioio module throws a Failed to initialize SDMMC host message it doesn't release (deinit) the SDIO_CLK (and presumably the SDIO_CMD) pin. So subsequent calls throw a "Pin in use" message. I have sometimes been able to get a stubborn SD card to mount by ejecting and re-inserting it, however if the pins are not released after a failure then more drastic (board reset) measures are required. 

On a more of a curious front, when I initialized the sdioio.SDCard object with only 1 data line, I didn't notice a speed difference. I copied a 50K file from the SD card and in both cases it took about 15-16 seconds. I did leave the other three data lines connected since when the were left floating, the card wouldn't mount.

RetiredWizard, 2024-09-13:
@jacobcrigby It will take some digging and I don't know how long it will take to update your code for the deinit issue Dan mentioned, but if you're lost to other projects, I could pick this up and try and get a PR published. 

Let me know :grin:

PR #9641 (2024-09-20): esp32s3: Implement sdioio
This closes #8498

@jacobcrigby created this update (Thanks so much!!!) to support sdioio for ESP32S3 and I think they planned on submitting it as a PR, however with the recent updates to the IDF I wanted to get this code integrated before the refactoring got more difficult. 

Dan had mentioned that updates from #9418 would probably be needed which I've gone ahead and done. The symptoms from #9417 appear to have been resolved with those updates.

I noticed some messages in circuitpython.pot that were very similar and I thought could be factored out so I did a little m [... cut]
closed 2024-09-20 by tannewt