adafruit/circuitpython #8499: 9.x AttributeError: 'Display' object has no attribute 'show'

ladyada, 2023-10-21:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.1-77-g2c47221cac on 2023-10-21; Adafruit Camera with ESP32S3
```

### Code/REPL

```python
def init_display(self):
        # construct displayio by hand
        displayio.release_displays()        
        self._display_bus = displayio.FourWire(self._spi, command=board.TFT_DC,
                                               chip_select=board.TFT_CS,
                                               reset=board.TFT_RESET,
                                               baudrate=60_000_000)
        self.display = board.DISPLAY
        # init specially since we are going to write directly below
        self.display = displayio.Display(self._display_bus, self._INIT_SEQUENCE,
                                         width=240, height=240, colstart=80,
                                         auto_refresh=False)
        print(dir(self.display))
        self.display.show(self.splash)
        self.display.refresh()
```

### Behavior

AttributeError: 'Display' object has no attribute 'show'

```
print(dir(self.display))
['__class__', 'auto_refresh', 'brightness', 'bus', 'fill_row', 'height', 'refresh', 'root_group', 'rotation', 'width']
```

### Description

scott mentioned this and there's no existing issue (that is aw) so we can have one here!

RetiredWizard, 2023-10-21:
I believe .show was depreciated in 8.x and removed in 9.x

https://github.com/adafruit/circuitpython/pull/8456

ladyada, 2023-10-21:
ooh ok that maybe what im forgetting!

DJDevon3, 2023-10-22:
show is still valid in 8.x and will be the last version it is valid.
transitioning to total deprecation of display.show in 9.x in favor of display.root_group
do something like this instead
```py
main_group = displayio.Group
display.root_group = main_group
```

tannewt, 2023-10-23:
We just discussed this in our internal meeting. We decided to add a show() function implementation that raises an exception when used to specify how to fix the code.

PR #8517 (2023-10-25): Add friendlier error when .show() used
Fixes #8499

```
Adafruit CircuitPython 9.0.0-alpha.1-99-gecaf9e6b14-dirty on 2023-10-24; Adafruit Feather ESP32-S3 TFT with ESP32S3
>>> import board
>>> board.DISPLAY.show(None)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: .show(x) removed. Use .root_group = x
>>> 
```
commit 0a1e36d2 (2023-10-25): Add friendlier error when .show() used

Fixes #8499
closed 2023-10-25 by dhalbert