adafruit/circuitpython #8503: Web workflow deletes chunk of code.py on save

BlitzCityDIY, 2023-10-22:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.1-64-gf14ca95df4 on 2023-10-06; Adafruit MatrixPortal S3 with ESP32S3
```

### Code/REPL

```python
import os
import gc
import ssl
import time
import wifi
import socketpool
import adafruit_requests
import adafruit_display_text.label
import board
import terminalio
import displayio
import framebufferio
import rgbmatrix
import adafruit_json_stream as json_stream
import microcontroller
from adafruit_ticks import ticks_ms, ticks_add, ticks_diff
from adafruit_datetime import datetime, timedelta
from adafruit_io.adafruit_io import IO_HTTP
import neopixel
import vectorio

displayio.release_displays()

# font color for text on matrix
font_color = 0xFFFFFF
# your timezone UTC offset and timezone name
timezone_info = [-4, "EDT"]

month_name = ["Jan.", "Feb.", "Mar.", "Apr.", "May", "June", "July", "Aug.",
              "Sept.", "Oct.", "Nov.", "Dec."]
weekday = ["Mon.", "Tues.", "Wed.", "Thurs.", "Fri.", "Sat.", "Sun."]

# the name of the sports you want to follow
sport_name = ["football", "baseball", "hockey", "basketball"]
sport_league = ["nfl", "mlb", "usa.1", "nhl", "nba"]
# the team names you want to follow
# must match the order of sport/league arrays
# include full name and then abbreviation (usually city/region)
team0 = ["New England Patriots", "NE"]
team1 = ["Boston Red Sox", "BOS"]
team2 = ["New England Revolution", "NE"]
team3 = ["Boston Bruins", "BOS"]
team4 = ["Boston Celtics", "BOS"]
# how often the API should be fetched
fetch_timer = 300 # seconds
# how often the display should update
display_timer = 30 # seconds

pixel = neopixel.NeoPixel(board.NEOPIXEL, 1, brightness = 0.3, auto_write=True)

# matrix setup
base_width = 64
base_height = 32
chain_across = 2
tile_down = 2
DISPLAY_WIDTH = base_width * chain_across
DISPLAY_HEIGHT = base_height * tile_down
matrix = rgbmatrix.RGBMatrix(
    width=DISPLAY_WIDTH, height=DISPLAY_HEIGHT, bit_depth=3,
    rgb_pins=[
        board.MTX_R1,
[... 150 lines cut ...]
```
connected
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 38
SyntaxError: invalid syntax

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```
 reopening the code.py file, line 38 now showed this:

```
sport_name = ["football", "baseball", "hockey", "basketba��������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������ins = 0
```
deleting up to line 108. 

This happened twice, but the first time it only deleted up to line 102. However, it did insert the mystery characters in the sport_name array in the same spot. unfortunately i didn't have the chrome console open to see any possible error logs

BlitzCityDIY, 2023-10-22:
i was able to have it happen again with the chrome console open:

```
Connected!
3index.js:37 write
index.js:37 Initializing File Transfer Client...
index.js:37 Waiting for connection status to change...
index.js:37     PUT http://cpy-2048dc.local/fs/code.py net::ERR_CONNECTION_RESET
_fetch @ index.js:37
writeFile @ index.js:37
await in writeFile (async)
writeFile @ index.js:24
writeFile @ index.js:35
op @ index.js:37
saveFile @ index.js:35
Gd @ index.js:37
await in Gd (async)
u @ index.js:9
Du @ index.js:9
keydown @ index.js:9
runCustomHandlers @ index.js:8
t @ index.js:8
(anonymous) @ index.js:8
index.js:37 write failed ProtocolError: Host 'cpy-2048dc.local' not found.
    at mc._fetch (index.js:37:2577)
    at async mc.writeFile (index.js:37:1942)
    at async xS.writeFile (index.js:24:4482)
    at async KS.showBusy (index.js:35:106428)
    at async KS.writeFile (index.js:35:108288)
    at async KS.op [as _saveFileContents] (index.js:37:14800)
    at async KS.saveFile (index.js:35:107513)
    at async Gd (index.js:37:11447) ProtocolError: Host 'cpy-2048dc.local' not found.
    at mc._fetch (https://code.circuitpython.org/assets/js/index.js:37:2577)
    at async mc.writeFile (https://code.circuitpython.org/assets/js/index.js:37:1942)
    at async xS.writeFile (https://code.circuitpython.org/assets/js/index.js:24:4482)
    at async KS.showBusy (https://code.circuitpython.org/assets/js/index.js:35:106428)
    at async KS.writeFile (https://code.circuitpython.org/assets/js/index.js:35:108288)
    at async KS.op [as _saveFileContents] (https://code.circuitpython.org/assets/js/index.js:37:14800)
    at async KS.saveFile (https://code.circuitpython.org/assets/js/index.js:35:107513)
    at async Gd (https://code.circuitpython.org/assets/js/index.js:37:11447)
index.js:37 write
index.js:37 write failed Error: Unable to perform file operation. Not Connected.
    at mc._checkConnection (index.js:37:1156)
    at mc.writeFile (index.js:37:1783)
    at xS.writeFile (index.js:24:4493)
    at KS.writeFile (index.js:35:108324)
    at op (index.js:37:14809) Error: Unable to perform file operation. Not Connected.
    at mc._checkConnection (https://code.circuitpython.org/assets/js/index.js:37:1156)
    at mc.writeFile (https://code.circuitpython.org/assets/js/index.js:37:1783)
    at xS.writeFile (https://code.circuitpython.org/assets/js/index.js:24:4493)
    at KS.writeFile (https://code.circuitpython.org/assets/js/index.js:35:108324)
    at op (https://code.circuitpython.org/assets/js/index.js:37:14809)
```

i had a few successful saves before this and was editing a different part of the code. it hung, i got that print out in the chrome console continuously, i refreshed the browser, reconnected to the board, opened the REPL, had this message:

```
connected
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 165
SyntaxError: invalid syntax

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

```

opened code.py and this was now line 165:

```
condition_text = adafruit_display_text.label.Label(terminalio.FONT, color=font_color,
                                              text=" ",
                                              x=5, y=35��������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������er, x = 0)
```

tannewt, 2023-10-23:
Do you know if the ESP board restarted?

BlitzCityDIY, 2023-10-23:
i'm not sure. i can try a newer build and see if the issue persists.

tannewt, 2023-10-24:
I assume it does. I may need to get you setup with separate UART debug logging for more details.

BlitzCityDIY, 2023-12-17:
just had this occur again with alpha.6

tannewt, 2024-02-27:
@BlitzCityDIY Have you had a chance to use web workflow recently? Is this still an issue?

BlitzCityDIY, 2024-02-27:
i tried again tonight. i erased the flash and installed 9.0beta 2. it has not been erasing chunks of code but it has been going into safe mode inconsistently (not a very helpful data point). it's happened 2/5 times on saving and then randomly while running the code. i'll be connected to the serial monitor on web workflow and it stops, doesn't show disconnect and i see the neopixel flashing yellow 3x. might be time for the debug build?

BlitzCityDIY, 2024-02-27:
actually the safe mode issue seems to be separate from web workflow. i power cycled and didn't try to get back into it and it went into safe mode on its own after a while. i'll try to get more info tomorrow

BlitzCityDIY, 2024-03-06:
i think this is good to close. i've been using absolute newest and adding/deleting chunks of code and the issue has not popped up where previously it would have during that kind of use. thank you! looking forward to using it more 🙂

tannewt, 2024-03-06:
> i think this is good to close. i've been using absolute newest and adding/deleting chunks of code and the issue has not popped up where previously it would have during that kind of use. thank you! looking forward to using it more 🙂

Great! Thanks for the update.

closed 2024-03-06 by tannewt