adafruit/circuitpython #8507: The doesn't appear to be a way to read buttons on the Qualia ESP32-S3

makermelissa, 2023-10-23:
### CircuitPython version

```python
Absolute Latest CircuitPython 9
```

### Code/REPL

```python
N/A
```

### Behavior

The buttons are on the GPIO Expander and the code for the expander (inside dotclockframebuffer) seems to only allow sending init sequences, but not reading values.

jepler, 2023-10-23:
there should be some library for this particular i2c chip that would be used after the LCD initialization. (maybe it doesn't exist yet?) We talked about adding I/O expander in the core (#8377) but didn't do it for some reasons described there.

makermelissa, 2023-10-23:
Ok, I'll see if I can get it working with Adafruit_CircuitPython_PCF8574 and if so, add an example and close this issue.

makermelissa, 2023-10-23:
It almost works. Pushing the button has no effect, but it does grab a value from the expander. I'll have to dive more into the specifics of why that may be when I have more than 10 minues.

makermelissa, 2023-10-23:
Whenever I try switching it to an input and I get the following error:
```
Traceback (most recent call last):
  File "code.py", line 87, in <module>
  File "/lib/adafruit_pcf8574.py", line 142, in switch_to_input
  File "/lib/adafruit_pcf8574.py", line 170, in direction
  File "/lib/adafruit_pcf8574.py", line 87, in write_pin
  File "/lib/adafruit_pcf8574.py", line 74, in write_gpio
OSError: [Errno 19] No such device
```
I also tried using bitbangio, but received this error:
```
Traceback (most recent call last):
  File "code.py", line 87, in <module>
  File "/lib/adafruit_pcf8574.py", line 142, in switch_to_input
  File "/lib/adafruit_pcf8574.py", line 170, in direction
  File "/lib/adafruit_pcf8574.py", line 87, in write_pin
  File "/lib/adafruit_pcf8574.py", line 74, in write_gpio
OSError: [Errno 5] Input/output error
```

makermelissa, 2023-10-27:
This could be related to #4621.

makermelissa, 2023-10-31:
It looks like the PCA9554 has special command bytes for configuring the port and having its own CP driver would probably be the best approach.

jepler, 2023-10-31:
Modifying tags to reflect that a new library is needed

makermelissa, 2023-10-31:
I'll update this if I find it continues to refuse to work, but I don't think it will be an issue.

makermelissa, 2023-10-31:
Ok, I have something functioning. New lib was definitely the way to go.

makermelissa, 2023-10-31:
New library created: https://github.com/adafruit/Adafruit_CircuitPython_PCA9554

closed 2023-10-31 by makermelissa