adafruit/circuitpython #8513: supervisor/shared/web_workflow/web_workflow.c: #endif placement

RetiredWizard, 2023-10-24:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.1-82-gec678a54ce on 2023-10-23; ESP32-S2-DevKitC-1-N4R2 with ESP32S2
```

### Code/REPL

```python
No code, compilation issue
```

### Behavior

A few weeks ago I was trying to perform a debug build on CircuitPython and I ran into an issue with the placement of the #endif on line 371 of web_workflow.c. I moved the #endif down two lines which I think resolved that issue and figured I get back to filing an issue/pr once I finished my debugging. Unfortunately, I can't recreate the compile error now, but looking at the code I'm not sure why it works when CIRCUITPY_WEB_WORKFLOW is set to zero.

The current code looks like this:
```c
bool supervisor_start_web_workflow(bool reload) {
    #if CIRCUITPY_WEB_WORKFLOW && CIRCUITPY_WIFI && CIRCUITPY_OS_GETENV

    char ssid[33];
    char password[64];

           // LOTS OF STUFF......

    if (initialized) {
        if (!common_hal_socketpool_socket_get_closed(&active)) {
            common_hal_socketpool_socket_close(&active);
        }

        #if CIRCUITPY_MDNS
        // Try to start MDNS if the user deinited it.
        if (mdns.base.type != &mdns_server_type ||
            common_hal_mdns_server_deinited(&mdns)) {
            mdns_server_construct(&mdns, true);
            mdns.base.type = &mdns_server_type;
            if (!common_hal_mdns_server_deinited(&mdns)) {
                common_hal_mdns_server_set_instance_name(&mdns, web_instance_name);
            }
        }
        if (!common_hal_mdns_server_deinited(&mdns)) {
            common_hal_mdns_server_advertise_service(&mdns, "_circuitpython", "_tcp", web_api_port);
        }
        #endif

        if (common_hal_socketpool_socket_get_closed(&listening)) {
            socketpool_socket(&pool, SOCKETPOOL_AF_INET, SOCKETPOOL_SOCK_STREAM, &listening);
            common_hal_socketpool_socket_settimeout(&listening, 0);
            // Bind to any ip. (Not checking for failures)
            common_hal_socketpool_socket_bind(&listening, "", 0, web_api_port);
            common_hal_socketpool_socket_listen(&listening, 1);
        }
        // Wake polling thread (maybe)
        socketpool_socket_poll_resume();
    #endif
    return true;
}
return false;
}
```
It looks to me like the indention at the end should be like this:
```c
        // Wake polling thread (maybe)
        socketpool_socket_poll_resume();
    #endif
        return true;
    }
return false;
}
```
And then the #endif should be moved down two lines to be outside the if(initialized) block. I'm surprised it even compiles, but won't the code as it stands now return **True** to a supervisor_start_web_workflow call if any of the three CIRCUITPY parameters are **False**? Perhaps supervisor_start_web_workflow isn't called under those conditions and it doesn't matter.

cuevana8, 2023-10-24:
+1..[.](https://github.com/flslib/fl-studio-cracked)[@@@](https://github.com/adafruit/circuitpython/issues/8513#issue-1958408291)

tannewt, 2023-10-24:
This just looks like a bug. Feel free to PR a change to move it.

PR #8515 (2023-10-24): web_workflow: adjust indent and move #endif
fixes #8513
closed 2023-10-24 by tannewt