adafruit/circuitpython #8525: `re` module not matching [\s\S] and equivalents

michalpokusa, 2023-10-26:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.4 on 2023-08-22; Adafruit Feather ESP32-S2 TFT with ESP32S2
```

### Code/REPL

```python
>>> import re
>>> re.search(r">\S*?<", ">test<") is None
False
>>> re.search(r">\s*?<", "> <") is None
False
>>> re.search(r">[\s\S]*?<", "> test <") is None
True
>>> re.compile(">.*?<", 16).search("> test <") is None
False
```

### Behavior

When searching for `\S` aka any non-whitespace character, `re` module finds a match.
Similarly, when searching for `\s` aka any whitespace character, `re` module also finds a match.

But, when combining them together, [creating a popular any character](https://stackoverflow.com/a/55149095) `re` module is unable to match.

This pattern is correct, and in theory should be possible to use in CircuitPython, as all the components (`\s`, `\S`, sets `[]`, non-greedy `*?`) are supported.
[image]

Similar issue hapens when using \d and \D, \w and \W.

Using `.compile()` with `re.DOTALL` which is a literal `16`, and with changed patterns (`[\s\S]` to `.`) seems to make it possible to match such strings, but still, the issue is present, it is just going around it.

### Additional information

Possibly related, **but maybe the cause is completely different**: https://github.com/adafruit/circuitpython/issues/6860

michalpokusa, 2023-10-26:
> Could you test this on a 9.0.0 build to see if the updates from MicroPython fixed it? Thanks.

Sure thing. I will test and come back with results.

michalpokusa, 2023-10-26:
Well, turns out it does in fact seem to fix the issue. I probably should have tested it myself before posting. Sorry for that.
[image]

dhalbert, 2023-10-26:
No problem - this is good. We'll want to decide whether to backport the fix to 8.2.x or simply leave your workaround in until 8.x.x has been supplanted by 9.x.x.

dhalbert, 2023-11-07:
I will close this, since it's fixed in 9.0.0. We could backport the fix to 8.2.x but having the workaround in the library is good for anyone running a slightly older version, so no need to backport.

closed 2023-11-07 by dhalbert