adafruit/circuitpython #8529: ESP32-S3 hard fault instantiating UART object after I2C object

DonutCables, 2023-10-27:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.7 on 2023-10-19; VCC-GND YD-ESP32-S3 (N16R8) with ESP32S3
```

### Code/REPL

```python
from busio import I2C, UART
import board
i2c = I2C(board.GPIO41, board.GPIO42)
uart = UART(board.GPIO1, board.GPIO2, baudrate=9600)
```

### Behavior

Board disconnects from the computer, power cycles, then flashes triple yellow lights on loop.
On plugging it in and connecting via serial, this is the readout
```
You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.
```

### Description

Initially discovered that instantiating an I2C 1602 display and a UART audio card in the same imported file would cause a crash into safe mode.

Running the following by going into the REPL while it's in safe mode:
```import supervisor
supervisor.runtime.safe_mode_reason
```
Returns `SafeModeReason.Hard_Fault`

### Additional information

Workarounds found:
 - Having no UART instantiation in the file no longer crashes on importing the file.
 - Moving the UART instantiation to *before* the I2C instantiation also gives no crash
 - Instatiating the I2C object at a lower speed such as `frequency=20000` does not crash

tannewt, 2024-01-18:
This works ok for me on a metro s3 with main and 8.2.9. Please retest with an "absolute newest" 9.x build and 8.2.9.

tannewt, 2024-01-24:
Please comment if this still happens and we'll reopen.

closed 2024-01-24 by tannewt