adafruit/circuitpython #8532: CircuitPython 9 does not reliably import modules

rgov, 2023-10-27:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.2 on 2023-10-27; Raspberry Pi Pico with rp2040
```

### Code/REPL

```python
>>> import adafruit_pcf8575
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'adafruit_pcf8575'

>>> import sys
>>> sys.path
['', '/', '/lib']

>>> open('/lib/adafruit_pcf8575.mpy')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 2] No such file/directory: /lib/adafruit_pcf8575.mpy

>>> open('/lib/adafruit_pcf8575.mpy')
<io.TextIOWrapper 0x2000eb20>
```

### Behavior

The filesystem is inconsistent about whether the modules exist.

tannewt, 2023-10-30:
Had you recently copied the file over? Sometimes the host OS caches writes. Running `sync` on Mac or Linux can help.

dhalbert, 2023-10-30:
macOS Sonoma acts differently than previous versions: it delays writes, sometimes significantly. This is a new and significant problem: see #8449. As @tannewt says, `sync` can help.

rgov, 2023-10-30:
Yes but I am pretty sure I had reset the board a few times in my testing.

bill88t, 2023-11-01:
Resetting the board (even with the `microcontroller` module) without ejecting will leave the files unwritten and perhaps even corrupt the filesystem. Those writes will not be performed after a remount.
If your code resets the board, make sure to eject before running.

rgov, 2023-11-02:
For what it's worth when I reverted to 8.2 I didn't encounter this issue anymore, whereas on 9 it was seriously inhibiting my ability to work.

dhalbert, 2024-01-24:
@rgov What was the host operating system you were using, and how were you copying the files over?

rgov, 2024-01-24:
macOS Sonoma, which as mentioned may delay writes.

closed 2024-02-10 by rgov