adafruit/circuitpython #8539: CPY script halts at i2c.writeto_then_readfrom()

PaulskPt, 2023-10-29:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.2 on 2023-10-27; ProS3 with ESP32S3
```

### Code/REPL

```python
# Test I2C between UM ProS3 and UM TinyPICO RTC shield (MCP7940)
# Circuitpython V 9.0.0.alpha-2
# Testing the function: i2c.write_then_readinto() 
# by Paulus Schulinck (Github @PaulskPt)
#
import board
import rtc
import mcp7940
import time
import sys, os
import wifi
import ipaddress
import socketpool
import adafruit_ntp_v2

my_debug = False
ip = None
s__ip = ""
NTP_dt_is_set = False
ntp_last_sync_dt = None
MCP_dt = None

id = board.board_id

if id == 'unexpectedmaker_pros3':
    import pros3
    import neopixel
    
mRTC = rtc.RTC()  # create internal RTC object

try:
    i2c = board.STEMMA_I2C()
    if my_debug:
        print(f"i2c: {i2c}")
except RuntimeError as e:
    # print(f"Error while creating i2c object: {e}")
    if e:
        e = None
    raise

mcp = mcp7940.MCP7940(i2c)

import adafruit_ntp_v2
pool = socketpool.SocketPool(wifi.radio)
ntp = adafruit_ntp_v2.NTP(pool, tz_offset=0)

def do_connect():
    global ip, s__ip
    TAG = "do_connect(): "

    # Get env variables from file settings.toml
    ssid = os.getenv("CIRCUITPY_WIFI_SSID")
    pw = os.getenv("CIRCUITPY_WIFI_PASSWORD")

    try:
        wifi.radio.connect(ssid=ssid, password=pw)
    except ConnectionError as e:
        print(TAG+f"WiFi connection Error: \'{e}\'")
    except Exception as e:
        print(TAG+f"Error: {dir(e)}")

[... 777 lines cut ...]
+--------+---------+-------+----------------------------------+
See: MCP7940N datasheet DS20005010H-page 27

Alarm interrupt status:
Expect next alarm at minute: 43
+-------------+----------+-------+-----+------+--------+--------+---------+---------------------+--------------------+
|  ALARM  Nr  | ENABLED? | MONTH | DAY | HOUR | MINUTE | AM/PM  | WEEKDAY | INTERRUPT OCCURRED? | NOTES:             |
+-------------+----------+-------+-----+------+--------+--------+---------+---------------------+--------------------+
|      X      |    X     |  10   |  20 |   7  |   43   |   PM   |   Fri   | X                   | CURRENT DATETIME   |
+-------------+----------+-------+-----+------+--------+--------+---------+---------------------+--------------------+
|      1      |   Yes    |  10   |  20 |   7  |   43   |   PM   |   Fri   | Yes                 | ALARM1 SET FOR     |
+-------------+----------+-------+-----+------+--------+--------+---------+---------------------+--------------------+
main():                 RING RING RING we have an interrupt from the RTC shield!
KeyboardInterrupt. Exiting...

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 8.2.6 on 2023-09-12; ProS3 with ESP32S3
```

### Additional information

Even when I replace the command  `self._i2c.writeto_then_readfrom(MCP7940.ADDRESS, out_buf, in_buf)`   by             `self._i2c.writeto(MCP7940.ADDRESS, out_buf)`, the script seems to hang.

PaulskPt, 2023-10-29:
The test above is an extract of the code in my repo at: `https://github.com/PaulskPt/Circuitpython_MCP7940`.
I created this test because of the 'lock-up' that occurred with CPY version: Adafruit CircuitPython 9.0.0-alpha.2 on 2023-10-27; ProS3 with ESP32S3.

PaulskPt, 2023-10-30:
Today I re-flashed the UM ProS3 with the latest stable CPY: V8.2.7. Same problem. I added an i2._scan(). That reported:
```
i2c device address: 0x3d
i2c device address: 0x6f
```
The 0x3d is an Adafruit 1.12in 128x128 mono OLED display (SH1107)
The 0x6f is the UM TinyPICO RTC shield (MCP7940)

PaulskPt, 2023-10-30:
Next I re-flashed the UM ProS3 with CPY V8.2.6. Same result. I am afraid that the problem is on my side. I close this issue.

closed 2023-10-30 by PaulskPt

dhalbert, 2023-10-30:
I was thinking this was showing an I2C bug in 9.0.0-alpha.2. Why do you think the problem is on your side if it works in 8.2.x but not in 9.0.0-alpha.2? If you can generate simple test program, that would be great. Maybe open a new issue. Thanks.

PaulskPt, 2023-10-30:
Thank you Dan. As I wrote (maybe not clear enough), the problem also appeared this morning using V8.2.7 and V8.2.6. I am checking my script and then start a new issue if necessary.

dhalbert, 2023-10-30:
Ah, sorry, I mistakenly thought that the I2C scans were indicating it was working in 8.x.x. No problem.

PaulskPt, 2023-10-30:
Further investigation revealed that: yes, it was a bug in my mcp7940.py file.

PaulskPt, 2023-10-30:
Latest test using: Adafruit CircuitPython 9.0.0-alpha.2-5-g6b62df4054 on 2023-10-29; ProS3 with ESP32S3
runs OK