adafruit/circuitpython #8543: [espressif / lolin_s2_mini] Inconsistency in defined pins

bato3, 2023-10-30:
There is some inconsistency in the Lolin S2 Mini plate.

The chip itself may have freely configured pins as i2c, sleep, uart, and the tile itself has no descriptions, only numbers.

Some of the pins were defined as compatible with the Lolin D1 Mini.

Similarly with S3 Mini, where there are only numbers directly on the tile, but in Pinout there are descriptions: https://www.wewos.cc/en/latest/s3/s3_mini.html

SCL, SDA, SCK, Miso, Mosi, but not RX and TX (SS - I suggest skipping) were defined.

The method of `Board.spi ()`, but there is no `Board.I2c ()` (not to mention UART ...)

I am asking for unification: adding missing pins / methods.

dhalbert, 2023-10-30:
Are you interested in doing a pull request yourself? That would be great. Most of us don't have a board to test.

bato3, 2023-10-31:
Is there any procedure for them to accept the code? (Some confirmation of license acceptance or something similar?)

And direct me to what tests should be performed, because the instructions are probably this: https://learn.adafruit.com/building-circuitpython/espressif-build

tannewt, 2023-10-31:
No license acceptance. The license that applies is in the file headers.

Test the build on your board to make sure it works. After you PR is proposed, further tests will be run automatically.

More details here: https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github

romkey, 2023-11-01:
I don't think you should add labels for RX and TX; the board doesn't expose those pins on its headers.

I've used the Lolin S2 mini in a few designs. It's a nice board and its footprint compatibility with the D1 mini is helpful. I was very frustrated by the lack of access to RX and TX when I had crashes and couldn't get the exception information over USB CDC and had no access to TX.

You can confirm this in [ESP32-S2 data sheet](https://www.espressif.com/sites/default/files/documentation/esp32-s2_datasheet_en.pdf); on page 14 in section 2.2 "Pin Description" it specifies that RX and TX (U0RXD and U0TXD) are GPIO44 and GPIO43.

Unfortunately the [Lolin S2 mini](https://www.wemos.cc/en/latest/s2/s2_mini.html) doesn't make those pins available, so I think it would be misleading to define them.

deshipu, 2023-11-01:
> I don't think you should add labels for RX and TX; the board doesn't expose those pins on its headers.
>
> I've used the Lolin S2 mini in a few designs. It's a nice board and its footprint [...]

I think there is some confusion here. The ESP32-S2 and ESP32-S3 chips have real USB support, and the USB serial for the REPL is a simulated serial device – it has no pins, it just writes to USB directly, not like on the ESP32 chips that don't have USB support, and have to rely on a separate serial-to-usb chip, connected to real gpio pins on the board.

That doesn't mean that there can't be pins named `board.RX` and `board.TX` that are used by default when creating `board.UART()` – it's a completely separate thing from the USB serial.

romkey, 2023-11-01:
I understand that.

On an ESP32 series chip, I think that `board.RX` and `board.TX` should reflect UART0's RX and TX pins, and on the Lolin S2 mini those pins aren't exposed on the headers, which is why I don't think they should be added to the pin definitions (and why they aren't there currently).

deshipu, 2023-11-01:
They are just the default UART. It doesn't matter which one. It's just a convenience.

dhalbert, 2023-11-01:
To be clear here, it sounds like GPIO43 and GPIO44 are not exposed at all on the Lolin S3 Mini, and would not be available except via bodge wiring to the chip pins. In that case, there should not be `board.RX` and `board.TX` pins

@deshipu I don't understand if the pins are not accessible why they would be listed. Maybe you meant something else?

romkey, 2023-11-01:
Why would you define `board.RX` and `board.TX` when you can't actually wire anything up to the module's default UART pins on that board? Why would you define the "default UART" pins to be something other than what the data sheet specifies? What pins would you propose that `board.RX` and `board.TX` be in that case?

romkey, 2023-11-01:
@dhalbert they're not exposed on the S2 mini - they are exposed on the S3 mini. But I think this issue was about the S2 mini.

dhalbert, 2023-11-01:
> @dhalbert they're not exposed on the S2 mini - they are exposed on the S3 mini. But I think this issue was about the S2 mini.

There is a link in the OP to the S3 mini. Maybe @deshipu was referring to the S3 instead of the S2. That was my mistake as well.

romkey, 2023-11-01:
@dhalbert yeah I think there may be some conflating going on :) I had to triple check to make sure of what the issue was about ;)

romkey, 2023-11-01:
To be clear, my understanding is that @bato3 was saying "the Lolin S2 mini `board` object doesn't define `RX` and `TX` and the Lolin S3 mini does, so we should update the Lolin S2 mini pin definitions to as well".

I was pointing out that the Lolin S2 mini doesn't expose those pins and therefore the `board` object shouldn't define them. Datasheet and board spec links in my [original comment](https://github.com/adafruit/circuitpython/issues/8543#issuecomment-1789387531) to back that up. The Lolin S3 mini does expose those pins, so it's appropriate that its `board` define them.

deshipu, 2023-11-01:
What I was trying to say is that it doesn't have to be the same GPIO numbers as on the S3, just the same positions on the physical development board. The justification for this is that if there is any kind of shield or project PCB that uses the TX and RX pins on the D1 Mini, it will work by default on the S2 and S3 as well. It's the physical positions of the pins that matter, not their internal numbers.

bato3, 2023-11-06:
> To be clear, my understanding is that @bato3 was saying "the Lolin S2 mini `board` object doesn't define `RX` and `TX` and the Lolin S3 mini does, so we should update the Lolin S2 mini pin definitions to as well".

Eh... My original intention was to draw attention to the fact that there is `board.SPI()` and not `board.I2C()`.

Additionally, while looking for information on where the SPI pins are located, I read that all boards from the *Wemos mini series are pin-compatible.* (or at least the external ones). Therefore, knowing from the ESP32-S2 datasheet that UART, I2C, SPI2, SPI3 can be any GPIO pin, I suggested additionally defining `RX`, `TX`.

Btw, I'm not only one https://github.com/wemos/docs/issues/54

romkey, 2023-11-07:
@deshipu Of course I totally agree with you about maintaining footprint compatibility. Unfortunately Wemos already broke footprint compatibility by not exposing U0RXD and U0TXD. Any confusion stems from their design decision. I don't think that CircuitPython should double down on that confusion by defining pins that the manufacturer did not.

Consider the case of users who have already used those pins for something else in their design. Or users who expect exception and first stage ROM boot loader output on the board.RX and board.TX (despite that having nothing with CircuitPython, on ESP32 series chips that is a reasonable and default assumption) and then not getting it. Experienced users will be fine but they're the minority.

Ultimately this is a pretty minor issue. I really don't think it's worth much of any of our time.

romkey, 2023-11-07:
@bato3 It would be nice if "_Wemos mini series are pin-compatible_" were true, but due to Wemos' design decision on the S2 mini board, it's not. And yes, you can map UART signals to any GPIO pin - while your application is running. On this particular board there simply are no default UART pins even though there should be, which is why I believe that `board.RX` and `board.TX` shouldn't be defined. I'm only advocating not to make that change; I'm not offering an opinion on any other issues you mentioned.

The suggestion at (demos/docs#54)[https://github.com/wemos/docs/issues/54) doesn't really help the overall situation. The pins in the traditional RX/TX positions on the footprint are not U0RXD and U0TXD. They won't function as RX and TX until application software remaps the GPIO mux. UART0 has some specific uses during the boot process, and for flashing (if one were to flash via serial rather than USB). If those pins on the footprint are marked RX and TX, it would be reasonable to assume they would work to access UART0, and they won't. I don't think that trying to remedy a confusing design decision by further confusing the issue is a good idea. It's unfortunate that Wemos designed the board this way, but it's not a problem CircuitPython or any application framework should or can realistically fix.

romkey, 2023-11-08:
BTW the CircuitPython download page for the Lolin S2 mini doesn't promise footprint compatibility and mentions there's only a "high chance" of compatibility.

Perhaps to help stem confusion it might help to note "Because the board has no default RX and TX pins, CircuitPython does not define `board.RX` or `board.TX`"?

romkey, 2023-11-08:
I apologize for continuing to post about this :) I have one of these boards and just remembered this.

There's an existing product called ["Lumostick"](https://www.tindie.com/products/bradanlane/lumosstick-circuitpython-led-long-matrix/) which runs CircuitPython and embeds a Lolin S2 mini in its hardware, driving an LED matrix.

The [Lolin S2 mini](https://www.wemos.cc/en/latest/s2/s2_mini.html) exposes GPIO37 and GPIO39 where RX and TX traditionally are on its footprint. Because these are not actually UART0 pins and because Wemos' documentation for the board doesn't call them out for that function, the person who designed the board uses these pins to [drive rows of Neopixels](https://gitlab.com/bradanlane_cp/lumosstick/-/blob/main/lib/lumosstick.py?ref_type=heads#L186). If these pins were actually connected to UART0 there would be output on them at power-on or reset and the board designer wouldn't have connected them to the LEDs.

I'm offering this as a real life example of how defining `board.RX` and `board.TX` to use the traditional pins on this board could cause problems. This is an existing product running CircuitPython which has these pins already in use for something else because the board documentation said they could do that. It's not a purely hypothetical problem.

It's also a pretty cool board.

deshipu, 2023-11-08:
@romkey I still don't see how that is a problem. CircuitPython will not output anything on the default uart pins until you actually call `board.UART()` and create the uart object. Until you do that, you can use those pins for anything you want, including driving neopixels.

romkey, 2023-11-08:
@deshipu The assertion is that the Lolin S2 mini is footprint compatible with the original D1 mini and therefore should have the same pins made available through `board`.

It is not. The manufacturer's own technical documentation shows that it is not. I've been sharing examples of how it is not, including an existing product built using it that runs CircuitPython that would not even work if it _were_ footprint compatible.

This is a problem that the manufacturer created when they designed the board this way. Yes, many of us wish they had maintained footprint compatibility. It is frustrating and confusing that they didn't, but that's the reality of the board.They did not, despite claims otherwise. It caused me some pain, which is why I originally commented on the proposal. 

I am advocating that it not a good thing for CircuitPython to ignore the technical specs and existing uses of the board and say "that's okay, we're just going to pretend it is actually footprint compatible with the D1 and assign these pins even though they're not actually used that way". That would cause further confusion among people who actually paid attention to the technical specs of the board. That would be committing a second wrong to try to make it right, which isn't going to happen.

deshipu, 2023-11-08:
I think that if someone designs their product based solely on how pins for a part are defined in a third party product like CircuitPython, then they have bigger problems than whether there is a default uart defined. It seems like you are arguing for the sake of winning the argument by now, not for any practical reason.

If we add a default UART object to the S2 Mini, it will work perfectly fine, and it will save some code and checking pin numbers for anyone using a shield that uses an UART. The fact that some other board can't use those pins for some other purpose is irrelevant. It's also irrelevant what those pins do before CircuitPython boots, because we are talking about using it in CircuitPython here.

I get that you have some kind of a grudge towards the designers of that board, but that really is no reason to make its use any more inconvenient for the users.

romkey, 2023-11-08:
CircuitPython doesn't exist in a vacuum. If it were irrelevant what happened before CircuitPython booted then footprints wouldn't matter outside of power and CPU control pins, and on CPUs that could accommodate it, CircuitPython could just declare that pin 1 was always RX because wouldn't that be nice for everyone? Having different pin definitions inside and outside of CircuitPython seems a sure recipe for more confusion.

I actually really like the manufacturer; I've used the D1 mini and the old Lolin32 boards quite a bit. We still have many of them in use at our hackerspace. They hit a very nice sweet spot on price, reliability and functionality.

dhalbert, 2023-11-08:
I would like to cool this down.  I think different people have different definitions of the word "default".

On Adafruit CircuitPython Espressif boards, we are generally careful _not_ to assign TXD0 and RDX0 to `board.TX` and `board.RX`, because they are special and can send data during boot, etc. We are also careful not to label those pins as just "TX" and "RX" on the silkscreen: we use "TX0" or "TXD0" and "RX0" or "RXD0".

However, we have many third-party boards (see the list below) that have assigned `board.TX` and `board.RX` to GPIO43 and GPIO44, and maybe it's too late to undo that. We should have asked them not to do that during the review process and to use `board.TX0` or `board.TXD0`, etc.

If these third parties label these pins as TX/RX on the silkscreen, that further complicates things. And now with Lolin not being consistent, I'm not sure what to do. We could change the names for `board.TX` and `board.RX` on the Lolin S3 Mini, but removing pins will break existing code.

Or, we can just say it's not _that_ important, but in the future, try to encourage people not to do this, and maybe start adding `TXD0` etc. as aliases on the boards that do this.

```
adafruit_qualia_s3_rgb666  # this is a mistake and we remove those definitions from this board
ai_thinker_esp_12k_nodemcu
atmegazero_esp32s2
bpi_bit_s2
bpi_leaf_s3
bpi_picow_s3
brainboardz_neuron
crumpspace_crumps2
deneyap_mini
deneyap_mini_v2
electroniccats_bastwifi
es3ink
espressif_esp32s2_devkitc_1_n4
espressif_esp32s2_devkitc_1_n4r2
espressif_esp32s2_devkitc_1_n8r2
espressif_esp32s3_devkitc_1_n32r8
espressif_esp32s3_devkitc_1_n8
espressif_esp32s3_devkitc_1_n8r2
espressif_esp32s3_devkitc_1_n8r8
espressif_esp32s3_devkitc_1_n8r8_hacktablet
espressif_esp32s3_devkitm_1_n8
espressif_esp32s3_lcd_ev
espressif_esp32s3_usb_otg_n8
espressif_hmi_devkit_1
espressif_kaluga_1
espressif_kaluga_1.3
espressif_saola_1_wroom
espressif_saola_1_wrover
franzininho_wifi_wroom
franzininho_wifi_wrover
gravitech_cucumber_m
gravitech_cucumber_ms
gravitech_cucumber_r
gravitech_cucumber_rs
hexky_s2
hiibot_iots2
lilygo_tembed_esp32s3
lilygo_ttgo_t8_esp32_s2_wroom
lilygo_ttgo_t8_s2_st7789
lolin_s2_mini
lolin_s3
lolin_s3_mini
magiclick_s3_n4r2
makerfabs_tft7
microdev_micro_s2
mixgo_ce_serial
mixgo_ce_udisk
muselab_nanoesp32_s2_wroom
muselab_nanoesp32_s2_wrover
smartbeedesigns_bee_data_logger
smartbeedesigns_bee_motion_s3
smartbeedesigns_bee_s3
targett_module_clip_wroom
targett_module_clip_wrover
unexpectedmaker_blizzard_s3
unexpectedmaker_feathers2
unexpectedmaker_feathers2_neo
unexpectedmaker_feathers2_prerelease
unexpectedmaker_feathers3
unexpectedmaker_nanos3
unexpectedmaker_pros3
unexpectedmaker_tinys2
unexpectedmaker_tinys3
waveshare_esp32_s2_pico_lcd
waveshare_esp32_s3_zero
yd_esp32_s3_n16r8
yd_esp32_s3_n8r8
```

romkey, 2023-11-08:
I had a pretty contrived example of another bad thing that can happen that I totally won't post now :)

I know that my advocacy here has been frustrating. I'm not trying to be a troll, just trying to fight entropy and I'm sorry for being annoying. There are going to be confused and frustrated users no matter how this matter goes.

dhalbert, 2023-11-08:
> There are going to be confused and frustrated users no matter how this matter goes.

Yes, it's about reducing the support burden, and often there's no good choice. As another example, we didn't provide `board.I2C()`, etc. on the Pi Pico. This was not so much because there weren't silkscreen markings, but because there were different "default" I2C pins defined by different language implementations and also different educational material. See #4121. "That's the great thing about standards: there are so many."

So we avoided the support burden of "sorry, the default pins aren't what you thought", but added a new support burden "how come `board.I2C()` doesn't work?".

bato3, 2023-11-09:
@deshipu you stuck to this UART, and the problem is that the SCK, MISO, MOSI lines are defined, but there is no board.i2c().

# Problem:

As everyone knows, there are only numbers on the Lolin s2 mini board.

The problem is what you get when you type "Lolin s2 mini pinout" into Google. This is either an [official pinout from Wemos](https://www.wemos.cc/en/latest/s2/s2_mini.html#pin), without I2C/SPI, [or an "independent" work](https://openmuscle.org/esp32-s2-mini-pinout-2/), not as defined in Circuitpython.

## The current situation looks like a job half done:

- There is SCL, SDA, SPI(), where SCL, SDA are [defined only in the schema.](https://www.wemos.cc/en/latest/_static/files/sch_s2_mini_v1.0.0.pdf)
- There is SCK, MISO, MOSI, _there is no I2C()_, with SCK, MISO, MOSI being defined only in the schema.
- There are defined A0, D0-D7, which I don't know where they came from. (OK, I know, they are on D1 mini Lite, and in the D1 mini v3.1 documentation. D1 mini v4 no longer mentions them. And they are added because: "Micropython has it")

## BTW: 
The SCL, SDA, SCK, MISO, MOSI positions are what Wemos invented. According to the datasheet [they should be here](https://www.studiopieters.nl/esp32-s2/):

[image]

# Solution(s)

I see two paths:
- remove all non-standard things (A0, D0-D7, according to [rpi situation](https://github.com/adafruit/circuitpython/issues/4121#issuecomment-775457783)) and lose backwards compatibility, or:
- take it upon yourself _to fix the world_ and publish "Circuitpython Lolin S2 mini pinout", with an indication of what has been "improved" and why.

### And what with `board.i2c()`?

dhalbert, 2023-11-09:
I suggest this to start:
- Make the S2 mini and S3 mini as consistent as possible, given that there are add-on boards (an "ecosystem") that expect I2C, etc. to be in certain positions.
- Remove all the `Dn` pin aliases, because they don't seem to be consistent, and are not on the original Wemos diagrams. There are several third-party diagrams, but they are not official, as you point out.
- _Maybe_ remove the `An` pins also, because their numbers correspond to analog peripherals, not GPIO numbers. But there's an argument to leave them.

Now, questions:
- Do the add-on boards all (or nearly all) use the same pins for I2C? If so, add `board.I2C()` as appropriate.
- Do the add-on boards use SPI, and do they use consistent pins? If so add `board.SPI()` as appropriate.
- Do the add-on boards use UART, and do they use consistent pins? This is more questionable, given the TXD0/RXD0 situation.

I did some searching for Wemos add-on boards, and did not find a lot of useful stuff. We have even had an issue with counterfeit Wemos boards in the past. See #7256 .

In general I would lean toward not adding a plethora of pin aliases. The users can always look up what pins do, given the pin mappings.

If there is official documentation that suggests certain pin names (e.g., from Wemos, "here's how to use CircuitPython on a Lolin <something>"), then that is guidance. But the official pin diagrams I see are nothing but GPIO numbers.

It is not really our responsibility to try to clean up the messes of third-party manufacturers and ecosystems that are not consistent. We rely on the people who submit PR's to understand the ecosystems. Ideally the manufacturers should submit the PR's, but sometimes they don't care. We don't usually own these boards, and we don't provide official support.

If those of you who own these boards would like to discuss what the pins should be, that is great. Please do it in a polite manner and reach a consensus. I've given some guidance in the previous posts. If there's a disagreement, go with leaving out aliases. Maybe a discord thread or a live meeting would be a better place to have a discussion. GitHub issues are not a great conversational tool.

deshipu, 2023-11-09:
> Do the add-on boards all (or nearly all) use the same pins for I2C? If so, add board.I2C() as appropriate.

Yes, there is a great number of various sensor boards that use I2C, and following a trend set back in the ESP8266 days by Arduino, they all use D1 Mini pins 4 and 5. A lot of other boards that are not compatible with D1 Mini also use pins 4 and 5 for I2C, I have even seen that on Pi Pico shields.

I also made a number of shields for the D1 Mini myself, and even sold a couple hundred of them, and I also used pins 4 and 5 for I2C on them:
- https://hackaday.io/project/13278-servo-breakout-for-wemos-d1-mini
- https://hackaday.io/project/18120-d1-mini-matrix-shield
- https://hackaday.io/project/19371-d1-mini-x-pad-shield
- https://hackaday.io/project/21500-d1-mini-analog-shield
- https://hackaday.io/project/27566-d1-mini-expander-shield
- https://hackaday.io/project/164561-accelerometer-shield-for-d1-mini
- https://hackaday.io/project/182342-camera-shield-for-s2-mini

The last one is specifically for the S2 Mini and S3 Mini boards.

> Do the add-on boards use SPI, and do they use consistent pins? If so add board.SPI() as appropriate.

There is a number of display shields that use SPI. I made two such shields myself, and sold a couple dozens of them:
- https://hackaday.io/project/19371-d1-mini-x-pad-shield
- https://hackaday.io/project/165983-d1-mini-ui-shield

> Do the add-on boards use UART, and do they use consistent pins? This is more questionable, given the TXD0/RXD0 situation.

I'm actually not aware of such a shield. I thought I've seen a GPS shield, but I can't find it right now.

romkey, 2023-11-10:
@dhalbert That aligns with my philosophy on these kinds of issues. Thank you for weighing in, I'm sorry this is consuming any of your time.

bato3, 2023-11-10:
I'm sorry if you got the impression that I was rude. I could have been carried away by emotions. (I use Google Translate for this - my English is too poor).
I'm looking for someone with whom I could repair at least a small part of the world. **_If this does not agree with the rules adopted by the Circuitpython maintainers - then I accept it._** (Not that I like it, but I don't have enough energy to discuss politics.)

OT: I looked through the Wemos github, and it looks abandoned.

### Currently, we can we split Wemos products on the following groups:
- full hardware and descriptive compatibility: D1 mini Lite (board), 95% D1 mini Shields
- full hardware compatibility: D1 mini, D1 mni Pro (green), S3 mini, C3 mini, c3 pico
- full software compatibility: S2 mini
- compatible with an additional shield: (shields) e-ink, OLED 0.96 - they "invented" [LOLIN I2C Port](https://www.wemos.cc/en/latest/d1_mini_shield/tft_i2c_connector.html#i2c-connector), [TFT Connector](https://www.wemos.cc/en/latest/d1_mini_shield/tft_i2c_connector.html#tft-connector)
- Connector compatible: D32 pro, S2 Pico, S3, S3 PRO
- wtf ;) : D32, W600-PICO

## Or maybe we could approach the topic from the other side: 
What do the [official expansion shields](https://www.wemos.cc/en/latest/d1_mini_shield/index.html) say about this?
- Wemos focuses on I2C communication. These are the same pins on all boards. Also has `I2C Connector`.
- There are RX, TX labels on all shields (but I didn't see a shield, where they were connected/used)
- On all shields, the labels A0, D0-D7, 3V3, GND are in the same places. Additionally: RST, 5V are also in the same places, but may have different names.
- e-Papier boards are incompatible with the D1 mini pinout (require an intermediary: TFT I2C Con)
- Some [boards](https://www.wemos.cc/en/latest/d1_mini_shield/oled_0_96.html) only contain: LOLIN I2C Port
- There are two pin-compatible SPI() boards: [TFT I2C Con](https://www.wemos.cc/en/latest/d1_mini_shield/tft_i2c_connector.html), [Micro SD Card Shield.](https://www.wemos.cc/en/latest/d1_mini_shield/micro_sd.html) However, none of them uses CS/SS defined as [Lolin D1 mini](https://www.wemos.cc/en/latest/d1/d1_mini.html) board. (LOL) Some [external work](https://www.aliexpress.com/item/32879814667.html), that uses `D8` as `CS`.

# Answers:
> Do the add-on boards all (or nearly all) use the same pins for I2C? If so, add board.I2C() as appropriate.

- all D1 mini style IC boards and shields (add-on boards) use this same pins for `I2C`.

> Do the add-on boards use SPI, and do they use consistent pins? If so add board.SPI() as appropriate.

- all D1 mini style IC boards and shields (add-on boards) use this same pins for `SPI`. All 3 shields, uses different IO for CS/SS

> Do the add-on boards use UART, and do they use consistent pins? This is more questionable, given the TXD0/RXD0 situation.

1. All shields (add-on boards) are labeled as `RX`, `TX`, but I haven't seen any board using UART. (Funny situation: describe and not use)
2. IC boards:
 - all D1: labeled `RX`, `TX`, schematic labeled as `RX`, `TX`
 - all C3: labeled `RX`, `TX`, schematic labeled as `U0TXD`, `U0RXD`
 - S2 mini: default JTAG, can be used as UART (U0: NC; U1 are on DAC pins: 17, 18)
 - S3 mini: no labels, pinout labeled `RX`, `TX`, schematic annotated as `RX`, `TX` chip footprint uses `U0TXD`, `U0RXD`
 - Wemos and Espressif don't use `TXD0`, `RXD0`  ;)

> Remove all the Dn pin aliases, because they don't seem to be consistent, and are not on the original Wemos diagrams. There are several third-party diagrams, but they are not official, as you point out. 

- All shields (add-on boards)  use `Dn`, also some D1 IC boards use `Dn`. In later products only GPIOs are used with possible additional labels for UART, SPI, I2C.

> Maybe remove the An pins also, because their numbers correspond to analog peripherals, not GPIO numbers. But there's an argument to leave them.

- Many boards (shields, IC) has `A0` labels, as main label
- S2, S3, are not labeled, but GPIO can be used as ADC.
- C3 pico, C3 mini v1 are extra labeled as `A0`
- **_C3 mini v2.1 are labeled as `A3`, `A0` is in different place (LOL)_** (GPIO to ADC is the same, but they reversed position GPIO0-GPIO3)

# My propositions:
- I see no contraindications for each D1 type board to have board.SPI(), board.I2C() defined
- I think that in the circuitpython documentation it should be additionally (and very strongly) highlighted that "Wemos D1 mini pinout" is used. (or remove `Dn`)
- I would like to ask you to fix the world and publish a picture with the pinout.

cristians-12, 2024-01-11:
Hello, i'm new on programming the lolin s2, which pins are TX/RX? Should i specify any pins as TX/RX like this?
SerialPort.begin(9600, SERIAL_8N1, RXD, TXD);

bato3, 2024-05-24:
@cristians-12 There are no default RX, TX pins on this board, but according to the Lolin board convention you should use pins 39, 37, so:

```py
import board

SerialPort.begin(9600, SERIAL_8N1, board.IO37, board.IO39);
```

According to ESP policy you can use/define any pin as hardware I2C, SPI, and UART