adafruit/circuitpython #8544: New pin label: DAC

bato3, 2023-10-30:
In Espressif chips there are usually 2 DAC pins available. In the uC of the ESP32-S2, ESP32-S3 family, the majority of the pins can be used freely as SPI, UART, but this does not apply to DAC.

[The DAC examples](https://learn.adafruit.com/circuitpython-essentials/circuitpython-analog-out) uses A0 labels (from M0), but this is not true for S2: DAC1 = IO17, DAC2 = IO18 (A0 = IO3).

By the way: What do you think about the 'TOUCH' labels? It would be a nice option so that you can see available options in `dir(board)`

tannewt, 2023-10-30:
`board` names pins based on the silkscreen of the board. So, this naming is up to the folks who design the board. You can also use `microcontroller.pin` to access the espressif names of the pins. That'll be consistent across boards.

bato3, 2023-10-31:
In fact, when I think about it deeper now, I think it was a stupid proposition.

On one hand, it would be nice to know about the functions of the tile - with labels like in internet pinouts.

However, even if I use `board.DAC`, I would still have to check the GPIO corresponding to him.

However, I will use the entry:
```py
dac1 = AnalogOut(board.IO17)
```
This code will still remain legible, and I will have fewer problems to verify the correctness of connecting the elements to the tile.

My proposal of the new label, partly resulted from the internal resistance of using the A0 label - [which is used in the guide.](https://learn.adafruit.com/circuitpython-essentials/circuitpython-analog-out) I wanted to divide that A0 was clearly associated with an analogue input. And that there is 2 form for an analog output.

### Updated proposal:
That is why I think that a better solution will be a change in uC in an example, to one that forces us to use IO labels and use `dac` variable instead `analog_out`.

It is possible?

tannewt, 2023-10-31:
It is possible, but not our policy. Typically in Arduino board designs, A0 and maybe A1 have a DAC. It is up to the board designer to label the board that way if it is possible. It isn't up to CircuitPython.

We could add a section similar to this though: https://learn.adafruit.com/circuitpython-essentials/circuitpython-i2c#wheres-my-i2c-2985160

dhalbert, 2024-01-24:
Closing as I think the discussion is complete.

closed 2024-01-24 by dhalbert