adafruit/circuitpython #8547: 'Keys' object has no attribute 'events'  and no attribute 'keys_count'

Anutrix, 2023-10-31:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.2 on 2023-10-27; ESP32-S3-DevKitC-1-N8 with ESP32S3
```

### Code/REPL

```python
print("ESP32-S3 here")

import board
import keypad

keys = keypad.Keys([board.IO6,], value_when_pressed=False, pull=True)

while True:
    print(dir(keys))
    event = keys.events.get()
    # event will be None if nothing has happened.
    if event:
        print(event)
```

### Behavior

```
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
main.py output:
ESP32-S3 here
['__class__', '__enter__', '__exit__', 'deinit', 'reset']
Traceback (most recent call last):
  File "main.py", line 10, in <module>
AttributeError: 'Keys' object has no attribute 'events'

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```

### Description

This is a basic example from https://learn.adafruit.com/key-pad-matrix-scanning-in-circuitpython/keys-one-key-per-pin.
Added `print(dir(keys))` to further highlight the issue.

deshipu, 2023-10-31:
Fixed in https://github.com/adafruit/circuitpython/pull/8535

Anutrix, 2023-10-31:
@deshipu Thx a lot. Had been struggling with this for days thinking it was issue in my code. Guess I just need to wait for next alpha(9.0.0-alpha.3) now.

closed 2023-10-31 by dhalbert

tannewt, 2023-10-31:
> Guess I just need to wait for next alpha(9.0.0-alpha.3) now.

The latest "Absolute Newest" builds off of S3 will have the fix. No need to wait.