adafruit/circuitpython #8554: AttributeError: .show(x) removed. Use .root_group = x File "adafruit_portalbase/graphics.py", line 60, in __init__

aalhard, 2023-11-02:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.2-8-g7715ff09cc on 2023-10-31; Adafruit PyPortal Titano with samd51j20
>>> [D
... 
>>>
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2020 Richard Albritton for Adafruit Industries
#
# SPDX-License-Identifier: MIT

import time
import board
import microcontroller
import displayio
import busio
from analogio import AnalogIn
import neopixel
import adafruit_adt7410
from adafruit_bitmap_font import bitmap_font
from adafruit_display_text.label import Label
from adafruit_button import Button
import adafruit_touchscreen
from adafruit_pyportal import PyPortal

# ------------- Constants ------------- #
# Sound Effects
soundDemo = "/sounds/sound.wav"
soundBeep = "/sounds/beep.wav"
soundTab = "/sounds/tab.wav"

# Hex Colors
WHITE = 0xFFFFFF
RED = 0xFF0000
YELLOW = 0xFFFF00
GREEN = 0x00FF00
BLUE = 0x0000FF
PURPLE = 0xFF00FF
BLACK = 0x000000

# Default Label styling
TABS_X = 0
TABS_Y = 15

# Default button styling:
BUTTON_HEIGHT = 40
BUTTON_WIDTH = 80

# Default State
view_live = 1
icon = 1
icon_name = "Ruby"
button_mode = 1
switch_state = 0

# ------------- Functions ------------- #
# Backlight function
# Value between 0 and 1 where 0 is OFF, 0.5 is 50% and 1 is 100% brightness.
def set_backlight(val):
    val = max(0, min(1.0, val))
    try:
        board.DISPLAY.auto_brightness = False
    except AttributeError:
        pass
    board.DISPLAY.brightness = val
[... 464 lines cut ...]
                if i == 6 and view_live == 3:  # only if view3 is visible
                    b.selected = True
                    while ts.touch_point:
                        pass
                    print("Sound Button Pressed")
                    pyportal.play_file(soundDemo)
                    b.selected = False
```

### Behavior

soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 148, in <module>
  File "adafruit_pyportal/__init__.py", line 128, in __init__
  File "adafruit_pyportal/graphics.py", line 44, in __init__
  File "adafruit_portalbase/graphics.py", line 60, in __init__
AttributeError: .show(x) removed. Use .root_group = x

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

dhalbert, 2023-11-02:
This is not a bug, but a deprecation introduced in 8.x.x, and finalized in 9.0.0.
Change
```py
board.DISPLAY.show(splash)
```
to
```py
board.DISPLAY.root_group = splash
```

RetiredWizard, 2023-11-02:
The user code needs the fix referenced by Dan above, but the library also needs the fix. I'll go ahead and submit a PR on that.