adafruit/circuitpython #8559: sdcardio example unable to successfully 'storage.mount(vfs, "/sd")'

ladyada, 2023-11-05:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.2-13-gdc5845fa41 on 2023-11-04; Adafruit Camera with ESP32S3
```

### Code/REPL

```python
import sdcardio
import board
import storage

sdcard = sdcardio.SDCard(board.SPI(), board.CARD_CS)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")
```

### Behavior

Traceback (most recent call last):
  File "code.py", line 12, in <module>
OSError: [Errno 19] No such device

### Description

the SD card is definitely 'started' - i dont get `OSError: no SD card`
note same thing occurs when using adaruit_sdcard library `sdcard = adafruit_sdcard.SDCard(board.SPI(), digitalio.DigitalInOut(board.CARD_CS))`
so its something in `storage`?

ladyada, 2023-11-05:
assignin' dan/scott because this is possibly related to the 9x merge

tannewt, 2023-11-06:
On the metro M7 SD with a version of 9.x I was able to list stuff until I ran out of memory. I had to increase the pystack too.

tannewt, 2023-11-07:
I tried on an S3 devkit c + sdcard breakout but I'm not seeing CS toggle. I have no idea why. Blink works on an S3 feather from the same source.

tannewt, 2023-11-14:
I think this is a hardware issue, not a software one. The same build for a slightly different S3 ESP kit works ok. Printing out all of the io mux and gpio registers looks correct.

closed 2023-11-14 by tannewt