adafruit/circuitpython #8564: Expand synthio.Note.amplitude range to allow CV-like outputs

CedarGroveStudios, 2023-11-06:
When using a control voltage (CV) -like object in synthio such as Envelope and LFO to modulate a Note's `amplitude`, only positive control values are used with the resultant output value placed above the baseline. For example, a sine-wave LFO created to span negative and positive values to modulate a note's `frequency` is rectified if used to modulate a note's `amplitude`. This becomes important when an analog output of just the Envelope or LFO is needed (rather than the modulated sound of the Note oscillator).

I believe the solution would be to expand the range of `Note.amplitude` from `ZERO` and `ALMOST_ONE` to `-ALMOST_ONE` and `ALMOST_ONE`. The result would be the same for amplitude values >=0. When amplitude < 0, the signal phase would be inverted (180-degrees) while proportionally adjusting the amplitude.

An alternative would be to create a `synthio.CV` object that operates similarly to `synthio.Note` so that CV signals can be altered by other CV signals, LFOs, mixers, and filters. This would approach would permit tuning the UI to include parameters strictly for CV signals (such as offset, phase, and range) and could reduce the potential confusion created by changing the range of allowable `Note.amplitude` values.

A full discussion including test code can be found on this Playground page: https://adafruit-playground.com/u/CGrover/pages/eurorack-control-voltage-cv-signals-from-synthio

jepler, 2023-11-18:
I do not remember what my motivation was for not permitting a negative amplitude.  _however_ note that in `synthio_note_step` the type of the `loudness[2]` parameter is `uint16_t`, so converting a negative number may lead to unexpected results.

I think that if the functionality you're looking for can be implemented without introducing a new "CV" class I'd rather do it that way.

CedarGroveStudios, 2023-11-21:
Ah, of course there will be a thread on that parameter to pull that will stretch into other areas of synthio -- I appreciate your reminder to look deeper. My limited knowledge of C++ might slow me down a bit, but I'll see what I can find. Hopefully it's just used as a multiplier to control gain.

Regarding negative amplitude values, allowing the amplitude to control a signal's phase is very "op-amp" like and would likely resolve the current CV issue without resorting to a new class.

PR #8642 (2023-11-21): Synthio: Allow amplitude to be signed (+ other stuff)
This closes #8564 (ping @cedargrovestudios)

While working on this issue, I noticed that the "circuitpython" tests were not being run by "make test" and therefore not during CI. (this probably happened during one of the recent micropython merges; I marked the changes as CIRCUITPY-CHANGE) There were several problems with the tests themselves. I fixed them, updated some expected outputs, and restored running the tests when expected.

The final unrelated change to tools/git-checkout-latest-tag.sh ensures that if a tag isn't fetched, the script [... cut]
closed 2023-11-21 by CedarGroveStudios