adafruit/circuitpython #8570: NotImplementedError: main_clock for espressif port

lookfwd, 2023-11-09:
### CircuitPython version

```python
CircuitPython 9.0.0-alpha.2 on Adafruit Feather ESP32-S3 4MB Flash 2MB PSRAM (and I guess everything on the espressif port
```

### Code/REPL

```python
import audiobusio
with audiobusio.I2SOut(bit_clock=board.A0,
                       word_select=board.A1,
                       data=board.A2,
                       main_clock=board.A3
                      ) as audio:
    pass
```

### Behavior

I'm get `NotImplementedError: main_clock`

### Description

I attempt to use ESP32-S3 with SGTL5000. I get `NotImplementedError: main_clock` when I try to use the `I2SOut`. An implementation similar to [mimxrt10xx](https://github.com/adafruit/circuitpython/blob/59463683dd207925f93f71d252a7ebd2deb4cb29/ports/mimxrt10xx/common-hal/audiobusio/I2SOut.c#L83)'s must get ported for espressif [here](https://github.com/adafruit/circuitpython/blob/main/ports/espressif/common-hal/audiobusio/I2SOut.c#L61) and the `common_hal_audiobusio_i2sout_deinit` function in the same file. The relevant ESP32-S3 library is [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html).

dhalbert, 2023-11-09:
Note that for the SGTL5000 the MCLK has a PLL that allows you to use any convenient clock signal that is 8-27 MHz. So for now maybe you could hook up the MCLK input to a PWM output, assuming it will run that high for Espressif.

> CLOCKING
> Clocking for the SGTL5000 is provided by a system
> master clock input (SYS_MCLK). SYS_MCLK should be
> synchronous to the sampling rate (Fs) of the I2S port.
> Alternatively any clock between 8.0 and 27 MHz can be
> provided on SYS_MCLK and the SGTL5000 can use an
> internal PLL to derive all internal and I2S clocks. This allows
> the system to use an available clock such as 12 MHz
> (common USB clock) for SYS_MCLK to reduce overall
> system costs.

lookfwd, 2023-11-09:
Great idea! I used a signal generator to give it 12MHz and SGTL5000 "woke up" and started to respond in I2C, but I2S wasn't working. Turning the PLL on and doing lots of configuration is a nice interim solution.

dhalbert, 2023-11-09:
Looks like PWM will do 12MHz:
```py
import pwmio
p = pwmio.PWMOut(board.IO6, duty_cycle=32768, frequency=12000000)
```

lookfwd, 2023-11-09:
That's great! If only it was in sync with `SCLK` and `LRCLK`, it could save us from the extra configuration and the power consumption of the PLL.

lookfwd, 2023-11-09:
Beeps nicely! It even plays MIDI :D

```
with audiobusio.I2SOut(bit_clock=board.A0, word_select=board.A1, data=board.A2, main_clock=board.A3) as audio:
    init_sgtl5000()
    melody = synthio.MidiTrack(b"\0\x90H\0*\x80H\0\6\x90J\0*\x80J\0\6\x90L\0*\x80L\0\6\x90J\0" +
                       b"*\x80J\0\6\x90H\0*\x80H\0\6\x90J\0*\x80J\0\6\x90L\0T\x80L\0" +
                       b"\x0c\x90H\0T\x80H\0\x0c\x90H\0T\x80H\0", tempo=640)
    audio.play(melody)
```

[image]

closed 2023-11-09 by dhalbert