adafruit/circuitpython #8574: Adafruit-Qualia-S3 "pystack exhausted"

jacobmarble, 2023-11-10:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.2-18-gb7b62873f8 on 2023-11-08; Adafruit-Qualia-S3-RGB666 with ESP32S3
```

### Code/REPL

```python
import adafruit_qualia
```

### Behavior

```console
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_qualia/__init__.py", line 32, in <module>
  File "adafruit_qualia/network.py", line 31, in <module>
  File "adafruit_portalbase/network.py", line 28, in <module>
  File "adafruit_io/adafruit_io.py", line 30, in <module>
  File "adafruit_minimqtt/adafruit_minimqtt.py", line 47, in <module>
  File "adafruit_minimqtt/matcher.py", line 20, in <module>
RuntimeError: pystack exhausted
```

### Description

Based on [an older issue](https://github.com/adafruit/circuitpython/issues/7643), I tried setting `CIRCUITPY_PYSTACK_SIZE=4000` in `settings.toml`. The result of `import adafruit_qualia` is then:

```console
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_qualia/__init__.py", line 32, in <module>
  File "adafruit_qualia/network.py", line 31, in <module>
  File "adafruit_portalbase/network.py", line 28, in <module>
  File "adafruit_io/adafruit_io.py", line 483, in <module>
  File "adafruit_io/adafruit_io.py", line 678, in IO_HTTP
MemoryError: memory allocation failed, allocating 184 bytes
```

### Additional information

- I've erased and factory-reset this device, then installed "absolute newest" CircuitPython 9.0.0-alpha.2-18-gb7b62873f8.
- I've installed the base bundles for this device via "circup install adafruit_qualia"; no surprises there.

RetiredWizard, 2023-11-10:
This is the same behavior that led me to open #8573.

jacobmarble, 2023-11-10:
I found the "breaking version". This version allows me to `import adafruit_qualia` without error:
```console
Adafruit CircuitPython 9.0.0-alpha.2-16-gccd667d97a on 2023-11-06; Adafruit-Qualia-S3-RGB666 with ESP32S3
```
This subsequent version does not:
```console
Adafruit CircuitPython 9.0.0-alpha.2-17-g01be5f402e on 2023-11-08; Adafruit-Qualia-S3-RGB666 with ESP32S3
```

RetiredWizard, 2023-11-10:
@jacobmarble Try placing `CIRCUITPY_HEAP_START_SIZE=1024000` in your settings.toml file.

Mark-MDO47, 2023-11-10:
I tried CIRCUITPY_HEAP_START_SIZE=1024000 with adafruit-circuitpython-adafruit_qualia_s3_rgb666-en_US-20231109-3ca9802.uf2 and it still gave pystack exhausted.

I tried adafruit-circuitpython-adafruit_qualia_s3_rgb666-en_US-20231106-ccd667d.uf2 and this worked for me.

tannewt, 2023-11-10:
> I found the "breaking version". This version allows me to `import adafruit_qualia` without error:

Do you still have the pystack size change with the old version? I think there are two issues here:

1. We need to adjust pystack size for newer MP core. (Covered by this issue)
2. We need to tune the split heap behavior. (Covered by #8573 )

jacobmarble, 2023-11-10:
> > I found the "breaking version". This version allows me to `import adafruit_qualia` without error:
> 
> Do you still have the pystack size change with the old version?

I commented the `CIRCUITPY_PYSTACK_SIZE` line in settings.toml before finding the breaking version.

tannewt, 2023-11-10:
Ok, that's interesting. Maybe I broke the pystack size with the split heap PR. Still worth a separate issue (this one.)

Mark-MDO47, 2023-11-11:
Using these suggested params in settings.toml
CIRCUITPY_HEAP_START_SIZE=1024000
CIRCUITPY_PYSTACK_SIZE=4000

FWIW I did the following:
factory reset
installed adafruit-circuitpython-adafruit_qualia_s3_rgb666-en_US-20231106-ccd667d.uf2
ran qualia_paint.py
RESULT: WORKS

factory reset
installed adafruit-circuitpython-adafruit_qualia_s3_rgb666-en_US-20231109-3ca9802.uf2
ran qualia_paint.py
RESULT: FAILS see below

soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
fw_version: 10, chip_type: 11
Traceback (most recent call last):
  File "code.py", line 23, in <module>
MemoryError: memory allocation failed, allocating 460800 bytes

Code done running.

NOTE: line 23 is as follows:
bitmap = displayio.Bitmap(graphics.display.width, graphics.display.height, 65535)

RetiredWizard, 2023-11-11:
That's progress :grin: try doubling the heap start to 2048000

Mark-MDO47, 2023-11-13:
Steps:
- Erase and factory reset
- copied in adafruit-circuitpython-adafruit_qualia_s3_rgb666-en_US-20231109-3ca9802.uf2
- D:\lib is empty; copied in adafruit-circuitpython-bundle-9.x-mpy-20231109
- D:\settings.toml now includes
    CIRCUITPY_HEAP_START_SIZE=1024000
    CIRCUITPY_PYSTACK_SIZE=4000
- copied example qualia_paint.py to D:\code.py changing to "Displays.ROUND21" on line 13
  $$$$$$$$ FAIL $$$$$$$$
    code.py output:
    fw_version: 10, chip_type: 11
    Traceback (most recent call last):
      File "code.py", line 23, in <module>
    MemoryError: memory allocation failed, allocating 460800 bytes
- D:\settings.toml now includes
    CIRCUITPY_HEAP_START_SIZE=2048000
    CIRCUITPY_PYSTACK_SIZE=4000
- qualia_paint.py to D:\code.py changing to "Displays.ROUND21" on line 13
  $$$$$$$$ FAIL $$$$$$$$
    code.py output:
    fw_version: 10, chip_type: 11
    Traceback (most recent call last):
      File "code.py", line 23, in <module>
    MemoryError: memory allocation failed, allocating 460800 bytes
- D:\settings.toml now includes
    CIRCUITPY_HEAP_START_SIZE=4096000
    CIRCUITPY_PYSTACK_SIZE=4000
- qualia_paint.py to D:\code.py changing to "Displays.ROUND21" on line 13
  $$$$$$$$ SUCCESS - paint program runs $$$$$$$$
    code.py output:
    fw_version: 10, chip_type: 11

RetiredWizard, 2023-11-13:
I think this is helpful information, especially since you're example is pretty easy to reproduce. I've been using a START_SIZE of 3072000 on boards that have enough RAM and 1536000 on the boards with less.

PR #8598 (2023-11-13): Fix pystack size
Original code used uint32_t* so `/ sizeof(size_t)` was needed. It is a uint8_t* now so that division makes it 4x smaller. Whoops!

Fixes #8574
commit ae07ac17 (2023-11-13): Fix pystack size

Original code used uint32_t* so `/ sizeof(size_t)` was needed. It
is a uint8_t* now so that division makes it 4x smaller. Whoops!

Fixes #8574
closed 2023-11-13 by dhalbert