adafruit/circuitpython #8597: Memory error on metro esp32s3 rev b

jerryneedell, 2023-11-13:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.4 on 2023-11-12; Adafruit Metro ESP32S3 with ESP32S3
```

### Code/REPL

```python
import ipaddress
import wifi
import socketpool
import time
import adafruit_requests
```

### Behavior

>>> import adafruit_requests
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
MemoryError: 
>>>

RetiredWizard, 2023-11-13:
The newest releases need CIRCUITPY_HEAP_START_SIZE set to a pretty large value in settings.toml. Take a look at #8574

jerryneedell, 2023-11-13:
Works normally with 8.2.7  -- same settings.toml

tannewt, 2023-11-13:
> The newest releases need CIRCUITPY_HEAP_START_SIZE set to a pretty large value in settings.toml. Take a look at #8574

This is a temporary workaround. We shouldn't need it for 9.x.

Is `adafruit_requests` a .py file or .mpy file?

jerryneedell, 2023-11-13:
OK 
```CIRCUITPY_HEAP_START_SIZE=1024000
CIRCUITPY_PYSTACK_SIZE=4000
``` works

jerryneedell, 2023-11-13:
> > The newest releases need CIRCUITPY_HEAP_START_SIZE set to a pretty large value in settings.toml. Take a look at #8574
> 
> This is a temporary workaround. We shouldn't need it for 9.x.
> 
> Is `adafruit_requests` a .py file or .mpy file?

.py

jerryneedell, 2023-11-13:
I had to set both:
CIRCUITPY_HEAP_START_SIZE=1024000
CIRCUITPY_PYSTACK_SIZE=4000

setting only
CIRCUITPY_HEAP_START_SIZE=1024000

resulted in

Adafruit CircuitPython 9.0.0-alpha.4 on 2023-11-12; Adafruit Metro ESP32S3 with ESP32S3
>>> import wifi_test
ip 10.0.0.106
8.8.4.4
ping 0.0
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "wifi_test.py", line 19, in <module>
  File "adafruit_requests.py", line 711, in get
  File "adafruit_requests.py", line 650, in request
RuntimeError: pystack exhausted
>>>

tannewt, 2023-11-14:
pystack should be fixed by #8598

jerryneedell, 2023-11-14:
With the latest PR now only the 
CIRCUITPY_HEAP_START_SIZE=1024000
is needed.

Adafruit CircuitPython 9.0.0-alpha.4-5-ged7c714fe2 on 2023-11-13; Adafruit Metro ESP32S3 with ESP32S3
>>> import wifi_test
ip 10.0.0.106
8.8.4.4
ping 0.0
200
This is a test of Adafruit WiFi!
If you can read this, its working :)

200
{'timezone': 'America/New_York', 'utc_datetime': '2023-11-14T17:40:00.336035+00:00', 'raw_offset': -18000, 'client_ip': '71.232.151.158', 'dst_from': None, 'unixtime': 1699983600, 'utc_offset': '-05:00', 'datetime': '2023-11-14T12:40:00.336035-05:00', 'week_number': 46, 'abbreviation': 'EST', 'day_of_year': 318, 'day_of_week': 2, 'dst': False, 'dst_offset': 0, 'dst_until': None}
done
>>> 

> pystack should be fixed by #8598

Yes it is. Now only the 
CIRCUITPY_HEAP_START_SIZE=1024000
is needed

Adafruit CircuitPython 9.0.0-alpha.4-5-ged7c714fe2 on 2023-11-13; Adafruit Metro ESP32S3 with ESP32S3
>>> import wifi_test
ip 10.0.0.106
8.8.4.4
ping 0.0
200
This is a test of Adafruit WiFi!
If you can read this, its working :)

200
{'timezone': 'America/New_York', 'utc_datetime': '2023-11-14T17:40:00.336035+00:00', 'raw_offset': -18000, 'client_ip': '71.232.151.158', 'dst_from': None, 'unixtime': 1699983600, 'utc_offset': '-05:00', 'datetime': '2023-11-14T12:40:00.336035-05:00', 'week_number': 46, 'abbreviation': 'EST', 'day_of_year': 318, 'day_of_week': 2, 'dst': False, 'dst_offset': 0, 'dst_until': None}
done
>>>

jerryneedell, 2023-11-14:
and as suggested, reducing it by a factor of 4 still works
CIRCUITPY_HEAP_START_SIZE=256000
works

anecdata, 2023-11-14:
What's the determining factor on whether the new heap will grow automatically vs. needing a custom `CIRCUITPY_HEAP_START_SIZE=` in `settings.toml`... some interaction with `esp-idf` memory usage?

jerryneedell, 2023-11-14:
CIRCUITPY_HEAP_START_SIZE=128000 also works now.
CIRCUITPY_HEAP_START_SIZE=65536 does not....

anecdata, 2023-11-14:
I can run the Quotes Request Test from:
https://learn.adafruit.com/pico-w-wifi-with-circuitpython/pico-w-requests-test-adafruit-quotes
on
`Adafruit CircuitPython 9.0.0-alpha.4-5-ged7c714fe2 on 2023-11-13; Raspberry Pi Pico W with rp2040`
and
`Adafruit CircuitPython 8.2.7 on 2023-10-19; Adafruit QT Py ESP32-S3 4MB Flash 2MB PSRAM with ESP32S3`
without a custom value for `CIRCUITPY_HEAP_START_SIZE=`

tannewt, 2023-11-14:
> What's the determining factor on whether the new heap will grow automatically vs. needing a custom `CIRCUITPY_HEAP_START_SIZE=` in `settings.toml`... some interaction with `esp-idf` memory usage?

It will grow automatically either way when it needs more memory. I'm not sure why setting the start size helps. *Maybe* fragmentation of the IDF heap? I'm planning on looking into it more.

RetiredWizard, 2023-11-14:
I was in over my head but when I looked at it, it looked to me like each allocation wasn't continuous so the small allocations were fragmenting the memory.

PR #8605 (2023-11-14): Fix split heap on ESP
It was setting the DMA capability requirement which excludes PSRAM.

Fixes #8597. Fixes #8573.
commit 1f235574 (2023-11-14): Fix split heap on ESP

It was setting the DMA capability requirement which excludes PSRAM.

Fixes #8597. Fixes #8573.
closed 2023-11-14 by tannewt