adafruit/circuitpython #8612: Check CIRCUITPY_ESP_FLASH_MODE settings on Espressif boards

dhalbert, 2023-11-15:
`CIRCUITPY_ESP_FLASH_MODE` used to only be used in a Makefile target for flashing the board. Now it is used to select the proper `sdkconfig*` file. Many ESP32-S3 boards, and maybe other ESP32-xx boards have it set to `dio`, when it may be able to be set to `qio`.

My understanding is that `dio` will work, just slower. (Am I misunderstanding?). However, if we fix the ones that are now `dio` to `qio`, we should see improved performance, I believe.

We should audit the board `mpconfigboard.mk` files to see which ones can be changed.

tannewt, 2023-11-16:
> My understanding is that `dio` will work, just slower. (Am I misunderstanding?). However, if we fix the ones that are now `dio` to `qio`, we should see improved performance, I believe.

This is my understanding too.

bill88t, 2023-11-16:
If this change is what I think it is, (actual flash speed while running), the speed improvement should be pretty considerable, just like with psram.
Actual seconds for big projects like mine.

Though it needs to be actually tested for every board.
I will do it for every board I have, though it takes time.

Also, won't esp32 (non -Sx) also benefit from it?
iirc my camera board has octal connections and is esp32.

dhalbert, 2023-11-16:
>Though it needs to be actually tested for every board.

For various classes of boards, we shouldn't need to re-test. For instance, the Espressif chips with integral flash, and  modules with in-module flash, will all have the same settings (for each chip or module). It's only when there are external, not pre-packaged flash chips that we have to check the board for how many data lines are connected.

commit 582ac6e6 (2024-02-07): Boost flash and RAM speeds on ESP boards

We inherited a bunch of DIO from 8.x which used it for ROM flash
settings. Pretty much everything should be able to do QIO.

S2, S3 and ESP32 all support 80mhz as well and ESP modules tend to
work at that speed too. Only on ESP32 do we leave PSRAM at 40mhz
because the 80/80 mode uses an extra SPI resource that we don't
want to use.

Fixes #8612
closed 2024-02-07 by dhalbert