adafruit/circuitpython #8618: PyGamer Thermal Camera with AMG8833 needs CIRCUITPY_PYSTACK_SIZE = 2900 to avoid "pystack exhausted"

miltieIV2, 2023-11-15:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.4 on 2023-11-12; Adafruit PyGamer with samd51j19
```

### Code/REPL

```python
https://learn.adafruit.com/elements/3054787/download?type=zip
PyGamer_Thermal_Camera//CircuitPython 9.x
```

### Behavior

Default CIRCUITPY_PYSTACK_SIZE 1536 won't even get to splash screen:
```
code.py output:
Traceback (most recent call last):
  File "code.py", line 17, in <module>
  File "adafruit_pybadger/__init__.py", line 16, in <module>
  File "adafruit_pybadger/pygamer.py", line 132, in <module>
  File "adafruit_pybadger/pygamer.py", line 62, in __init__
  File "adafruit_lis3dh.py", line 427, in __init__
RuntimeError: pystack exhausted
```

with less than CIRCUITPY_PYSTACK_SIZE = 2900, there is a crash when pushing buttons:
```
Traceback (most recent call last):
  File "code.py", line 354, in <module>
  File "code.py", line 124, in update_histo_frame
  File "adafruit_display_text/__init__.py", line 420, in text
  File "adafruit_display_text/label.py", line 440, in _set_text
  File "adafruit_display_text/label.py", line 423, in _reset_text
  File "adafruit_display_text/label.py", line 280, in _update_text
  File "adafruit_bitmap_font/glyph_cache.py", line 54, in get_glyph
RuntimeError: pystack exhausted
```

### Additional information

This makes me wonder how many projects will need a stack bump and if the default should be cranked up.

miltieIV2, 2023-11-15:
@@FoamyGuy
FoamyGuy

dhalbert, 2023-11-15:
This is probably fixed by #8598. I am releasing 9.0.0-alpha.5 very soon and you can re-test.

closed 2023-11-16 by miltieIV2