adafruit/circuitpython #8631: QTPY ESP32S3 9.0.0A5: fails on BLERadio startScan()

kyril96, 2023-11-18:
**Adafruit CircuitPython 9.0.0-alpha.5 on 2023-11-15; Adafruit QT Py ESP32-S3 no psram with ESP32S3**

**code.py output (the first few times) :**
]0;🐍192.168.0.144 | code.py | 9.0.0-alpha.5\scanning
Traceback (most recent call last):
  File "code.py", line 5, in <module>
  File "adafruit_ble/__init__.py", line 271, in start_scan
ValueError: Prefix buffer must be on the heap
]0;🐍192.168.0.144 | 271@adafruit_ble/__init__.py ValueError | 9.0.0-alpha.5\
Code done running.

**After repeated runs without the reset button or an unplug, I instead get this:**
code.py output:
]0;🐍192.168.0.144 | code.py | 9.0.0-alpha.5\Traceback (most recent call last):
  File "code.py", line 1, in <module>
  File "adafruit_ble/__init__.py", line 28, in <module>
  File "adafruit_ble/advertising/__init__.py", line 98, in <module>
MemoryError: memory allocation failed, allocating 96 bytes
]0;🐍192.168.0.144 | 98@adafruit_ble/advertising/__init_ MemoryError | 9.0.0-alpha.5\
Code done running.

**The code itself:**
```
ble = BLERadio()
print("scanning")
bleConn=None
for ad in ble.start_scan():
    print ("9.0.0A5 goes kaboom before this can be printed.")
    break
print ("done")

```

**Switching UF2 and lib back to 8.2.6 works as expected:**
code.py output:
]0;🐍192.168.0.144 | code.py | 8.2.6\scanning
9.0.0A5 goes kaboom before this can be printed.
done
]0;🐍192.168.0.144 | Done | 8.2.6\
Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
]0;�192.168.0.144 | REPL | 8.2.6\
Adafruit CircuitPython 8.2.6 on 2023-09-12; Adafruit QT Py ESP32-S3 no psram with ESP32S3

tannewt, 2023-11-20:
This may be split heap related.

dhalbert, 2023-11-20:
I have an easy fix for this, but I'm debugging other things at the same time.

PR #8645 (2023-11-22): ESP32-S3 BLE fixes
Fixes #8631.

Seems to fix ESP32-S3 BLE acting as a simple central. See https://forums.adafruit.com/viewtopic.php?t=206024 for user problems. I will point the users in that thread to the artifacts once this has built.

Tested on a BerryMed Pulse Oximeter with the library's simpletest.

Notes:
- #8631 fix fixes a problem exposed by the new split heap use.
- `BLE_ERR_SUCCESS` is 0, but was being used in places where the return values are actually `BLE_HS_*`. Nimble has multiple sets of error return codes, but no corresponding types, so it's hard to figure out which codes to us [... cut]
closed 2023-11-22 by dhalbert