adafruit/circuitpython #8635: 9-alpha5 reported wrong line number for exception by 1 more than reality

tyeth, 2023-11-18:
### CircuitPython version

```python
9.0.0.0-alpha5
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2021 Jose David M.
# SPDX-FileCopyrightText: 2023 Melissa LeBlanc-Williams for Adafruit Industries
#
# SPDX-License-Identifier: MIT

# NOTE: Make sure you've set up your settings.toml file before running this example
# https://learn.adafruit.com/getting-started-with-web-workflow-using-the-code-editor/
"""
This example shows a web address QR on the display
"""

import time
import adafruit_qualia
import os
import json
from adafruit_qualia.graphics import Graphics, Displays
from adafruit_qualia.peripherals import Peripherals
import displayio
import terminalio
import vectorio

base = adafruit_qualia.Qualia(Displays.BAR320X820, rotation=90)

# Background Information
graphics = base.graphics

# Set up Peripherals
peripherals = base.peripherals

# Set display to show
display = base.display

# display.auto_refresh=False

def display_qr_and_text(qr_data, text, x=0, y=0, relative_x_from_center=None, relative_y_from_center=None, scale=10, include_qr_offset=False, include_text_x_offset=-20):
    global graphics, display, base
    qr_size = scale
    if relative_x_from_center is not None:
        x = (display.width // 2) + relative_x_from_center
    if relative_y_from_center is not None:
        y = display.height // 2 + relative_y_from_center
    if include_qr_offset:
        x_offset = (qr_size + 5) * scale
        y_offset = (qr_size + 4) * scale
        x -= x_offset
        y -= y_offset
    qr = graphics.qrcode(qr_data, qr_size=scale, x=x, y=y, qr_color=0x121212, qr_bg_color=0x0000aa, return_group=True)
    display.root_group.append(qr)
    base.add_text(text_position=(x + include_text_x_offset, 0.9 * display.height), text=text, text_scale=3, text_wrap=0, text_maxlen=180, text_color=0xFFFFFF)
    return (x,y)

def get_item_at(x, y, group):
    for item in group:
        if not hasattr(item, "x") or not hasattr(item, "y"):
            continue
        if isinstance(item, displayio.Group):
            return get_item_at(x, y, item)
        elif isinstance(item, displayio.TileGrid):
            print("Checking TileGrid", json.dumps(item)
            if item.x <= x and x < (item.x + item.width) and item.y <= y and y < (item.y + item.height):
                print("Touched TileGrid", json.dumps(item))
                return item
    return None

# Define the triggerTouch function or import it from a module
def triggerTouch(x, y, finger):
    global display
    print("Touched", x, y, finger)
    #iterate through the tilegrids and groups (which contain tilegrids) inside root_group, testing the x,y against the bounding box of each tilegrid
    item = get_item_at(x, y, display.root_group)
    if item and hasattr(item, "touch_callback"):
        item.touch_callback(x, y, finger)
        return True
    return False

def example_touch_callback(x, y, finger):
    print("Touched Example Callback, toggling hidden status")
    item.hidden = not item.hidden

wifi = adafruit_qualia.network.WiFi()

if wifi.is_connected and wifi.ip_address in (None, "0.0.0.0"):
    print("Wifi almost connected, Waiting for IP")
    counter=50
    while not wifi.is_connected:
        print(".", end="")
        time.sleep(0.1)
        if(counter==0):
            break

if not wifi.is_connected and wifi.ip_address not in (None, "0.0.0.0"):
    # using circuitpython get portalbase wifi information so we can construct the url from IP and port and password (both read from settings.toml using os.getenv("field", default_value) )
    ip = wifi.ip_address
    port = os.getenv("CIRCUITPY_WEB_API_PORT", "80")
    password = os.getenv("CIRCUITPY_WEB_API_PASSWORD", "password")

    # WebPage to show in the QR
    webpage = f"http://{ip}:{port}/"

else:
    # if not then craft a fake url that starts with http:// but infact has a clever use of @ (%40) and :(%3a) etc to really just link to about:blank except about:wifi_not_connected, like a low grade url escape that only a QR reader would fall for as it's technically not a valid url; be better if it was a javascript: type url that supported alerts etc, maybe just a website with good customisation, a bit like lmgtfy.com or a meme generator
    #webpage="http://%40%3a%2f%2fabout%3awifi_not_connected"
    webpage = "https://learn.adafruit.com/circuitpython-with-esp32-quick-start/setting-up-web-workflow"
    ip=None

# QR size Information
qr_size = 9  # Pixels
scale = 8

# Generate QR code bitmap for webpage
display_qr_and_text(webpage, "IP: " + str(ip) if ip else "", relative_x_from_center=100,y=10, scale=scale, include_text_x_offset=-30)

# Generate Wifi joining QR code
qrdata=f"WIFI:S:{os.getenv('CIRCUITPY_WIFI_SSID', '*Unset*')};T:{os.getenv('CIRCUITPY_WIFI_TYPE', 'WPA')};P:{os.getenv('CIRCUITPY_WIFI_PASSWORD', '*Unset*')};" if ip else "https://www.adafruit.com"
(x,y)=display_qr_and_text(qrdata, "SSID: " + os.getenv("CIRCUITPY_WIFI_SSID", "*Unset* read help->"), relative_x_from_center=-330,y=10, scale=scale, include_text_x_offset=-30)
os.listdir("/")
if not ip:
    # Create a Bitmap from a PNG file
    image = displayio.OnDiskBitmap("/circuitpythonlogo-color888.bmp")

    palette = displayio.ColorConverter(input_colorspace=displayio.Colorspace.RGB888)

    # # Create a Palette
    # palette = displayio.Palette(32)
    # for i in range(2,32):
    #     palette[i] = 0x000000 if i % 2 == 1 else 0xFFFFFF
    # palette[0] = 0XFFFFFF
    # palette[1] = 0x000000
    # palette.make_transparent(31)
    # # palette[2] = 0xFFFFFF
    # # palette[3] = 0xFF0000
    # # palette.make_opaque(0)
    # # palette.make_opaque(1)
    # # palette.make_opaque(2)
    # # palette.make_transparent(3)

    # Create a TileGrid
    tilegrid = displayio.TileGrid(
        bitmap=image,
        pixel_shader=palette,
        x=x,
        y=y,
    )
    tilegrid.touch_callback = example_touch_callback
    img_group = displayio.Group()
    img_group.append(tilegrid)

    display.auto_refresh=False
    # Add the TileGrid to the display
    display.root_group.append(img_group)
else:
    for item in display.root_group:
        if isinstance(item, displayio.TileGrid):
            item.touch_callback = example_touch_callback

display.refresh()
display.auto_refresh=True

counter=0
while True:
    counter+=1

    if graphics.touch.touched:
        try:
            finger=-1
            for touch in graphics.touch.touches:
                finger+=1
                x = touch["x"]
                y = touch["y"]
                if (
                    not 0 <= x < graphics.display.width
                    or not 0 <= y < graphics.display.height
                ):
                    continue  # Skip out of bounds touches
                triggerTouch(x, y, finger)
        except Exception as e:
            print(e)
            pass
    if peripherals.button_up:
        peripherals.backlight = True
    if peripherals.button_down:
        peripherals.backlight = False
    time.sleep(0.1)
```

### Behavior

soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 61
SyntaxError: invalid syntax

Code done running.

### Description

You'll need the qualia cpy library and my modified fork of the portalbase for the multiple qrcodes and bgcolour. I also have a circuitpython logo taken from the homepage in the root folder
https://github.com/tyeth/Adafruit_CircuitPython_PortalBase/tree/multiple_qrcodes

### Additional information

Fixing the print statement on line 60 rather than 61 solved the error.
It's worth stating that I was exhausting the pystack with some poorly written recursion shortly before, and may not have rebooted, but the board seemed happy to continue after further code changes, up to this point of not spotting line 61 was 60 🤦

tyeth, 2023-11-18:
I rebooted with the error back in place, here's the attached code file with all needed assets + libraries (in py format) in case it's some formatting issue with the example and copyright I only half modified
[line61isNot60.zip](https://github.com/adafruit/circuitpython/files/13401626/line61isNot60.zip)

anecdata, 2023-11-19:
I think that's normal.
```py
=== # test 1
=== 
=== foo = "test1"
=== 
=== print("Checking TileGrid", foo
=== )  # close the parens on the next line is OK
=== print("some more stuff")
=== 
=== 
Checking TileGrid test1
some more stuff

=== # test 2
=== 
=== foo = "test1"
=== 
=== print("Checking TileGrid", foo
=== print("some more stuff")  # line 6
=== 
Traceback (most recent call last):
  File "<stdin>", line 6
SyntaxError: invalid syntax
```

tyeth, 2023-11-19:
I get ya, I suppose I've been spoilt by other languages that show the offending lines contents and first character/token that breaches syntax. I wonder if the exception shouldn't be the line that never finishes correctly, so on a multi-line if / print, the print is never complete so the start is the beginning of the error. Maybe a more informative syntax error could be given containing the offending statements with line numbers.

tyeth, 2023-11-19:
maybe this migrates to a feature request / close as wont fix

dhalbert, 2023-11-19:
The CPython parser is better about this:
```py
print (3
if 0:
    pass
```
CPython:
```
  File "/home/halbert/foo.py", line 1
    print (3
          ^
SyntaxError: '(' was never closed
```
CircuitPython/MicroPython:
```
  File "/home/halbert/foo.py", line 2
SyntaxError: invalid syntax
```
But the MicroPython parser just works differently, so I wouldn't characterize this as "wrong".

Closing as inherent, but it's an interesting observation.

closed 2023-11-19 by dhalbert