adafruit/circuitpython #8643: Hard fault: memory access or instruction error found intermittently on RP2040 CAN Bus Feather

0nix, 2023-11-21:
### CircuitPython version

```python
8.2.7
```

### Code/REPL

```python
RCVRID = 0x001 #CHANGE THIS LINE
# RP2040 CAN Bus Feather

from time import sleep
import time
import board
import neopixel
from rainbowio import colorwheel
from digitalio import DigitalInOut, Pull
from adafruit_mcp2515.canio import Message, RemoteTransmissionRequest
from adafruit_mcp2515 import MCP2515 as CAN
import adafruit_mcp2515.canio as canio

COLORS = {
    0x0: (0,0,0,0),
    0x1: (255,0,0,0),
	0x2: (0,255,0,0),
	0x3: (0,0,255,0),
	0x4: (255,255,0,0),
	0x5: (0,255,255,0),     
	0x6: (255,0,255,0),
	0x7: (0,0,0,255),
	0x8: (255,128,0,0),
	0x9: (0,128,255,0),
	0xA: (128,0,255,0), 
	0xB: (255,0,128,0), 
	0xC: (0,128,0,0),
	0xD: (100,149,237,0),
	0xE: (139,0,0,0),
	0xF: (218,165,32,0),
	0x10: (32,178,270,0),
	0x11: (255,69,0,0),
	0x12: (0,255,255,128),
    0x13: 0x13,  #Rainbow 		= 0x13, 
    0x14: 0x14 #FastRainbow	= 0x14,
    #Skip			= 0xA0
} 
GREEN = (0,255,0)
RED = (255,0,0)
WHITE = (255,255,255)
NONE = (0,0,0)

LLPin = DigitalInOut(board.D5)
ULPin = DigitalInOut(board.D6)
URPin = DigitalInOut(board.D9)
LRPin = DigitalInOut(board.D10)

LL_LightOn = COLORS[0x3]
LL_LightOff = COLORS[0x0]
UL_LightOn = COLORS[0x3]
UL_LightOff = COLORS[0x0]
UR_LightOn = COLORS[0x3]
UR_LightOff = COLORS[0x0]
LR_LightOn = COLORS[0x3]
LR_LightOff = COLORS[0x0]

RainbowColor = 0x0
FastRainbowSet =  ((255, 0, 0, 0), (255, 40, 0, 0), (255, 150, 0,0), (0, 255, 0,0), (0, 0, 255, 0), (180, 0, 255,0))

LightsPin = board.D11

[... 254 lines cut ...]
                can_send_msg(RCVRID,bytes(msg,'ascii'))
                
                lastMsg = msg
        else:
            if (msg == blankMsg):
                pass
            else:
               #print(msg)
               statusPixel.fill((0,0,255))
            
    except Exception as e:
        print(e)
        statusPixel.fill((255,255,0))

# rest of code

```

This has not been as consistently tested as the previous string approach, but it is showing some promise. This does not explain how some boards are able to deal with this issue just fine while others do not.

Furthermore, the boards may be exposed to a higher chance of ESD since they are in proximity to acrylic boards where people are expected to step on and slide. While the acrylic does not come into direct contact with the boards, I have noticed they accumulate static electricity so this may also play a factor. 

The code also does not sleep between while loops, would this be a factor?

Any and all help is tremendously appreciated.

bill88t, 2023-11-21:
You may want to test 9.0-alpha builds, since the memory management system changed a good bit.

0nix, 2023-11-22:
Would you advise the use of 9.0-alpha on a production setting with code like this? It is not going to change in functionality.

dhalbert, 2023-11-22:
Testing the alpha build might give us some clues about what the problem is. I don't think we would necessarily advise it for use in production. but if it works better than 8.2.x that is a selling point. "Stable" is in the eye of the beholder. Alpha builds are generally not deliberately unstable.

0nix, 2023-11-25:
This script lacks a sleep at the end of the loop. Would that improve anything? All examples include a call to sleep, and I assume this is a best practice scenario.

anecdata, 2023-11-25:
Senders may typically have a delay between messages, but if the device is primarily listening for messages, there should be no delay and you want to minimize the other processing in the main loop. There's not much buffering in the CAN hardware (2 receive buffers and 3 send buffers), so loops need to complete in a timely manner relative to the frequency of incoming messages.

tannewt, 2024-03-05:
Please retest this with an absolute newest build and try to trim down the test code. I was unable to reproduce this with a disconnected bus and no other devices.

SamantazFox, 2025-01-20:
Could it be related to https://github.com/adafruit/Adafruit_CircuitPython_MCP2515/pull/28 ?