adafruit/circuitpython #8652: occasional HardFault on esp32-s3 with camera & httpserver

jepler, 2023-11-22:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.2-13-gdc5845fa41 on 2023-11-04; Adafruit Camera with ESP32S3
```

### Code/REPL

```python
# I do not have a minimized standalone program that reproduces the problem
```

### Behavior

```
You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.
```

### Description

The camera itself is stable, it's only since adding use of wifi & httpserver to my program that this has started occurring.

### Additional information

yes I know this is an old version of circuitpython and I should try a current one

michalpokusa, 2023-11-24:
I have seem multiple reports of hard-fault with httpserver, although none with clear indication of what might be causing them.

If I may suggest one thing, please try setting the `poll_interval` on `serve_forever()` or add `time.sleep()` between calling `.poll()`.
https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/blob/1c5c41a2c4cf76bf1305b5b77cf3911b8dd977ad/adafruit_httpserver/server.py#L175

Even better would be making it a async function with `asyncio.sleep()` and running the program like in [this example](https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/blob/main/examples/httpserver_websocket.py)

When I was trying to find a reason for hard-faults, it seemed like it helped with stability of the server.

tannewt, 2024-01-31:
I'm going to close this because there hasn't been any specific reproducing code added. Feel free to file another issue if it crops up again.

closed 2024-01-31 by tannewt