adafruit/circuitpython #8657: Errors writing files to root of CIRCUITPY on MacOS Sonoma

dmcomm, 2023-11-25:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.0-alpha.5 on 2023-11-15; Raspberry Pi Pico W with rp2040
8.2.8; Raspberry Pi Pico W
8.2.2; Raspberry Pi Pico W
8.2.2; Arduino Nano RP2040 Connect
```

### Code/REPL

```python
None
```

### Behavior

Hi, 3 people in my group are unable to copy files to the root of CIRCUITPY on MacOS Sonoma. Sorry for the second-hand report - if I start things off here, hopefully others can comment if needed.

On first sight, everyone thinks this is #8449, but it's different.

- Trying to copy a file to the root of CIRCUITPY often fails.
- Had a report of 'The Finder can't complete the operation because some data in "README.md" can't be read or written. (Error code -36)'
- Python running on MacOS gives OSError 5 (creating additional confusion with 8449).
- Copying into "lib" works fine.
- Copying to a subfolder and then moving to the root works more often, but not reliably.
- Flash Nuke and clean install of CircuitPython didn't help.

dhalbert, 2023-11-25:
We think this is the same underlying issue as #8449. It manifests in more than one way

dmcomm, 2023-11-25:
I hadn't seen any mentions of the writes outright failing from the perspective of MacOS (except when neglecting to eject properly).

Also I'm not quite clear in the other issue about 8MB vs 16MB. The Nano Connect's CIRCUITPY is a little under 16, does that count as "small" or not?

eightycc, 2023-11-25:
> Also I'm not quite clear in the other issue about 8MB vs 16MB. The Nano Connect's CIRCUITPY is a little under 16, does that count as "small" or not?

The mounted filesystem size affects whether or not Sonoma delays writes to the USB MSC. We've found that filesystems in excess of 8M do not exhibit delayed writes. A board with a 16MB flash will not experience delayed writes. But, because its resulting filesystem will be smaller than 8MB, one with an 8MB flash will.

dmcomm, 2023-11-25:
> We've found that filesystems in excess of 8MB do not exhibit delayed writes

The Arduino Nano RP2040 Connect has a filesystem bigger than 8MB, so that means it experiences this issue but not what's described in 8449.

dhalbert, 2023-11-25:
Below is a writeup I did onrequest to send to someone external, on November 13. It describes both problems.

When attempting to characterize the problem, I instrumented the writes that were being requested via USB. I could not actually find a clear pattern of delayed writes on <8MB vs no delayed writes on larger filesystems. I saw delayed writes on both, but it was not clear how the writes were being delayed in metadata vs data, and when what should have been an atomic transaction was complete. This was with the latest Sonoma developer beta at the time, so it's possible that the bug is evolving over time.

@dmcomm - exactly which version of Sonoma are you using? Please also file a Feedback report with Apple describing your particular problem. Include details of the CIRCUITPY you are using, by doing Get Info on thefilesystem or similar.

-----

We are seeing issues with macOS Sonoma which are greatly complicating the use of small FAT external disk drives. CircuitPython presents a CIRCUITPY drive via USB MSC, and there are pico-sdk and Arduino examples which can also present USB MSC drives.

Before Sonoma, macOS wrote data and metadata to USB drives essentially immediately. We now see that Sonoma can delay writing metadata for drives <= 8MiB in size. We have seen delays of up to 25 seconds. Critically, the delay between writing data and metadata is long, which means that the filesystem can exist in an inconsistent state for many seconds.

Windows and Linux in the past have also had write delays. But their delays do not appear to be during a single filesystem transaction: they do not leave the filesystem in a quiescent inconsistent state for many seconds at a time.

The write delays cause havoc with CircuitPython's autoreload behavior, which assumes that the filesystem is stable after about 0.75 seconds after a write has finished.

Writing more than one file at once can also cause I/O errors. That is an even more serious problem, not fixable by doing `sync` or similar. Here is an example of write errors:

```
%  cp -X 8k1.bin 8k2.bin 8k3.bin 8k4.bin /Volumes/CIRCUITPY
cp: /Volumes/CIRCUITPY/8k1.bin: fcopyfile failed: Input/output error
cp: /Volumes/CIRCUITPY/8k1.bin: fchmod failed: Input/output error
cp: /Volumes/CIRCUITPY/8k2.bin: fcopyfile failed: Input/output error
cp: /Volumes/CIRCUITPY/8k2.bin: fchmod failed: Input/output error
cp: /Volumes/CIRCUITPY/8k3.bin: fcopyfile failed: Input/output error
cp: /Volumes/CIRCUITPY/8k3.bin: fchmod failed: Input/output error
cp: /Volumes/CIRCUITPY/8k4.bin: fcopyfile failed: Input/output error
cp: /Volumes/CIRCUITPY/8k4.bin: fchmod failed: Input/output error
```

The UF2 bootloader on RP2040 is not affected by this, because it presents as a much larger drive. But other UF2 bootloaders may present as smaller drives.

We are discussing and investigating this issue in https://github.com/adafruit/circuitpython/issues/8449.

We have attempted to look at traces of the write behavior by instrumenting filesystem writes in CircuitPython. We have not gotten clear data yet. There appear to be delayed writes in Ventura as well, but these are probably other filesystem transactions. We need to add more semantic instrumentation that details what the writes are doing.

eightycc, 2023-11-25:
> The Arduino Nano RP2040 Connect has a filesystem bigger than 8MB, so that means it experiences this issue but not what's described in 8449.

That's not good news. If you don't mind, please note this in #8449 so that we can track it all in one place. In the end, #8449 comes down to unpredictable delays between writes regardless of the symptoms described in the various entries.

dhalbert, 2023-11-25:
> I hadn't seen any mentions of the writes outright failing from the perspective of MacOS (except when neglecting to eject properly).

See https://github.com/adafruit/circuitpython/issues/8449#issuecomment-1804861258 (and what I just posted).

eightycc, 2023-11-25:
@dhalbert That failures are being seen on a ~15MB filesystem is concerning.

dhalbert, 2023-11-25:
Did we test filesystems >8MB but <16MB? Are those FAT16? I don't have the boards at hand to check at the moment.

eightycc, 2023-11-25:
I tested on an RP2040 Metro with 16MB. The filesystem was FAT16 and CP had my FAT12/16 partition ID patch. The resulting filesystem was ~15MB.

dhalbert, 2023-11-25:
Earlier today I wrote up what appears to be the best workaround right now:
https://learn.adafruit.com/welcome-to-circuitpython/troubleshooting#macos-sonoma-14-dot-x-disk-errors-writing-to-circuitpy-3160304

I'll refine this when I have time next week.

eightycc, 2023-11-25:
That's my favorite workaround, too. I can do some testing on the Metro tomorrow if you like. Let me know what you'd like me to try.

dhalbert, 2023-11-26:
OK, I'm going to close this in favor of #8449, based on @dmcomm's comment here: https://github.com/adafruit/circuitpython/issues/8449#issuecomment-1826863626

> Another member reported that Arduino Nano RP2040 Connect is affected, which it shouldn't be based on previous results here, because it has a filesystem >8MB. I'm trying to follow up with him about that.
> 
> > It was giving exactly the same symptoms, but flash nuke and fresh CircuitPython fixed the problem. Also tried another flash nuke and downgrading to 7.0.0 in case it was an issue left over from an older filesystem, but still fine. Must have been random filesystem corruption or something. Sorry for the false alarm!

closed 2023-11-26 by dhalbert